SpenceKonde/ATTinyCore #405: Arduino-cli support?

jmusther, 2020-05-27:
I've got arduino-cli version 0.10.0 installed and ATTinyCore 1.3.2, and when I try to compile with arduino-cli I get:

`#error "This version of the Arduino IDE is not supported for this part, upgrade Arduino 1.8.6 or higher or downgrade ATTinyCore to 1.2.1"`

I'm assuming that ATTinyCore is treating arduino-cli version 0.10.0 as an older version of the IDE, similarly to #389

But I'm assuming there's no fundamental reason ATTinyCore wouldn't work with arduino-cli?

per1234, 2020-05-27:
The `ARDUINO` macro is defined in platform.txt like so:
```
-DARDUINO={runtime.ide.version}
```
When using Arduino CLI or Arduino Pro IDE, the `runtime.ide.version` property is currently set to `10607`.

I'm not aware that there is any macro that indicates Arduino CLI is in use. It would be even less helpful if it set the `runtime.ide.version` property to the Arduino CLI version (currently `001000`), since the `ARDUINO` macro is expected to provide the IDE version. I have made a suggestion for how this could be improved on the arduino/arduino-cli repo's issue tracker: https://github.com/arduino/arduino-cli/issues/725. Feel free to provide input with any ideas for how this situation could be improved by the Arduino CLI developers over there.

I did add a note last month to the Arduino Platform Specification that the property's value is meaningless when using Arduino development software other than the Arduino IDE (you can still use the `ARDUINO` macro having been defined as an indicator that the code is being compiled by Arduino development software):
https://arduino.github.io/arduino-cli/dev/platform-specification/#global-predefined-properties
So at least this behavior is documented, but that doesn't provide a solution to this issue.

jmusther, 2020-05-27:
Ok thanks for the explanation, that makes sense.  Do you have any idea how I could work around this current limitation?

per1234, 2020-05-27:
The workaround is to add this option to your `arduino-cli compile` command:
```
--build-properties compiler.cpp.extra_flags=-DARDUINO=10812,compiler.c.extra_flags=-DARDUINO=10812
```
That overrides the default definition of the `ARDUINO` macro with one that sets its value to 10812.

commit 52e3d025 (2020-06-02): Fix #405 - work around arduino-cli bug

arduino-cli incorrectly reports version at 1.6.7, rather than whatever version it is based on. 

This will result in the error message not being shown on IDE 1.6.7 - but if they're using that godawful version of the IDE, they deserve what they get. On 1.6.7 and all versions of arduino-cli which report that version, a #warning is shown instead.
closed 2020-06-02 by SpenceKonde