SpenceKonde/ATTinyCore #409: No option for CLKOUT fuse

dperlman, 2020-06-12:
Hi there, I'm almost embarrassed to even mention something so simple, but the bootloader burning menu doesn't have the ability to set the CLKOUT fuse which causes the internal RC clock oscillator to output on XTAL2, for those ATTinys that have that option. This is a useful capability because it allows easy calibration of the oscillator using a frequency counter or oscilloscope if greater accuracy is needed.

Thanks, I really appreciate this project, it is remarkably easy to use!

SpenceKonde, 2020-06-18:
I can see the usefulness of the application - though you could also upload a sketch that outputs a squarewave based on a timer, which would work on a crappier scope.... 

My feeling with these "niche" fuse settings was that it was better to keep the core simple to use and reduce the tools submenu clutter (it's already too damned messy) - that's why we don't have options for CLKOUT (or external CLOCK, except on a couple of parts that don;t support external crystal)

As an aside, I did realize that, for tuning, I ought to add a tools menu to tell it a location to look for the OSCCAL value to use (probably EEPROM 0, last EEPROM address, or two bytes in flash before the Optiboot version for Optiboot boards (the latter has the advantages of being much less likely to get inadvertantly broken by user code, and that you can tell the difference between "set OSCCAL to 0xFF" and "no value saved here" though I'm pretty sure nothing that tried to tune the device to the "proper" clock speed would end up trying to set it to 0xFF.... 

Come to think of it, I'll be the shark way to tune would be to output serial characters (bitbang is fine - maybe that 1010101010 (counting the start and stop bits) one for ease of generation?), and have it start from OSCCAL=0, and when an I/O pin (connected to serial adapter DTR) went briefly,  print the character (maybe like, a few right after eachother?), then increment OSCCAL, repeat until OSCCAL=255.... and the adapter would be controlled by a python script, opening briefly (hence pulsing DTR low).... and then it would look for where it got correctly formed characters, and the middle of that group of correctly formed characters is where the correct cal was.... and then it could call AVRDude (tied to an ISP programmer connected to the device) to write the proper calibration value to that spot in the flash or EEPROM....

SpenceKonde, 2020-06-18:
Anyway, for advanced users that need such uncommon fuses to be set, they should be manually set via AVRdude. 

One of the most frequent usability complaints I receive is that I have too many tools submenus already, so I don't really want to add more without a really compelling use case.... and a tool that gets used once per part by an advanced user when connected to a 'scope doesn't, IMO, qualify.... Remember that with verbose upload, you can do "burn bootloader" to get the AVRdude command line, copy it out into a text editor and change the fuse(s) you want, and paste it into a command window - it gives complete paths to everything, so you don't need to go digging for avrdude binary or avrdude.conf or anything else.

closed 2020-06-18 by SpenceKonde