SpenceKonde/ATTinyCore #411: [Question] Possible to force soft Wire/I2C slave?

prandeamus, 2020-06-16:
Not sure if this is possible, so I'm asking for guidance. I am using ATTinyCore with an ATTiny85. It uses the differential ADC between ADC2 and 3, and reports values as an I2C slave device. All works well, and I've learned a lot more than I expected about I2C implementations which is a bonus.

I would love to add an additional (single ended, not differential) ADC on ADC1, but using PB2 for ADC1 is incompatible with using PB2 as SCK.  PB1 is unused in my current design. I understand that the current Wire implementation is a mash-up several different Wire implementations to cover all the hardware variants supported by the core. Here's the question: can I force a software/bitbanged version on an ATTiny85 to put the I2C slave on PB1 for SCK and PB0 for SCA? This would free up PB2 for additional ADC.

I can see an alternative approach to use the PB0/ADC0 pin as a weak single ended ADC - apparently if the voltage can be kept from dropping below 0.9*Vcc the reset won't be triggered. But I wondered if I had any software options for implementing I2C on different pins before I took that approach.

I know this is asking for the "moon on a stick" so if it can't be done, I do get it. No point in complaining too much when there are only 8 pins on the darn thing :)

prandeamus, 2020-06-17:
I may be able to answer my own question, and the answer is probably that you can't do it with configuration options. It seems to me that the I2C slave options either use USI or inbuilt I2C support, and are not bitbanged. Some of the I2C master options are done as bitbanging, but that's not what I'm looking for. I guess I need an independent I2C slave that uses bitbanging.

SpenceKonde, 2020-06-18:
Aaah, yeah sorry for the slow response here - you are correct. Since there's no part in ATTinyCore that doesn't have either a USI, a full-fledged TWI module, or a hardware-slave-only TWI module, there was never a need for me to hunt down a software I2C slave library.. 

It looks like there are a lot more I2C software masters than slaves - but this guy looks like he's made a... really scary looking Software I2C slave implementation? https://forum.mysensors.org/topic/6779/software-i2c-slave-for-arduino-uno-tft-shield-code 

pity he didn't separate out the I2C slave code from his application.... 

I suspect there are others which may be easier to use if you google vigorously.

closed 2020-06-18 by SpenceKonde

prandeamus, 2020-06-18:
You've gone beyond my expectations already for finding that software I2C slave which I had not previously found. Many thanks for your time spent on this!