SpenceKonde/ATTinyCore #412: OSCCAL tuning support

SpenceKonde, 2020-06-18:
The current way that OSCCAL tuning is done is pointless, as it requires modifying variants or board def to specify the tuned OSCCAL value, but this will vary from part to part. 

Instead.... for non-Optiboot boards, we should have Tools submenu to support loading OSCCAL from... maybe EEPROM 0, EEPROM (last), maybe? Something like that would be viable. And for Optiboot boards, we could look right before the Optiboot version number in flash.... I think we have enough space that we could have optiboot use that - if it's been written... I think it would be straightforward to have a python script that would use serial output to figure out the correct value of OSCCAL, and when found, call AVRdude to write that to flash.... this would let us finally put the goddamned tuning issue to bed. - an issue that I am entirely sick of being asked about...

nerdralph, 2020-06-25:
Coincidentally, I've been planning to implementing something like this with picoboot-lib.  When you flash the bootloader, it will have a tuner program included.  When avrdude first runs, the tuner program will time the duration of the first 1 or 2 STK_GET_SYNC bytes and adjust OSSCAL.  The OSCCAL value will be written to the last byte of flash (unused by the bootloader), and the user program will overwrite the tuner.  On subsequent reset, the bootloader will update OSCCAL with the value saved in flash.

SpenceKonde, 2020-06-25:
Sweet - how do you tune it though, is the question? Like - I can think of lots of ways to tune.... but not a clear way to do that *automatically* after bootloading...

I am very interested in this..... (and whatever method could be adapted to also handle the optiboot bootloaders....

nerdralph, 2020-06-25:
> 
> Sweet - how do you tune it though, is the question? Like - I can think of lots of ways to tune.... but not a clear way to do that _automatically_ after bootloading...
> 

Yes, it's fully automatic, based on how avrdude implements the STK500 protocol.  It's essentially building on what I've previously done to implement autobaud in the mega version of picoboot.
http://nerdralph.blogspot.com/2018/05/picoboot-adruino-with-autobaud.html

nerdralph, 2020-06-27:
picoboot-lib v0.3 with autotuning has been pushed, and is ready for testing.  I'll update the README later today.  I did my initial testing with a t85 that is 1-2% fast with the factory 0x5A OSCCAL.  After uploading the first sketch, the tuner usually picks 0x59, and sometimes 0x58.  The value is saved to the last byte of flash and then read at the start of the bootloader after every subsequent reset.
I managed to implement it without removing the EEPROM read functionality.

closed 2023-06-25 by SpenceKonde