SpenceKonde/ATTinyCore #414: Add Micronucleus support

freemovers, 2020-06-19:
@SpenceKonde, I started working on integrating the Micronucleus  support in your core. The board manager in Arduino has to know where the Micronucleus tools are located. As far as I can tell this can only be done by creating toolsDependencies in the JSON file. Are you able to change yours as follows:
`          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            },
           {
              "packager": "californiasteam",
              "name": "micronucleus",
              "version": "2.0a4"
           }
          ]`

and in tools you need the following (I linked mine to DigiStump)

`"tools":
            [
                {
                    "name": "micronucleus",
                    "version": "2.0a4",
                    "systems": 
                    [
                        {
                            "host": "x86_64-apple-darwin",
                            "archiveFileName": "micronucleus-2.0a4-osx.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-osx.tar.gz",
                            "checksum": "SHA-256:B5EB0C7B251CD88F4816186BB931855834141E71A28D90FB9E46788E483AA421",
                            "size": "51203"
                        },
                        {
                            "host": "i686-mingw32",
                            "archiveFileName": "micronucleus-2.0a4-win.zip",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-win.zip",
                            "checksum": "SHA-256:7027971118FDE88484AACB5D8CA7867D66E273839EC3B2592616829317BB70E4",
                            "size": "1712005"
                        },
                        {
                            "host": "i686-linux-gnu",
                            "archiveFileName": "micronucleus-2.0a4-linux32.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-linux32.tar.gz",
                            "checksum": "SHA-256:0D4286388EED28D1ECB29AFE81253F24F54D4F0A5C1B2C17507EABD504C595F8",
                            "size": "21909"
                        },
                        {
                            "host": "x86_64-linux-gnu",
                            "archiveFileName": "micronucleus-2.0a4-linux64.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-linux64.tar.gz",
                            "checksum": "SHA-256:1F545C0BB60E85A604901C2D7044772AC91776594C209C571DFEDAD4A70195B8",
                            "size": "22874"
                        }
                    ]
                }
            ]`
I the meantime I will work on the menu.

SpenceKonde, 2020-06-19:
For sure!

And yeah, github does that - issues always eat whitespace when you try to post formatted code. I mean, it's not like there are many people trying to discuss programming or show code here *shrug*

SpenceKonde, 2020-06-19:
Maybe email it to me with the formatting non-fucked?

SpenceKonde, 2020-06-19:
Also, I think I just realized that all these years, I could have been dealing with the 5v internal oscillator issue on the 841 much more gracefully!

freemovers, 2020-06-19:
Great, I just sent you the e-mail.

per1234, 2020-06-19:
>issues always eat whitespace when you try to post formatted code

Use code fencing and life is good:
https://guides.github.com/features/mastering-markdown/

freemovers, 2020-06-19:
```
"toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino5"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            },
           {
              "packager": "californiasteam",
              "name": "micronucleus",
              "version": "2.0a4"
           }
          ]

         "tools":
            [
                {
                    "name": "micronucleus",
                    "version": "2.0a4",
                    "systems":
                    [
                        {
                            "host": "x86_64-apple-darwin",
                            "archiveFileName": "micronucleus-2.0a4-osx.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-osx.tar.gz",
                            "checksum": "SHA-256:B5EB0C7B251CD88F4816186BB931855834141E71A28D90FB9E46788E483AA421",
                            "size": "51203"
                        },
                        {
                            "host": "i686-mingw32",
                            "archiveFileName": "micronucleus-2.0a4-win.zip",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-win.zip",
                            "checksum": "SHA-256:7027971118FDE88484AACB5D8CA7867D66E273839EC3B2592616829317BB70E4",
                            "size": "1712005"
                        },
                        {
                            "host": "i686-linux-gnu",
                            "archiveFileName": "micronucleus-2.0a4-linux32.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-linux32.tar.gz",
                            "checksum": "SHA-256:0D4286388EED28D1ECB29AFE81253F24F54D4F0A5C1B2C17507EABD504C595F8",
                            "size": "21909"
                        },
                        {
                            "host": "x86_64-linux-gnu",
                            "archiveFileName": "micronucleus-2.0a4-linux64.tar.gz",
                            "url": "https://github.com/digistump/DigistumpArduino/releases/download/1.6.5a/micronucleus-2.0a4-linux64.tar.gz",
                            "checksum": "SHA-256:1F545C0BB60E85A604901C2D7044772AC91776594C209C571DFEDAD4A70195B8",
                            "size": "22874"
                        }
                    ]
                }
            ]
```

SpenceKonde, 2020-06-19:
Wonderful, thanks!

freemovers, 2020-06-19:
I get "Error resolving FQBN: board" when I try to compile, but I think that is related to the toolsDependencies. @per1234 do you know where Arduino stores the information received from the JSON file like the tools and compiler information? For now I can create a copy of the JSON file and simulate on install through the board manager, but I was hoping there would be an easier way to add dependencies to Arduino.

SpenceKonde, 2020-06-19:
Does it persist after restarting ide? I see that when I fiddle with
boards.txt while ide is open

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

freemovers, 2020-06-19:
Got it. I still had a version of your core which I downloaded by using the board manager AND the latest core from Github in the Arduino/hardware folder. Looks like that was causing a conflict.

freemovers, 2020-06-19:
Digispark is not compiling now and I might need your help. It throws the following error: 
```
C:\Users\Documents\Arduino\hardware\ATTinyCore\avr\cores\tiny\wiring.c:30:28: note: in expansion of macro 'CLOCK_SOURCE'

 #if (F_CPU == 16500000L && CLOCK_SOURCE==6)
```
Looks like it is trying to include avr/boot.h, but that file does not exist.

SpenceKonde, 2020-06-19:
Huh? That file is part of the toolchain files... Can you post the complete error message? There's more before it that will tell us what's going on; did you add the build.clocksource property to the board? We use that so code can figure out what kind of source their clock is coming from (it gets used from platform.txt, and in a few places, it's used internally...):

0 - Internal 8MHz oscillator (with or without prescaling to a speed lower than 8MHz)
1 - External Crystal
2 - External Clock (only available within the core on the 48, 88 and 828, as described above)
3 - Internal WDT oscillator (not available on the x41, 1634, and 828)
4 - Internal ULP oscillator (available only on the x41, 1634, and 828)
5 - Internal 4MHz oscillator (present only on the x313 parts - if the 8MHz internal oscillator is prescaled to 4MHz, CLOCK_SOURCE will be 0, not 5)
6 - Internal PLL (x5 and x61 only)

SpenceKonde, 2020-06-19:
Edit, oh, right, I see...

SpenceKonde, 2020-06-19:
But that error message indicates some problem expanding CLOCK_SOURCE, not a problem finding avr/boot.h?

freemovers, 2020-06-19:
This is embarrassing. Typo in the variable name and I renamed it to clocksource instead of build.clocksource.

freemovers, 2020-06-20:
Got the following boards compiling and uploading now:
ATtiny84a (California STEAM)
ATtiny85 (DigiSpark)
ATtiny167 (DigiSpark Pro)
ATtiny841 (Wattuino)
I know that the "California STEAM" board has the correct pin-out .....
But the DigiSpark pro is a disaster:
[image]
I have to create a variant to map the pins to the correct location. The regular Digispark probably has the same issue.

SpenceKonde, 2020-06-20:
Eh, it's kinda dumb- but it's not like it's particularly complicated to change. We should probably straight up include that image on the x7-specific page., and they could have

I can ask @MCUdude to just insert an extra set of pin numbers in the existing diagram so people can see the pin numbers on the digispark next to theirs to assist in porting

I am trying to really push the PIN_Pxn notation (you may notice I added that to all of ATTinyCore - I was horrified to see it wasn't in already. In the documentation for the x7 parts, including that one, I will highlight the fact that there are THREE different pinouts in use, and recommend using the PIN_Pxn notation when writing code for these parts to maximize portability.

SpenceKonde, 2020-06-20:
On the latter front, I'm working on an Arduino Best Practices (standin name) guide for non-standard hardware, which recommends PIN_Pxn (and the hardware packages that support it)- far from ready for public consumption - but when I feel like rambling about how peripherals work, I write stuff there, and my hope is that it will eventually compare and "classic" and "modern" versions of the peripherals as well... 

I had previously started using PIN_xn - but then Arduino official boards started using PIN_An for the pin that that "analog pin" referred to or something. So, PIN_Pxn it is. Unambiguous, and, if that's what you use, if you ever go deeper than the Arduino abstractions, you don't have to keep converting back and forth between pin numbers and port/bit....

SpenceKonde, 2020-06-20:
The pinout chart you posted is also wrong, by the way.... 

Correct one:
https://hamprojects.files.wordpress.com/2018/12/DigisparkProDiagram2.png

SpenceKonde, 2020-06-20:
Does the analog pin mapping work correctly? I don't have any digispark 167s - actually, I'm not sure I have any working, assembled, 167's... haven't touched one of those in eons...

I'm.... almost certain it does not. 

I did some things to the way analog pin numbers were treated by ATTinyCore to try to make it suck slightly less than it did before. 

If i ever got my hands on the guy who decided separate pin numbering schemes for pins with and without an ADC channel, and the names "analog pin" and "digital pin" were a good idea.... 

Well, let's just hope I don't...

SpenceKonde, 2020-06-20:
Your pinout diagram was the beta version - though most of what was in pins_arduino.h looked okay from that perspective. 

I cannot belive that just a few years ago this heinous mess was my main project.... I guess on the bright side, I sure have come a long way.... there are so many things done such stupid ways.... so many missed opportunities.....

freemovers, 2020-06-21:
LOL. I never looked at the core of the TinyCore. I was working on an art project and got stuck with the ATtiny84a (not enough timers), so it was a good reason to learn the ATtiny1616, but there was no core. And oh boy, that was a learning curve after you took off implementing it, but moving forward the MegaTinyCore is the way to go.
I didn't use the pinout diagram for the DigiSpark 167 but looked up the variance file from the Digispark repository on Github. Looked pretty similar to the variance files used on the TinyCore, so I just copied the variables from the file. I didn't know you just changed to PIN_Pxn 30 minutes after I pulled the latest version ... I have a 167 here so I can test the analog pins.

tedder, 2020-06-24:
Fwiw I can host the pinouts on https://pinout.guide/ if desired.

tomaskovacik, 2020-06-24:
so someone again reinventing a wheel ... 

boards.txt file is here with full menu : https://github.com/tomaskovacik/ATTinyCore/blob/DigistumpSupport/avr/boards.txt

[image]

take micronucleus binaries from micronucleus repository, also upgrade hexfiles for upgrading bootlader

https://github.com/micronucleus/micronucleus

freemovers, 2020-06-24:
Thanks Tomas! I have some questions. Where are the micronucleus binaries located? I looked into the releases but that only showed a zip-file with the source code. Looks like the boards.txt is referring to the standard variant pins_arduino.h file, or did you make changes in those as well? The Digispark has a different pin-out. Great idea to have the hexfiles for the bootloader in there as well.

tomaskovacik, 2020-06-24:
Hi, i am not near the comouter for few hours, I put all on my temporary website as example, please look at #210 all is there

SpenceKonde, 2020-06-25:
@freemovers 
Is there a reason you included the Retain EEPROM submenu for the usb boards? When uploading through a bootloader, the EEPROM is always retained (unless the bootloader goes to the trouble of wiping it) - so the only case in which the EESAVE setting will have any effect is when burning bootloader.... (that's why I don't use it for optiboot board defs - otherwise people complain that it doesn't work when uploading through the bootloader :-P)

SpenceKonde, 2020-06-25:
On the 841 usb def - I see that speed is 8MHz; what voltage does it run at? 3.3 or 5? If 5, does serial work on it without manual calibration (indicating that the bootloader cal's it - great!)

SpenceKonde, 2020-06-25:
Only thing left before the merge, I think is to add a doc page describing minimum connections. for those who insist on building their own....

SpenceKonde, 2020-06-25:
is upload.speed supposed to be in the Micronucleus versions? I didn't think that was used for USB uploads....

SpenceKonde, 2020-06-25:
Okay, I removed upload.speed. Fixed An constants for digispark board (ugh!)

I'm still making some modifications here - but I realized something.... uhhhhh.... where are the bootloader files? How do we bootload a board with micronucleus?

tomaskovacik, 2020-06-25:
if it helps loot at my fork

I put them into new dir 

https://github.com/tomaskovacik/ATTinyCore/tree/DigistumpSupport/avr/bootloaders/micronucleus

boards.txt:
attinyx5nucleus.bootloader.tool=micronucleus
attinyx5nucleus.bootloader.file=micronucleus/upgrade-t85_default.hex

for empty chip are here https://github.com/micronucleus/micronucleus/tree/master/firmware/releases

but for upgrade(microcleus is replaced after reboot) are here https://github.com/micronucleus/micronucleus/tree/master/upgrade/releases

SpenceKonde, 2020-06-26:
Okay.... current todo before 1.4.0:

* Do some sanity checking
* Get bootloader binaries in.
* Assuming my understanding is correct, add reset pin menu option to allow disabling reset on Micronucleus boards to get an extra I/O pin (as I understand, this is common)

SpenceKonde, 2020-06-26:
Could someone explain to me the whole "uipgrade" vs "not upgrade" thing?

tomaskovacik, 2020-06-26:
best is to read about it here: https://github.com/micronucleus/micronucleus/tree/master/upgrade

basically: if you have a chip with any sort of bootloader,  after uploading upgrade firmware this will install micronucleus as bootloader on that chip (After restart I think)

standard firmware can be instaled the same way as optiboot only and for blank chip this is the only option

hope this helps

freemovers, 2020-06-27:
@SpenceKonde Thanks for all the work so far. Found an issue in https://github.com/SpenceKonde/ATTinyCore/blob/eb35ea9b9cb277f047490fe2fa7be57366194e06/avr/variants/tinyX7_DigiSpark/pins_arduino.h#L260
TIMER1* should be TIM1*

freemovers, 2020-06-27:
I agree on the EEPROM retain option and I noticed you removed this on all except the ATtiny84a with micronucleus.

SpenceKonde, 2020-07-10:
* Added micronucleus 2.5 bootloader images
* Added support for burning bootloader (both upgrade and fresh install)
* Corrected maximum sketch size (@freemovers - were you aware that the max sketch sizes exceeded what was available for all of the board defs? )
* Lowered maximum data size on micronucleus boards to match what was done in https://github.com/ArminJo/DigistumpArduino
* Added support for slower speeds on t85 (obviously without USB
* Updated docs

Still to do: 
Try installing it via board manager and make sure that works
Throw together InstallingMicronucleus.md, UsingMicronucleus.md
Push out the release and go work on my other projects!

Orel-A, 2020-07-13:
Sir, something is missing from the build...

> Arduino: 1.8.13 (Windows 10), Board: "ATtiny85 (Micronucleus / DigiSpark), Enabled, CPU (CPU frequency), 16 MHz (no USB?), B.O.D. Disabled (saves power), Enabled, Upgrade (via USB), Reset"
> 
> Sketch uses 266 bytes (4%) of program storage space. Maximum is 6586 bytes.
> 
> Global variables use 9 bytes (1%) of dynamic memory, leaving 487 bytes for local variables. Maximum is 496 bytes.
> 
> java.io.IOException: Cannot run program "{runtime.tools.micronucleus.path}/launcher": CreateProcess error=2, The system cannot find the file specified
> 
> 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
> 
> 	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
> 
> 	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
> 
> 	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
> 
> 	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
> 
> 	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
> 
> 	at processing.app.SketchController.upload(SketchController.java:732)
> 
> 	at processing.app.SketchController.exportApplet(SketchController.java:703)
> 
> 	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
> 
> 	at java.lang.Thread.run(Thread.java:748)
> 
> Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
> 
> 	at java.lang.ProcessImpl.create(Native Method)
> 
> 	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
> 
> 	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
> 
> 	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
> 
> 	... 9 more
> 
> An error occurred while uploading the sketch
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

Is it related to micronucleus.exe ? Where to put it ?

SpenceKonde, 2020-07-13:
> at java.lang.ProcessImpl.(ProcessImpl.java:386)

Yes it must be installed via boardmanager to automatically pull that in, I
need to fix a few critical issues and will do board manager release. If you
want micronucleus support without board manager install, you need to have
already installed a core that pulls in the took, I think. I havent tested
uploads through micronucleus either - I've had my hands full fixing all the
other issues that were revealed during the process of getting this set up.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

Orel-A, 2020-07-13:
Okay I solved the problem by installing Digistump's library via board install manager. There was an error in the ide that their post_install.bat isn't trusted so it didn't run. Despite that hitting upload solved the problem for me. So I guess no need for any drivers too. Just need to include micronucleus.exe to ATTinyCore without relying on Digistump being installed.

SpenceKonde, 2020-07-14:
Yeah, that's what the board manager changes above will do, will be in the
1.4.0 release.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-07-25:
I think Micronucleus support Wave 1 is done!

closed 2020-07-25 by SpenceKonde