SpenceKonde/ATTinyCore #418: consider picoboot-lib support

nerdralph, 2020-06-24:
I've just released picoboot-lib, which I believe is the smallest and least intrusive bootloader for tiny AVRs.
https://github.com/nerdralph/picoboot-lib
It uses the last 256 bytes of flash + 2 bytes for calling the bootloader from .init1.  It doesn't modify/patch the reset vector, and doesn't touch the watchdog timer or MCUSR, which is what I mean by "least intrusive" bootloader.

I've already started integrating it into [picoCore](https://github.com/nerdralph/picoCore).  All I had to do was add upload.protocol=arduino and upload.speed=57600 to the boards.txt file.  Next I'll add it as a git submodule under the core libraries.

There's currently 8 unused bytes at the end of the bootloader, so I may be able to add extra functionality.  I was hoping to include eeprom write support (read is supported), but I'll need more than 8 bytes of flash.

SpenceKonde, 2020-06-25:
Wow! that looks AMAZING! Will add picoboot support (and probably steal that mechanism to jump to bootloader in place of virtualboot... that's just beautiful....)... AND it works at a higher baud too?!

How does it figure out that it's already run in order to jump back to the application after running (or uploading new sketch?)

Does picoBoot give a blink pattern? I find that *extremely* useful when working with bootloaded boards, as a way of verifying that the bootloader is there and the bootloader ran; *some* way to provide a function analogous to the triple blink would be highly recommended. 

Without hardware bootloader support, you have way more granularity with size of bootloader.... so you could totally sacrifice another 32b or 64b (depending on part page size) in the name of adding extra features.... with that... what could you get? Could you write EEPROM too? 

By the way, the REAL prize that you want.... is for that RJMP to be located in the first byte after the first page (or first block of 4 pages on 841, 441, 1634 and 828, since they have a "4 page erase" which is about the size of the pages on similar parts, and a page buffer 1/4th that size), so that when the power is weak and the power glitches, resetting the chip when the first page is erased, so the page is erased but nothing got written, it will skid along the 0xFFFF's (which are a noop) until reaches that rjmp, which brings it to the bootloader. That sequence of events DOES happen, and it DOES brick chips. Has happened to me... between 3 and 6 times IIRC.... and I know a guy who is using these in production who had the same problem (he had thousands of them deployed, and they were in the middle of wilderness bolted onto trees communicating via long range wireless... and every year, a few of them would die in this way).... this is worth the number of bytes between end of vector table and the end of first page, like, without even a second thought.

One other aside.... I would suggest some of those last bytes be used to define the bootloader as picoBoot....? And maybe a version? (optiboot uses the last two bytes for that)

nerdralph, 2020-06-25:
> 
> How does it figure out that it's already run in order to jump back to the application after running (or uploading new sketch?)
> 
If it times out waiting for serial communication, it just returns, continuing with the next instruction after .init1.
https://github.com/nerdralph/picoboot-lib/blob/master/pb-lib.S#L73

If it receives STK500 commands, it enters a command processing loop, and for quit it jumps back to the reset vector.
https://github.com/nerdralph/picoboot-lib/blob/master/pb-lib.S#L122

There's actually a bug in the quit that happens with custom crt0 like I use in picoCore.  My pcrt0 doesn't re-initialize the stack, so jumping back to the reset vector when the bootloader was entered by a call will "leak" 2 bytes of stack memory.  I've added two pop calls there, but haven't pushed the code yet.
>
> Does picoBoot give a blink pattern? I find that _extremely_ useful when working with bootloaded boards, as a way of verifying that the bootloader is there and the bootloader ran; _some_ way to provide a function analogous to the triple blink would be highly recommended.
>
Not yet.  I've used a couple different techniques in previous bootloaders.  One is to pull the Tx line low during the bootloader timeout interval, so the Rx LED on the attached USB-TTL adapter will light up.  That's my preferred technique for the mega devices with hardware UARTs.  For picoboot-lib, I'm planning on supporting single-pin operation like I've done with earlier bitbang uarts.
http://nerdralph.blogspot.com/2014/01/avr-half-duplex-software-uart.html
For single-pin operation, I can't hold the Tx line low, since it's also the Rx line.
For picoboot-lib, to indicate the bootloader is waiting, I think I'll dimly light a LED with the internal pullup.   Blinking is a possibility, at the cost of a bit more code.

>
> Without hardware bootloader support, you have way more granularity with size of bootloader.... so you could totally sacrifice another 32b or 64b (depending on part page size) in the name of adding extra features.... with that... what could you get? Could you write EEPROM too?
>

Yes.  It already has EEPROM read support, but I've thought about removing that since the Arduino IDE doesn't use it.  But if I add EEPROM write support, then I'd probably want the EEPROM read to support verify.  I'm also planning to modify picoCore's platform.txt to write the .eeprom section, so you can store constants in EEPROM like:
'const char fonts[] EEMEM = {24, 69};'

>
> By the way, the REAL prize that you want.... is for that RJMP to be located in the first byte after the first page [...]
>
That's a rather complex problem you are describing.  I'd have to understand it better, and reproduce it myself before I would try to implement a solution.

> 
> One other aside.... I would suggest some of those last bytes be used to define the bootloader as picoBoot....? And maybe a version? (optiboot uses the last two bytes for that)
>

I'll think about that.

SpenceKonde, 2020-07-30:
@nerdralph 
When I first saw the picoboot method of entering the bootloader, I got a bad feeling, but I couldn't figure exactly why...

It just hit me - and this is why I couldn't use picoboot in ATTinyCore with that entry mechanism. The problem comes if you ever upload with any other core, or the right core but tell it it has optiboot, or uses an old version of the core that only has optiboot... anything like that, and the board needs ISP programming to revive. This means that this code is far from resiliant - it's resiliant in the hands of a skilled and careful operator who is certain of which bootloaders are on which boards. , but not the average user. 

That said, I am in general incredibly impressed with what you managed in such a tiny amount of flash, and I love the software serial from sketch wizardry. I'm guessing the intent is that you'd put an interrupt on (oh hey, on the '85 RX comes out to AIN1, so you can use the ACO interrupt) and call that from the ISR? (I'm trying to reason out what the cost would be of passing something to it in order to tell it the delay times, but I'm pretty sure )

This also violates a (the?) key expectation of a bootloader, which is that as long as the code does not use SPM, nothing you can upload or fail to upload could ever prevent the bootloader from running at reset and permitting a new upload. This limitation is of course fundamental to the picoboot entry method; I don't consider a bootloader which violates this expectation to be acceptable for general use, only for careful individuals who are ready and prepared to re-bootload. I don't even like how many micronucleus builds rely on MCUSR getting cleared by the app (I've been advocating copying it to GPIOR0 and clearing it, so if user doesn't clear it (hardly anyone does) the bootloader would still be entered only under the conditions specified by the bootloader config...

SpenceKonde, 2020-07-30:
(as an aside, for ATTinyCore, there is also a need for builds that can use a hardware USART and do the reset vector things, plus the 828 which actually has a hardware bootloader with 256b flash option - and in both of those cases the tuning is especially important, because they're consistently 3-5% fast at 5v on default configuration... but super accurate between 2 and 4 and across temp range.... In general, where there's a bootloader that can tune, I would dearly like to make it remember the setting last time it was programmed (micronucleus puts it right before the appstart instruction), so that as long as it was programmed at the voltage, and wasn't running in a freezer or oven, the tuning would always be right...

nerdralph, 2020-07-31:
> @nerdralph
> When I first saw the picoboot method of entering the bootloader, I got a bad feeling, but I couldn't figure exactly why...
>
> It just hit me - and this is why I couldn't use picoboot in [...]

I see bootloaders as being something for development, but not deployment.  The bootloader limits what you can do with the Tx/Rx pins, and doesn't let you have instant-on after power-up due to the bootloader timeout.  So I wanted to have an easy way to remove the bootloader when development is finished.

That said, I recognize that there is a demand for things that are more idiot-proof, surprising as it is too me.  So what I'm thinking of is a separate version of picoboot that isn't as easy to disable/remove.  It'll need more than 256 bytes, but 1 extra 64-byte page should be plenty.

Part of the problem I have writing good bootloaders is I generally don't eat my own dogfood.  My preferred way to program AVRs is ICSP, with debugWire being my second choice.  For an old guy like me, it's hard to know/understand how the noobs want to use MCUs.  And I don't want to cater directly to noobs, as I'm not prepared to do all the handholding.  I don't mind helping people like you and Hans who are technically competent, and have a much higher tolerance for stupidity than I do.  :-)

> That said, I am in general incredibly impressed with what you managed in such a tiny amount of flash, and I love the software serial from sketch wizardry. I'm guessing the intent is that you'd put [...]
I wasn't thinking of that at the time, as I think lots of use cases are satisfied with a blocking read.  I implemented an interrupt-based receive in [picoUART](https://github.com/nerdralph/picoUART), so I could support something similar.  The hardest part I find is understanding how people are using a non-blocking UART.  I implemented it because I had a number of request for it, but nobody showed me the actual in-the-field code that they would be using it with.  If you know of a couple projects I can look at, that would help.

nerdralph, 2020-07-31:
> (as an aside, for ATTinyCore, there is also a need for builds that can use a hardware USART and do the reset vector things, plus the 828 which actually has a hardware bootloader with 256b flash [...]

I don't recall coming across the 828 until you mentioned it.  I plan to implement the autotuning in a version of picobootArduino.  The intended use is a mega88/168/328 running on the internal RC oscillator.  Do you have any stats on how popular the 828 is?  I'm not saying no to supporting it, but I'd be reluctant to add a bunch of conditionally compiled code to support a chip that I currently don't own and is not commonly used.

SpenceKonde, 2020-07-31:
I'm not sure how everyone has been using them - personally, I often use the bootloader on parts with hardware serial. I don't really have a rational reason not to use it on the others (it does seem to work fine), simply because you can leave the programmer (ie, serial adapter) connected at a cost of only 2 pins (which you want for serial port debug anyway). Of course, until the recent tweaks, I wouldn't use it if I wasn't using a crystal, because I normally run at 5v, and the interesting tinies need a crystal at that speed. Of course, I also habitually use crystals, because I almost *never* run something off battery that isn't spending almost all it's time in deep sleep with the clock off. Lots of people seem to want those two extra pins very badly. 

I have to admit that when i finally started playing around with the Digispark boards... uhhhh... the user experience is kinda nice. Just pop in a USB cable... Am *definitely* making T1634 and 861 versions!

Oh, what I was imagining is that the application places interrupt on AIN1 (you can configure it to compare with reference, making into something like a PCINT, only it doesn't eat up a whole port of PCINTs), and then jump to that blocking byte-read, get the byte back and stuff it into the ring buffer - and if more kept coming, you'd still have the  stop bit before the line went low again triggering the interrupt, so you'd have time to do that stuffing of a byte into the ring buffer and leave the ISR. I envisioned replacing tinySoftSerial with it on picoBoot boards. 

The point being so that you could receive data when you don't know when the data would be coming,  eg, a command from a master to do something, or from a slave managing sensors to report that a reading has met some criteria. 

T828 - Mmm, nope, it's not particularly popular. Too many pins, not enough capability behind it, never captured peoples' attention or imagination. Does anyone really need 28 ADC channels, all tied to the not-particularly-inspiring classic AVR ADC? and the unique capabilities it does have aren't super exciting, even if they are unusual (did you notice the asymmetric port with extra strong sink?) I mean, I'm happy to mail you one if you want (or other tinies, depending on what your collection is like... 

The 841/1634 though - those *are* really popular. OSCCAL0 works the same on those three - they're actually significantly easier to calibrate vs the normal ones. There's no split down the middle of OSCCAL, and it's almost linear across the whole range. (The 841's internal oscillator is insane though - you can crank that sucker up to 16MHz on almost all parts - though I did find one where it needed to be up at 254 to get there... (the problem is, you have to slow it back down while writing to EEPROM...)

The last few tinies seem really weird... the 841 got all the peripherals, but memory and pincount is confining, the 828 got pins but not much else, and the 1634 got memory, but not the peripherals other than the USARTs. Change to modern AVR peripherals hurt, but once you get used to them, they are just a joy to work with compared to the older tinies - input capture makes so much more sense... and there aren't wacky one-off features in every goddamned part

nerdralph, 2020-07-31:
I don't find the ICSP connection generally limits the use of the pins that it is connected to.  And when it does, debugWire is ideal.  I also don't use serial debugging that much.  Most of the time I use objdump and debug the code in my head.  Then there are the things that you can't debug in software like signal rise times and voltage thresholds, in which case I pull out my Rigol.

When I develop code, I use a very short build/test cycle.  Often I'll only change a few lines of code before re-flashing to test.  If I had to plug in and unplug a USB cable every time, it would get very tedious.  One of the (many) things on my to-do list is to optimize the usbasp or usbtiny firmware, writing it in 100% assembler.  I should be able to get the time to flash a 4kB image below 1 second.

The ring buffer idea is feasible, and is something I'll probably add to picoUART.  The current version implements a 2-byte FIFO, which is good enough for most non-blocking read requirements.

For tiny AVRs I have the t13, t84, t85, and t88.  For megas I have the m8a, m88p, m168p and m328p.  I'm not that keen on learning the quirks of another AVR.  Adding #ifdefs for various register and bit names (TIMSK/TIFR vs TIMSK0/TIFR0, SPMEN vs SELFPRGEN, etc) makes for ugly and hard to maintain code, so I try to limit what I write to a few popular devices with similar register names.  I recently started 8052 programming (late to the game I know), and love having the same registers for UART and timers between different manufacturers.  I've also got a few STM32 parts on my desk that I haven't done much with, and soon I'll pick up a GD32VF103 to experiment with.  So although I'll continue to develop AVR software, I think the newer AVRs like the 4809 and tinyAVR 0-series are a dead end.  Thanks for the offer to send me some chips though; it is nice of you even though don't plan to take you up on it.

Getting back on topic,  I think it will be slightly easier to use EE_RDY for the virtual reset vector like Optiboot does, rather than keeping it in the last page before the bootloader.  So unless you can think of a problem with the EE_RDY method, that's what I'll implement.

Another question I have is whether you think there is much value in bumping up the serial speed from 57.6 to 115.2.  At 57.6 (8Mhz), the bit-bang serial code has a timing error of <0.1%.  At 115.2, it would be about 0.6%.  With the automatic OSCCAL tuning getting the timing to within 1% of optimal, a total error of < 2% would still be well within the tolerances of just about any USB-TTL adapter.

nerdralph, 2020-08-04:
It took a bit more work than I thought, but the virtual reset vector is done.
https://github.com/nerdralph/picoboot-lib/releases/tag/v0.7

It's still using 57.6kbps, though I'm still considering a change to 115.2kbps for a future performance enhancement.

nerdralph, 2020-08-20:
Any comments?

SpenceKonde, 2020-08-21:
Somehow this thread fell off my radar. as i mentioned in another recent issue, it seems like sometimes email notifications haven't; been reaching me - or maybe I have been getting so much email that they have been getting buried before I see them, which tells you something about how busy I have been. lately.... I sympathize regarding the deplorable situation with registers on the classic AVRs... On the other hand, within the world of classic AVRs, the wierder tinies (mostly the 841 and 1634) are also the gems of the product like, which is rather frustrating..... I have one more significant addition planned for the classic tinies (some micronucleus hardware and software improvements, in addition to fast I/O and some other loose ends), and then I'll just doing core cleanup/streamlining for ATTinyCore I think, and focusing most of my attention on the Dx and tinyAVR 0/1/2-series. (A 2-series is coming, looks like mostly like the 1-series, but loses TCD in exchange for a second USART and TCB v. 2.0. (good trade!). (personally I think the new ones did an admirable job of keeping registers consistent - most of the delays in getting DxCore out the door were related to things other than making the core actually work, eg, getting my whole schedule filled up with small issues that snowballed massively (ex, attinycore micronucleus) and spending like a month on jtag2updi, which still isn't up to par! I'm forever amazed at how much time I can spend on projects, and how much time off I can spend on projects, and not seem to get any closer to feeling like I'm on top of things...  I'm glad I love doing this shit :-P But yeah, like 90% of the code for DxCore is unaltered megaTinyCore and things shared between my core and MegaCoreX. I also keep getting distracted designing new hardware. Did I really need to design a dozen boards in the past two weeks? Of course not... But I did just that.... when DxCore, ATTinyCore, and megaTinyCore all had thing that would be more useful to all involved for me to work on.... 

But yeah, the code for the "modern" AVRs is so much tidier than the classic AVR code; they seem to have been making a concerted effort at consistency, sacrificing it only when the nature of a new feature forces their hand. The new AVR DA and AVR-DB cores are so similar. Oh - did you hear that the modern AVRs also have improved instruction timing? They call it AVRxt, ST, SBI/CBI, ST and PUSH are all single cycle now, and CALL and it's variants lost a cycle. And LDS gained a cycle >.>

I've got some time off though, I'm feeling optimistic about getting to finally implement the fast digital I/O on my cores... especially with DxCore having gone pretty smoothly (other than discovering that my bootloader that I'd been talking up wasn't doing RAMPZ corrrectly. 

I usually limit myself to 57.6KBps on the classic AVRs even using hardware UART + crystal :-P but I tend to be overly cautious. I iterate quickly too - but not as quickly as you it sounds like, heh. If it works reliably at 115200 and can autocalibrate (it does that right, from the initial charachters the computer sends to start upload?), that's pretty damned cool, I say go for it - that is mighty hard to argue with. The main reason I tended to be so cautious with the tinies is that I not only didn't have autocal, I didn't have a way to cal, period. 

Regarding the ring buffer, I was imagining that being implemented at the core level in C, just reaching into the picoboot-lib for the bitbang routines - it seems like one could have a stub of an ISR to call it when the pin goes low (probably via analog comparator interrupt to keep PCINT open), while the print function could just stuff characters into the TX routine - resulting in a much faster/better soft serial port that reused the space the bootloader was already taken. 

The reason I like the idea of putting the vector to application at the very end of the user flash is that it lets you do what micronucleus does, erase it first, write it last, and never require ISP bootloading because of failed upload (reliable enough fuse reset as I/O without owning an HVSP). And what drove that home is realizing how durable micronucleus is, especially considering what a godawful kludge it is, and how often it non-fatally fails! About the only thing that does it in is a botched USB bootloader upgrade, or an app that writes to the flash, which would do in anything without hardware write protect). And, regardless of the wisdom or folly of putting something that doesn't have hardware bootloader support in a location where accessing it via anything other than the serial or USB cable and accompanying bootloader that you built into the design is a big deal, that is, apparenrly, very common practice as far as I can tell (I am aware of this being done IN COMMERCIAL PRODUCTION). When people see something working reliably, they, well, start to rely on it. (and if it isn't reliable, they just don't bother with it at all - at least that's the impression I get, and what I do). 

As an aside, I have come to agree with you regarding plugging/unplugging cables for micronucleus. I leave reset enabled and have the bootloader run on external reset, not just unplug/replug. A lot less maddening. now if only I could get it to reach out and touch them via USB to make them reboot :-P def. possible, but not something I have time to work on for now.

nerdralph, 2020-08-21:
Thanks for the detailed comments.

It's nice to hear that the newer AVR parts have maintained some consistency in IO register naming; better late then ever I guess.  One think looks good with the new AVRs is UPDI for programming.  I like using debugWire on the classic AVRs, but you have to use ICSP for things like fuse modifications.  If I were to try out one of the new AVRs, which would you recommend?  I'd like to avoid QFN, but I'm OK with TSSOP, TQFP, and SOIC packages.

At 115.2kbps, the autocalibration will loose some precision compared to 57.6k, but I still should be able to keep the timing well withing +/- 2%.  With a t85 at 57.6k, I was usually getting the serial timing to +/- 0.5%.  After I make the changes for 115.2, I'll do some testing with my scope to confirm the timing accuracy.

After your original suggestion about the ring buffer, I looked around for small implementation.  After not finding anything good, I wrote my own: https://github.com/nerdralph/RingBuffer
After I do changes for 115.2kbps, I'll update picoboot-lib to support non-blocking Rx using the ring buffer.  I'm thinking 16-bytes is a reasonable default size for the buffer?

You are preaching to the converted about putting the virtual reset vector at the end of the last user page.  I've been using that technique since the first tiny bootloader I wrote over 6 years ago.

If you have a smart hub, you can force a power cycle on the port you have the digispark plugged into using [uhubctl](https://github.com/mvp/uhubctl)
Another thing you could try is a circuit that ANDs UDM and UDP, with an 5ms time constant RC circuit attached to the RESET pin.  A USB bus reset is 10-20ms of UDP and UDM low, so the USB reset will actually reset your AVR.

SpenceKonde, 2020-08-21:
Nice! Will comment more tomorrow but re: modern AVR - assuming you like the tinies, nor big chips, my only hesitation would be 1614 (SOIC-14) vs 3216 (SOIC-20). Both dead easy to solder, and capabilities are virtually identical other than flash and pincount (they also, for reasons I don't entirely understand, both have two full fledged ADCs in them, identical to eachother; 

I did one paid project with 1614's  where we set them both up in freerunning mode, and the nhad the chip reading them as fast as it could, rescaling them with a 1024-entry lookup table, and blasting the value it looked up on SPI out to a dual DAC, 37k samples from each ADC, rescaled and sent out per second.

SpenceKonde, 2022-02-07:
Closing - this is not a viable solution for ATTinyCore because we have more parts than picoboot supports, and support a wide variety of clock speeds as well as hardware, software, or LIN port for the serial port, and for soft serial I'm able to easily get it to use any of the three pin pairs that I support on tinySoftSerial for the x61 (since the x61 can move the analog comparator's negative input to any of three pins - and there is a potential to expand tinySoftSerial in the future with very little effort to also work on INT0 and INT1 if desired, and since that's selected by a tools menu option, it won't interfere with application use of those vectors unless it legitimately is set to use them, in which case the error is correct, expected, and desired behavior. If that were done, one, sometimes two additional options would get added to each board's software serial menu, and each board without a hardware serial port would get an extra 24 binaries or so (12 clock speeds for autoreset (1 sec timeout, app started on POR) and non-autoreset (bootloader runs at POR); optiboot does that straight up no problem. I automatically generate the makefile that describes all 532 binaries used by ATTinyCore 2.0.0 in the same script that creates the 6370 line boards.txt file and the result works. Adding a second serial bootloader for parts that supported picoboot (since it would need to change the contents of menus, since it doesn't support all the dumbass clock speeds that ATTinyCore does, it has to have it's own top-level board description, which would  would add to an already cluttered boards menu (since we can't stop using optiboot for only some clock speeds, the optiboot options would have to stay). Not to mention that adding another bootloader. even for a subset of parts, would add at least another 1k lines to boards.txt, which is already absurdly long (~1/3rd of a MB)

closed 2022-02-07 by SpenceKonde