SpenceKonde/ATTinyCore #419: Add micronucleus support for ATtiny88 @ 16 MHz

SpenceKonde, 2020-06-25:
There's someone selling ATtiny88 boards with VUSB and a 16MHz external CLOCK (yes, this is out of manufacturer spec, but everyone knows that you can get 16MHz out of AVR's that top out at 12 and 20 out of ones that cap out at 16 at 5v and room temperature). They're cheaper than dirt too...

We should support this configuration.

At this point, I think it should be pretty trivial....

prandeamus, 2020-06-26:
Not much to add technically, but I fully support this!

I presume you mean something like the "MH-Tiny ET ATTiny88" dev boards? They are indeed often very cheap to cheap and widely cloned. I tried to get schematics from the original developer in the mistaken belief that it was open source hardware - my error, I think. As far as I can see the USB is wired up in much the same way as a digispark ATTINY85. They supply their own core and ship with micronucelus onboard. In their core, the reset button doesn't appear as a digital IO line as they assume you'd never blow the appropriate fuse, and for the same reason the external clock means that they don't expose the relevant IO line separately. I'm still a relative newbie, but they do seem to be fair value for money at the low end if you want an ATTINY chip with lots of IO lines.

[image]

SpenceKonde, 2020-06-26:
Yeah those are the ones! Am shocked that the schematic isn't available, though like, it ain't rocket science; you and I can figure out most of the connections without even having to bust out the ohm meter for definitive answers.

prandeamus, 2020-06-26:
Sure! I figured out the pin mappings by looking at their source code and messing around with blink application, and I'm no guru. Ditto for discovering which pins connect to VUSB. I note by the way that they are not keeping their repo up to date. It was thrown at github in 2018 and not changed much since, and they are happy to list digistump as the support email in the JSON, which is ... well, let's say it's cheeky. They quote a website address of mh.nodebb.com, which no longer exists ... Having a modern and maintained core to match this hardware and micronucleus would be great.

SpenceKonde, 2020-07-25:
This is now in!

closed 2020-07-25 by SpenceKonde

terrypin999, 2023-05-26:
I’m not a programmer just an experienced end user. Always find GitHub dauntingly complex. Could someone kindly tell me in plain English what steps I must take to add a bootlloader to my Tiny (clone) boards. I have successfully done so for my Nano boards. But from my reading so far I understand I must first add a new library? Where can I locate it please?

prandeamus, 2023-05-26:
You'd be better off starting a new issue/question; this one is closed.

If you have a chip without a bootloader, already installed, then the quick answer is that you'll need to get a programmer and use the "burn bootloader" option from Arduino IDE. Or use the avrdude utility directly. There are various USB-based programmers around and you should be able to get one + cables for a few dollareuropounds. If you have an existing Arduino Uno/Nano/whatever you can actually use that a programmer; in fact it's one of the sample apps shipped with the IDE. 

But don't carry on this conversation on this thread.

SpenceKonde, 2023-05-26:
Yeah open a new thread in discussion please @terrypin999 
I can't slice off pieces of an issue and put them into a discussion, I checked. I can move issues between cores, and I can turn discussions into issues and issues into discussions, but I can't split off part of one into the other.

terrypin999, 2023-05-27:
On Fri, 26 May 2023 12:05:45 -0700, you wrote:

>Yeah open a new thread in discussion please @terrypin999 
>I can't slice off pieces of an issue and put them into a discussion, I checked. I can move issues between cores, and I can turn discussions into issues and issues into discussions, but I can't split off part of one into the other.

Hi Spence,

Thanks. Meanwhile I posted in the Arduino Forum and that discussion is
on-going. I'm about to report on latest frustrating status, but in a
words it's 'unsolved.
https://forum.arduino.cc/t/adding-a-3rd-party-board-to-pc-without-internet/1131392

When I've done that (and a few weekend 'husband chores') I'll do as you
suggest, although, as I think I mentioned, as a non-programmer I do find
GitHub largely obscure. A great source, largely in cut & paste mode,
but...

Given your email address:
SpenceKonde/ATTinyCore
***@***.***>
hope this ends up reaching you, not in GitHub as per the earlier warning
;-)

Terry