SpenceKonde/ATTinyCore #420: Add micronucleus support to everything else that could plausibly use it!

SpenceKonde, 2020-06-25:
Assuming I can make it build, at least :-P
At least a first crack at this should be almost trivial - so why not?

ATtiny861 @ 16.5 PLL
ATtiny828 @ 16 external CLOCK
ATtiny828 @ 12.8 internal (tweaked OSCCAL)
ATtiny1634 @ 16 external xtal
ATtiny1634 @ 12.8 internal (tweaked OSCCAL)

nerdralph, 2020-09-01:
One thing to watch out for is OSCCAL tuning problems when there is more than one device on the bus.  The calibration routine needs empty frames, so data for another device can throw off the tuning.
https://github.com/micronucleus/micronucleus/issues/182

SpenceKonde, 2022-01-26:
tiny828 was dropped from the list because I realized that the part really f'ing sucks, Read between the lines of the datasheet, and it's abundantly clear that something awful happened to the ADC so late in development that a differential ADC had been written about, and all references to it were then hastily scrubbed from the datasheet. The whole bloody point of the chip was the ADC! It looks like it was going to get an 841-like ADC. I suspect that it still has that ADC (If management was willing to sign off on a respin, they'd have done something better than the what they released, I would hope... ) but the differential functionality is afflicted with some horrible errata so serious that thy couldn't keep a straight face while telling customers that it had a differential ADC. One wonders why they released it at all - my  personal theory is that a small number of friendly customers know about the issue and have a workaround that works for their use case and they're the ones buying those awful chips. 

Otherwise, every part with at least 8k of flash has micronucleus support., see v2.0.0-dev branch,....

closed 2022-01-26 by SpenceKonde