SpenceKonde/ATTinyCore #422: ATtiny x7 pinout diagram

SpenceKonde, 2020-06-25:
@MCUdude 
When you have a chance, I'm afraid we need a pair of images for the x7 ATtiny.... I don't quite know why all of a sudden the ATTinyCore stuff is crazy hot, but getting so many people talking to me about it.... we now have the VUSB bootloader too... insanity!

we need a Micronucleus pin mapping image for Digispark Pro. I'm envisioning the markings on the sides of an image of a Digispark Pro, kind of like what is there, only in the style of ATTinyCore, and without the reference to things they implemented differently. That one references features done differently on ATTinyCore, among other things. 

The reset pin (PB7) is digital pin 13, analog pin A13, ADC10 (ugh, I know). 
Not on pin:
PB4, digital 14, XTAL1
PB5, digital 15/A15, XTAL2, ADC8

PWM pins shall be: PA2 (OC0A), 

PB0 (OC1AU), PB1 (OC1BU), PB2 (OC1AV), PB3 (OC1BV) PB4 (OC1AW), PB5 (OC1BW), PB6 (OC1AX), PB7 (OC1BX)
Footnote on diagram: Only one OC1A and one OC1B pin can output PWM at a time.

PB3 is also USB D-
PB6 is also USB D+

The functions of the rest of the pins are per the Pxn port pins on the other diagrams. 

ADDITIONALLY:
The ATtiny x7 pinout image for the normal pinouts has two issues:
An constants should be the ADC channel numbers, ie, PA0~PA7 are A0~A7. PB5~PB7 are A8~A10 - currently, PB0~PB7 are marked A8~A15 
All port B pins should be marked as PWM: 
PB0 (OC1AU)
PB1 (OC1BU)
PB2 (OC1AV)
PB3 (OC1BV)
PB4 (OC1AW)
PB5 (OC1BW)
PB6 (OC1AX)
PB7 (OC1BX)

Footnote on diagram: Only one OC1A and one OC1B pin can output PWM at a time.

Current set of numbers should be marked "NEW". 

We need a second set of pin numbers (marked "OLD"):
Left side, top to bottom:
PA0   0/A0
PA1   1/A1
PA2   12/A2
PA3   3/A3
PA4   11/A4
PA5   13/A5
PA6   10/A6
PA7   14/A7

Right side, top to bottom:
PB0   4 
PB1   5
PB2   6
PB3   7
PB4   8
PB5   9/A8
PB6   2/A9
PB7   15/A10

Damned near as batshit crazy as the digispark pro pinout!

This isn't super urgent, as I'm going to go ahead and release 1.4.0 without it, because I want to get back to the more exciting cores and get these upgrades into the hands of the people.

MCUdude, 2020-06-25:
I'm currently on vacation, so I'm not able to do this for at least a couple of weeks. But about that more exciting core. I did spend quite a while creating those pinouts. It would be nice if you'd at least reply to my emails do I could finish them and make them the way you want them. I'm doing this out of goodwill; don't make me change my mind.

SpenceKonde, 2020-06-26:
I got back to you about that - I basically fell off te planet w/regards to work on ,my projects for a few weeks...

MCUdude, 2020-07-02:
[image]
 So is this the "old" or "new" pinout?

SpenceKonde, 2020-07-03:
That is the "new" pinout. 

The "old" pinout has the pins numbered:
Left side, top to bottom:
PA0 0/A0
PA1 1/A1
PA2 12/A2
PA3 3/A3
PA4 11/A4
PA5 13/A5
PA6 10/A6
PA7 14/A7

Right side, top to bottom:
PB0 4
PB1 5
PB2 6
PB3 7
PB4 8
PB5 9/A8
PB6 2/A9
PB7 15/A10

I'm sure you can see why I am recommending the "new" pinout over that convoluted mess :-P

I changed how the PWM from OC1A/OC1B will work - "PWM on OC1A and OC1B pins can each output only one duty cycle at a time. If analogWrite() is used on more than one pin that share a single channel, the most recent duty cycle will be output on all pins sharing that channel that are currently outputting PWM"

MCUdude, 2020-07-03:
OK, how about this:

[image]

SpenceKonde, 2020-07-03:
Why not do it like with the x4, with a second set of pin numbers, but only one diagram?

[image]

Also, PA2 is pin 12 in the "old" pinout, not pin 2 (what an obnoxious pinout - it was an attempt to make as many peripherals on the same pin numbers as on the '328p)

MCUdude, 2020-07-03:
That's fine, but what do you want me to call the "old" pinout (that's going to have arrows)? "Digispark Pro style pinout" or "Alternative pinout"?

MCUdude, 2020-07-03:
[image]

How about this?

SpenceKonde, 2020-07-03:
Nice! I like it! Thank youuuu! Added!

MCUdude, 2020-07-03:
No, that was just a poor screenshot! Here's the high-res photo. Just crop out the

[image]

SpenceKonde, 2020-07-03:
Now, at some point, we need a version of this: 
[image]

changed to match our normal style, with all the things the other diagrams show for pins, but with their wacky numbering.

SpenceKonde, 2020-07-03:
I wonder what it is about the t167 that makes everyone who looks at it feel inspired to renumber the pins....

MCUdude, 2020-07-03:
The problem is that the Digispark image is so low-res that it would look weird on larger screens. Are you able to find it in a higher resolution?

SpenceKonde, 2020-07-03:
That was the only version of it I was able to find - it looks like they never made a higher res version available :-(

MCUdude, 2020-07-03:
OK, I may be able to make something up. I'll get back to you!

MCUdude, 2020-07-03:
How about this one?

[image]

MCUdude, 2020-07-03:
OK here are both in the same image. Feel free to crop as you like!

[image]

SpenceKonde, 2020-07-04:
Lovely!! Thank you! Page has been updated!

closed 2020-07-04 by SpenceKonde

MCUdude, 2020-07-04:
No problem! But I would assume you'll need the ATtinyX5 pinout with USB D+ and D- as well? And perhaps a Digispark pinout pic?

SpenceKonde, 2020-07-04:
I don't think so? Unless I'm mistaken, the digispark people didnt renumber
the pins on the '85... dont the pin numbers match already?

I consider the 167 to be an exception to the "design-for-chips-not-boards"
approach because the only people using it with vusb are probably going to
be using the digispark pro boards... which are readily available fairly
cheaply... and have a different set of numbers printed on them.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com