SpenceKonde/ATTinyCore #425: fast digital I/O

SpenceKonde, 2020-06-26:
As with the modern AVR cores, "fast" digital I/O operations should be possible here. Not a huge priority, but when I add it to megaTinyCore, it should be trivial to come do the same thing over here too...

MCUdude, 2020-07-03:
I would love to see how you (or @westfw who wrote the fast IO code used in [MegaCoreX](https://github.com/MCUdude/MegaCoreX)) would solve this for classic AVRs. I've been trying to port the code written for MegaCoreX, but I can't get the compiler to figure it all out, and thus the speed isn't that great since it doesn't compile down to a single instruction

sleemanj, 2020-07-03:
https://github.com/sleemanj/ATTinyCore/blob/master/avr/cores/tiny/Arduino.h#L253

SpenceKonde, 2020-07-03:
It should compile down to a single instruction on both classic AVR and modern AVR, sbi(VPORTx.OUT,bit) for modern, sbi(PORTx,bit) for classic.

My plans for fast digital I/O (on all boards)
* digitalRead() will automatically get optimized if pin is compiletime known. (I already ripped out the moronic thing about digitalRead() turning off PWM. "read" should not change the behavior of the thing you're reading. That's why it's called "read" not "write"). 
* fastDigitalWrite() which will *just* write the pin high or low, nevermind PWM
* (ideally) automatically optimize digitalWrite() for compile time known situations such that it turns off PWM (if the pin has PWM) and writes the pin. (should be... in (sreg backup) cli, ld, andi/ori, st, sbi/cbi, out (restore sreg) - on modern avr, there's an extra ld, andi/ori, st to emulate the pullup stuff; on Dx it's just an st if you stage the PINCONFIG) and t841/441/1634/828 (on 1634/828, PUEx can be set with sbi/cbi... but on the 841/441, it can't be, as the PUEx registers aren't accessible to cbi/sbi - they're way the hell up at 0x62 and 0x63! )
*  pinMode() also needs the same treatment as digitalWrite() - I think by the time I'm deep enough that I've done it for the others, this will be a "while I'm here, I might as well" kind of thing...

MCUdude, 2020-07-03:
I tried to implement it in a sketch just to see how it works. My target (An ATmega1284P) is running at 16 MHz, and according to my scope, I can get a maximum toggle speed of 301.859kHz. 

I know that the function won't toggle the pin if `pin` isn't constant, but that's not important right now. Does anyone know why it doesn't get compiled down to a single sbi/cbi instruction?

```c++
const uint8_t fastPin = 0;

static inline void fastWrite(uint8_t , uint8_t ) __attribute__((always_inline, unused));
static inline void fastWrite(uint8_t pin, uint8_t val)
{
  if(__builtin_constant_p(pin))// && __builtin_constant_p(val)) 
  {
    if(val == HIGH)
    {
      ((void)(*((volatile uint8_t *)portOutputRegister(digitalPinToPort(pin))) |= digitalPinToBitMask(pin)));
    }
    else // Can only be LOW here, no need to check
    {
      ((void)(*((volatile uint8_t *)portOutputRegister(digitalPinToPort(pin))) &= ~digitalPinToBitMask(pin)));
    }
  }
}

void setup()
{
  pinMode(fastPin, OUTPUT);
}

void loop()
{
  cli();
  while(1)
  {
    fastWrite(fastPin, HIGH);
    fastWrite(fastPin, LOW);
  }
}

```

sleemanj, 2020-07-03:
Ah, right yes I forgot.

I also redefine `portOutputRegister`, `digitalPinToPort` or `digitalPinToBitMask` in the variant files to avoid pgm memory lookup tables  as the original Arduino.h ones do it.  That is probably your issue.

Example redefinition for Tiny 25/45/85:
https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX5/pins_arduino.h#L247

Looking at it I never got around to doing the 24/44/84

WestfW, 2020-07-04:
Are you compiling with -flto ?
This particularly clean version of the fast functions requires the "link time optimization" so that the sketch area has access to the digital_pin_to_xxxx[] arrays...

MCUdude, 2020-07-04:
> Are you compiling with -flto ?
This particularly clean version of the fast functions requires the "link time optimization" so that the sketch area has access to the digital_pin_to_xxxx[] arrays...

I'll assume this question is for me? I've tried to compile with and without -flto, but the speed is exactly the same ( 301.859 kHz).

WestfW, 2020-07-04:
>> I also redefine portOutputRegister, digitalPinToPort or digitalPinToBitMask

Ah.  This is important.  The normal portOutputRegister() and etc macros have explicit pgm_read_byte() calls that won't be optimized away (they're eventually inline assembler.)

that's why the code at
https://github.com/WestfW/Duino-hacks/blob/master/fastdigitalIO/fastdigitalIO.h#L80
explicitly accesses the digital_pin_to_port_PGM[] arrays instead of using the macros.

(That code should actually work fine on traditional AVRs.  I had originally suggested putting the fast hacks onto Mega0/etc mostly because there has seemed to be resistance to having "fast" functions for the traditional boards.)

nerdralph, 2020-07-13:
I used Bill's trick of evaluating the xxx_PGM arrays at compile time for [ArduinoShrink](https://github.com/nerdralph/ArduinoShrink/blob/master/src/wiring_digital.c)
I don't even bother checking for lto since the Arduino AVR core has been using it since the early 1.8.x versions.

SpenceKonde, 2021-02-15:
Have implemented them in my own branch (nothing compiles because I'm still early enough in the process that I'm breaking at least as many things as I fix, see.., so not pushed into repo yet :-)

SpenceKonde, 2021-02-24:
Note that there is some automatic optimization here that offers significant improvement for all digital IO where it's constant- but not quite the 2-cycle-write that we will get from fast I/O, which we don't want to happen automatically because code clearly relies on how slow they are :-(

SpenceKonde, 2022-02-03:
I've got implementations ready, but I cant figure out how to declare them such that they don't error out. I try to do what I do on megaTinyCore and DxCore, but it reports function body unavailable here. Why? I can have the declaration in Arduino.h and the body in wiring_digital.c on megaTinyCore. Why can't I do the same thing here?