SpenceKonde/ATTinyCore #426: is tools.avrdude.program used?

nerdralph, 2020-06-26:
I noticed it is referenced in your platform.txt, but I don't think the Arduino IDE uses it.
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/platform.txt#L121

I opened the same issue in MicroCore.
https://github.com/MCUdude/MicroCore/issues/108

SpenceKonde, 2020-06-26:
I don't know off the top of my head

SpenceKonde, 2020-06-26:
It's used when you do sketch -> upload using programmer, or when you upload to a board that doesn't have upload.protocol/etc set (ie, for ATTinyCore, any non-bootloader board). 

The way to check this is to insert some distinctive gibberish on those lines, enable verbose uploads, try different ways of uploading to see which ones make it manifest....

closed 2020-06-26 by SpenceKonde

nerdralph, 2020-06-27:
Thanks.  I was looking at the verbose output, but hadn't tried all permutations of upload.protocol.