SpenceKonde/ATTinyCore #427: License

basepr1me, 2020-07-02:
Hello,

I would love to include your ATTinyCore as a port in OpenBSD. It works great and I'm currently working with it and an attiny841 for testing. However, I can't find a license for this repo anywhere. If you can clearly license this software, I can get the port imported.

Would you be willing to do that?

Thanks,

Tracey

SpenceKonde, 2020-07-02:
What is needed for this? 

I am not an IP lawyer and I don't really understand this licensing shit.

basepr1me, 2020-07-02:
HAHAH, no one understands the licensing shit. If you own the software, you can license it how you like. For my own software, I use the ISC license, which is pretty much what is used in BSD land.

https://github.com/basepr1me/thingsd/blob/master/LICENSE.md

There's an example.

sleemanj, 2020-07-03:
Trouble is that there are a lot of contributors to ATTinyCore, and indeed it started as a fork of TCWorld core

https://github.com/SpenceKonde/ATTinyCore/commit/deaf3deb9e872ea6b94070163d1c51e6e6c18c80

You can't really make a unilateral decision on licencing without agreement from the people who have contributed to it.

sleemanj, 2020-07-03:
With that said, we can trace [TCWorld core's history](https://github.com/SpenceKonde/ATTinyCore/commit/38d1d21f7046405476585d54421118f2e383b619) back to http://code.google.com/p/arduino-tiny which itself is LPGL so there is an argument you could make that ATTinyCore is LGPL by extension of it's ancestry

SpenceKonde, 2020-07-03:
If the original core this all was based on was LGPL, wouldn't the terms of the LGPL require that it be released as LGPL?

basepr1me, 2020-07-03:
Yes, since it is a fork and not original code, the original licence should have remained with the code.

SpenceKonde, 2020-07-05:
Can someone scare up an appropriate license file for me to include?

SpenceKonde, 2020-07-10:
Anyone? Hoping to release 1.4.0 tonight, so if you want a license in that version, go scare up a copy of the LGPL for me :-P

sleemanj, 2020-07-10:
https://docs.github.com/en/github/building-a-strong-community/adding-a-license-to-a-repository#including-an-open-source-license-in-your-repository

sleemanj, 2020-07-10:
Actually maybe that doesn't work any more.

Here is the LGPL v3 text, just copy paste into a LICENSE file in the repository root

https://choosealicense.com/licenses/lgpl-3.0/

sleemanj, 2020-07-10:
And here is version 2.1
https://choosealicense.com/licenses/lgpl-2.1/

I don't know enough to say which should be selected.

Orel-A, 2020-07-13:
> The Free Software Foundation may publish revised and/or new versions of the GNU Lesser General Public License from time to time. Such new versions will be similar in spirit to the present version, but may differ in detail to address new problems or concerns.
Version 3 it is...
Need to copy from here: http://www.gnu.org/licenses/lgpl-3.0.html

closed 2020-07-21 by SpenceKonde

SpenceKonde, 2020-07-21:
Thanks for bringing this to my attention. 

Version 2.1 is more appropriate for embedded software - v3 adds additional terms to require that the user be given means to update the software. While honorable in theory, practical considerations can render this impossible, and it is not my intent to prevent the use of this library in commercial products - in fact, I am already aware of a number of such products in production!

commit 91588b29 (2020-07-21): LGPLv3

Fix #427