SpenceKonde/ATTinyCore #428: Definitions for ATtiny2313A

vproch, 2020-07-03:
Hi.
It would be great to add a full support of ATtiny2313A microcontroller. Even if it uses the same definitions as ATtiny4313, it has different device signature (and memory size, of course), so it is not possible to use new registers according to [http://ww1.microchip.com/downloads/en/appnotes/doc8261.pdf](http://ww1.microchip.com/downloads/en/appnotes/doc8261.pdf).
Would it be possible to add both ATtiny4313, ATtiny2313 and ATtiny2313A to the list?

MCUdude, 2020-07-03:
Uh, it does support ATtiny2313 and ATtiny4313?

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x313.md

vproch, 2020-07-03:
Yes, ATtiny2313, but not ATtiny2313A. For example using pin change interrupts on pins PA0..PA2 and PD0..PD6 is not possible because it is not supported by old ATtiny2313 microcontroller.

MCUdude, 2020-07-03:
I will assume you haven't tried to use PCINTs for the ATtiny2313A. As you can see here, ATtinyCore always builds for ATtiny2313A regardless.

https://github.com/SpenceKonde/ATTinyCore/blob/28f557fd594f7b158ed073cd0ee070d4d469a0b9/avr/boards.txt#L2192

vproch, 2020-07-03:
Yes, but the definitions are missing. I have one project which compiles ok for ATtiny4313 but there is a long list of errors if ATtiny2313 is chosen from chips list.
'PCMSK2' was not declared in this scope
'PCIE2' was not declared in this scope
'PCINT17' was not declared in this scope
etc.

MCUdude, 2020-07-03:
Ok. I had a look at the definitions files (iotn2313.h and iotn2313a.h), and it seems like iotn2313.h is included instead of iotn2313.h

Can you turn on verbose compilation in the IDE settings and post the output?

vproch, 2020-07-03:
```
opt/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/vp/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/vp/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/vp/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx313:LTO=enable,chip=2313,clock=8external,eesave=aenable,bod=disable,INITIALIZE_SECONDARY_TIMERS=0,millis=disabled,neopixelport=porta -ide-version=10813 -build-path /tmp/arduino_build_930768 -warnings=none -build-cache /tmp/arduino_cache_325812 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/vp/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/vp/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/vp/Arduino/4digit_timer/4digit_timer_message_eeprom_fast/4digit_timer_message_eeprom_fast.ino
/opt/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/vp/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/vp/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/vp/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx313:LTO=enable,chip=2313,clock=8external,eesave=aenable,bod=disable,INITIALIZE_SECONDARY_TIMERS=0,millis=disabled,neopixelport=porta -ide-version=10813 -build-path /tmp/arduino_build_930768 -warnings=none -build-cache /tmp/arduino_cache_325812 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/vp/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/vp/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/vp/Arduino/4digit_timer/4digit_timer_message_eeprom_fast/4digit_timer_message_eeprom_fast.ino
Using board 'attinyx313' from platform in folder: /home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3
Using core 'tiny' from platform in folder: /home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3
Detecting libraries used...
/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny2313 -DF_CPU=8000000L -DCLOCK_SOURCE=1 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX313 -DARDUINO_ARCH_AVR -DDISABLEMILLIS -DNEOPIXELPORT=PORTA -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/cores/tiny -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/variants/tinyX313 /tmp/arduino_build_930768/sketch/4digit_timer_message_eeprom_fast.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny2313 -DF_CPU=8000000L -DCLOCK_SOURCE=1 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX313 -DARDUINO_ARCH_AVR -DDISABLEMILLIS -DNEOPIXELPORT=PORTA -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/cores/tiny -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/variants/tinyX313 /tmp/arduino_build_930768/sketch/4digit_timer_message_eeprom_fast.ino.cpp -o /tmp/arduino_build_930768/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/opt/arduino-1.8.13/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_930768/preproc/ctags_target_for_gcc_minus_e.cpp
Kompiluji projekt...
/home/vp/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=attiny2313 -DF_CPU=8000000L -DCLOCK_SOURCE=1 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX313 -DARDUINO_ARCH_AVR -DDISABLEMILLIS -DNEOPIXELPORT=PORTA -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/cores/tiny -I/home/vp/.arduino15/packages/ATTinyCore/hardware/avr/1.3.3/variants/tinyX313 /tmp/arduino_build_930768/sketch/4digit_timer_message_eeprom_fast.ino.cpp -o /tmp/arduino_build_930768/sketch/4digit_timer_message_eeprom_fast.ino.cpp.o

```

SpenceKonde, 2020-07-03:
This is already fixed in the github version - github version uses the 2313a headers for 2313 and 2313a.

closed 2020-07-03 by SpenceKonde

vproch, 2020-07-03:
Oh, thank you very much, I have to learn using github more often.

vproch, 2020-07-12:
There is one remaining problem. Compilation works ok now, but when I try to upload a program to  the microcontroller,  it says:
avrdude: AVR Part "attiny2313a" not found.
Valid parts are:
  ...
  t2313    = ATtiny2313
  ...

SpenceKonde, 2020-07-12:
Thanks, fixed.

vproch, 2020-07-12:
Now the ATtiny2313A definitions (PCMSK2, PCINT15, ...) stopped working again :-(

SpenceKonde, 2020-07-12:
Ugh, yeah, I realized that would happen about 30 seconds after I submitted that change, working on it...

SpenceKonde, 2020-07-12:
There, that should fix it!

vproch, 2020-07-12:
Than you again, now it works perfect!