SpenceKonde/ATTinyCore #430: Should  docs (specifically pinout images) even be part of board manager install?

SpenceKonde, 2020-07-20:
I realized that like... most of the download consists of image files, even after dealing with two particularly oversized ones. 

With the image files, core is 4.2MB download 7.25MB extracted.

Without them, it's a 400K download, 2.7MB extracted. 

With larger images for 828, x7, those numbers were 10MB and 14MB,

Considering that in board manager, the files are stashed in a hidden folder and aren't meant to be worked with by hand, does it even make sense to include those images? Something to think about, but not something I will act on in near future.

pcfreak1201, 2020-08-01:
If I remember correctly, I've searched for a nice pinout graphic over Google picture search, with "only big size" - an found your Git-Repository! **_Than_** I realized there are lot of cores and now using them in the ArduinoIDE.
So, when you want to include the images, use smaller GIFs (?) or SVGs (even for automatic generation?) and add a description like "bigger images on Github". I for my own always use PDFs with vector images. They are small, a few k, and could easyly be printed and scaled on the screen.

SpenceKonde, 2020-08-06:
Let me assure you, I will *not* be removing them from the github repo, which is where they get linked to. My question was only regarding what gets automatically downloaded and installed when you use board manager. The main reason being that the images are nigh inaccessible, burried in a subdir of packages in your Arduino15 folder in the hidden appdata folder.. If I did this, I'd yank the images out only when I make the branch that gets "released" for board manager.Main reason being that I realized they're >90% of the size of the download (assuming you already have the right toolchain version, if not, that pulls in a >30mb download from arduino)

SpenceKonde, 2020-10-03:
Pinout images will be removed from board manager releases.

closed 2020-10-03 by SpenceKonde