SpenceKonde/ATTinyCore #433: Optimization: Replace PROGMEM tables for register lookup

SpenceKonde, 2020-07-22:
Currently, the PORTx, PINx, DDRx registers are stored as a table of uint16_t's in PROGMEM for all non "tinymodern". They are all at addresses under 0x20; they could certainly be stored as uint8_t's with a few casts to save space. This is done in tinymodern

Additionally on all parts, the PORTx, PINx, DDRx registers are related to each other by simple arithmetic rules (alas, not the same rules on all parts); Thus, they could be replaced with a short macro, instead of a table lookup. This would also allow greater optimization (even without fastDigitalIO.

digital_pin_to_timer_PGM is full of zeros on most parts, too, digitalPinToTimer macro should be moved to variant files, and replaced with a simple macro to handle the small number of PWM pins they have (except - maybe - on the x7 parts - though there there's another thing we can do there...) - it's sort of silly to have a whole table of values for that!

nerdralph, 2020-08-20:
You can keep the PROGMEM tables, and just access them directly without pgm_read_byte.  That way, gcc's LTO optimizes them away.  That's what I do in [ArduinoShrink](https://github.com/nerdralph/ArduinoShrink)

SpenceKonde, 2020-08-21:
How does that end up working? Ir requires the pins to be compiletime known, right? So if you ever have a situation where you're doing digitalRead() or digitalWrite() of non-compiletime-known value, it would suddenly and abruptly give crazy results without countermeasures like westfw's fast digital I/O does? or am I missing something?

yumkam, 2020-11-23:
Why error out, when you can just `__builtin_constant_p(foo_PGM[P]) ? foo_PGM[pin] : pgm_read_byte(&foo_PGM[P])`
E.g. 5f4a6b231bb9efe43c91cecbbb09ac8b6f988420
(passed limited compile-tested only, no hardware to test atm; should improve lto + constant pin case, no change in generated code for non-constant pin, and no per-board tweaks needed)

SpenceKonde, 2020-12-10:
Woooowwwww, thanks guys. Used this in megaTinyCore, I love it!

closed 2020-12-10 by SpenceKonde