SpenceKonde/ATTinyCore #434: Update ASCII art pinout in avr/variants/tinyX8_MH/pins_arduino.c

SpenceKonde, 2020-07-22:
They don't make this board based on DIP packages - yet that's what's in the ASCII art. 

I don't have the patience for this right now. Need:
* ASCII art pinout of their board, with their lousy numbering
* ASCII art for TQFP-32 pulled in from avr/cores/tinymodern/pins_arduino.c and renumbered for their lousy numbering scheme
* ASCII art for TQFP-32 with the normal t88 numbering for avr/variants/tinyX8/pins_arduino.c

AingeruJM, 2020-08-07:
Is this ok?

```
              PD2 PD0 PC5 PC3
                PD1 PC6 PC4 PC2
              32  30  28  26
                31  29  27  25
             ┌ ─ ─ ─ ─ ─ ─ ─ ─ ┐
    PD3   1  |°                |  24  PC1
    PD4   2  |                 |  23  PC0
    PA2   3  |                 |  22  PA1
    VCC   4  |                 |  21  GND
    GND   5  |                 |  20  PC7
    PA3   6  |                 |  19  PA0
    PB6   7  |                 |  18  AVCC
    PB7   8  |                 |  17  PB5
             └ ─ ─ ─ ─ ─ ─ ─ ─ ┘
               9  11  13  15  
                10  12  14  16
              PD5 PD7 PB1 PB3
                PD6 PB0 PB2 PB4
```

```
              PD2 PD0 PC5 PC3
              32  30  28  26
                PD1 PC6 PC4 PC2
                31  29  27  25
             ┌ ─ ─ ─ ─ ─ ─ ─ ─ ┐
    PD3   1  |°                |  24  PC1
    PD4   2  |                 |  23  PC0
    PA2   3  |                 |  22  PA1
    VCC   4  |                 |  21  GND
    GND   5  |                 |  20  PC7
    PA3   6  |                 |  19  PA0
    PB6   7  |                 |  18  AVCC
    PB7   8  |                 |  17  PB5
             └ ─ ─ ─ ─ ─ ─ ─ ─ ┘
               9  11  13  15  
              PD5 PD7 PB1 PB3
                10  12  14  16
                PD6 PB0 PB2 PB4
```

```
               2 1 0 6 5 4 3 2
               D D D C C C C C
               P P P P P P P P

               2 1 0 9 8 7 6 5
               3 3 3 2 2 2 2 2
             ┌ ─ ─ ─ ─ ─ ─ ─ ─ ┐
    PD3   1  |°                |  24  PC1
    PD4   2  |                 |  23  PC0
    PA2   3  |                 |  22  PA1
    VCC   4  |                 |  21  GND
    GND   5  |                 |  20  PC7
    PA3   6  |                 |  19  PA0
    PB6   7  |                 |  18  AVCC
    PB7   8  |                 |  17  PB5
             └ ─ ─ ─ ─ ─ ─ ─ ─ ┘
               9 0 1 2 3 4 5 6
                 1 1 1 1 1 1 1

               P P P P P P P P
               D D D B B B B B
               5 6 7 0 1 2 3 4
```

SpenceKonde, 2020-08-09:
```
              PD2 PD0 PC5 PC3
                PD1 PC6 PC4 PC2
              32  30  28  26
                31  29  27  25
             ┌ ─ ─ ─ ─ ─ ─ ─ ─ ┐
    PD3   1  |°                |  24  PC1
    PD4   2  |                 |  23  PC0
    PA2   3  |                 |  22  PA1
    VCC   4  |                 |  21  GND
    GND   5  |                 |  20  PC7
    PA3   6  |                 |  19  PA0
    PB6   7  |                 |  18  AVCC
    PB7   8  |                 |  17  PB5
             └ ─ ─ ─ ─ ─ ─ ─ ─ ┘
               9  11  13  15  
                10  12  14  16
              PD5 PD7 PB1 PB3
                PD6 PB0 PB2 PB4
```
Looks like the right base for the TQFP ones. - The ones with pins numbered vertically has aesthetic appeal, but is incredibly hard to read, at least for me. 
Need to replace the physical pin numbers with Arduino pin numbers, and that knocks out the two TQFP action items. I'm not sure I care about the first one anymore.

AingeruJM, 2020-08-10:
I had to fix it a bit, two options:

```
                                 (A5)  (A3)
                                    (A4)  (A2)
                    (D 2) (D 0) (D22) (D20)
                       (D 1) (D27) (D21) (D19)
                    PD2    PD0   PC5   PC3
                        PD1   PC6   PC4   PC2
                     32    30    28    26
                        31    29    27    25
                    ┌ ─  ─  ─  ─  ─  ─  ─  ─ ┐
    (D 3)  PD3   1  |°                       |  24  PC1  (D18/A1)
    (D 4)  PD4   2  |                        |  23  PC0  (D17/A0)
    (D25)  PA2   3  |                        |  22  PA1  (D24/A7)
           VCC   4  |                        |  21  GND
           GND   5  |                        |  20  PC7  (D16   )
    (D26)  PA3   6  |                        |  19  PA0  (D23/A6)
    (D14)  PB6   7  |                        |  18  AVCC
    (D15)  PB7   8  |                        |  17  PB5  (D13   )
                    └ ─  ─  ─  ─  ─  ─  ─  ─ ┘
                      9    11    13    15  
                        10    12    14    16
                     PD5   PD7   PB1   PB3
                       PD6    PB0   PB2   PB4
                    (D 5) (D 7) (D 9) (D11)
                       (D 6) (D 8) (D10) (D12)
```
                    (D 2) (D 0) (D22/ (D20/
                                 /A5)  /A3)
                       (D 1) (D27) (D21/ (D19/
                                    /A4)  /A2)
                    PD2    PD0   PC5   PC3
                        PD1   PC6   PC4   PC2
                     32    30    28    26
                        31    29    27    25
                    ┌ ─  ─  ─  ─  ─  ─  ─  ─ ┐
    (D 3)  PD3   1  |°                       |  24  PC1  (D18/A1)
    (D 4)  PD4   2  |                        |  23  PC0  (D17/A0)
    (D25)  PA2   3  |                        |  22  PA1  (D24/A7)
           VCC   4  |                        |  21  GND
           GND   5  |                        |  20  PC7  (D16   )
    (D26)  PA3   6  |                        |  19  PA0  (D23/A6)
    (D14)  PB6   7  |                        |  18  AVCC
    (D15)  PB7   8  |                        |  17  PB5  (D13   )
                    └ ─  ─  ─  ─  ─  ─  ─  ─ ┘
                      9    11    13    15  
                        10    12    14    16
                     PD5   PD7   PB1   PB3
                       PD6    PB0   PB2   PB4
                    (D 5) (D 7) (D 9) (D11)
                       (D 6) (D 8) (D10) (D12)

```

```

closed 2020-08-10 by SpenceKonde

SpenceKonde, 2020-08-10:
Bam! Thanks! (1.4.1 is out now too)

commit 54f6fbf5 (2020-08-10): Close #434