SpenceKonde/ATTinyCore #435: New pinout images for ATtiny88

SpenceKonde, 2020-07-22:
Just needs second set of digital pin numbers for the MH-Tiny pinout.

Pins 0~13 are the same
PB6 is not numbered (used as CLKI)
PB7 is 14. 
Then PA2, PA3 are 15, 16
PA0, PA1 are 17, 18
PC0~PC5 are 19~24
PC7 is 25
PC6 (reset) is 26
USB on PD1, PD2

The boards are actually pretty neat...

pcfreak1201, 2020-07-31:
[image]
Äh, I'm not sure if I made a mistake on the USB pinout, but I'll check this.
Here are two links:
1. cutted bitmap of the picture above, grabbed on the sellers page (https://www.aliexpress.com/i/32953202750.html):
https://drive.google.com/file/d/1zXLN2fGS2JdM5OJh_QdomA0WIv5q1Zgs/view?usp=sharing
2. Picture of the PCB, cutted, done by me:
https://drive.google.com/file/d/1R6Rtnqiu24vsnLpNykzursJc3aVWpz09/view?usp=sharing
May be it helps someone in designing nice pinout graphics :-)

prandeamus, 2020-08-01:
Nice work! It's a shame that this board has been "orphaned" by the original designer.  (Best I could find is https://mhetlive.nodebb.com/category/12/main-boards-new-mh-et-live-tiny88-16-0mhz). The board has presumably been cloned ad infinitum and is available from multiple suppliers but the official arduino board definition is little more than a patched Digispark. I'm really glad to see ATTinyCore take up the support challenge!

SpenceKonde, 2020-10-03:
Okay, I am going to pull this in for next release.

SpenceKonde, 2021-03-06:
Which I did...

closed 2021-03-06 by SpenceKonde

arduino12, 2022-05-23:
**Hi** @SpenceKonde !

Can you add the cheap and common MH-Tiny pinout to the [ATtiny88.md page](https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x8.md),

Also please tell us how you create the nicely labeled graphics pinouts on that page, so we can improve @pcfreak1201 pinout:
> [image] Äh, I made a mistake on the USB pinout, ignore it! USB is on PD1 (D+), PD2 (D-)...

I got **[10pcs of them for 17$](https://www.aliexpress.com/item/1005002561237243.html)** so I can verify the pinout if needed :)

Also found this top-bottom picture if it helps:
[image]

Can also add a section about achieving low current consumption (0.1uA!) for this board,
I found it by playing with one and I am sure it will help others:
 * Desolder 78L05 **to save 6mA**.
 * Desolder 16MHz clock oscillator (EN pin is shorted underneath to VCC so have to desolder) **to save 2mA**.
 * Desolder or cut green PWR LED 5V trace between regulator and RST PU resistor **to save 3mA**.
 * Desolder or cut USB_D- 1K PU resistor 5V trace between resistor and 5V **to save 1.7mA** (after this you have to program using ISP because the micronucleus USB won't work without this resistor and the external clock oscillator).
 * Disable BOD, DWEN, and set internal RC OSC using [those fuses](http://eleccelerator.com/fusecalc/fusecalc.php?chip=attiny88&LOW=6E&HIGH=DF&EXTENDED=FE&LOCKBIT=FF).
```
#include <avr/sleep.h>

int main()
{
	ADCSRA &= ~(1<<ADEN);					// Disable ADC
	set_sleep_mode(SLEEP_MODE_PWR_DOWN);			// Set deepest sleep mode
	sleep_mode();						// Go to sleep
}
```
[image]

**Thanks and let me know if I can help with the pinout!**

leNnard-808, 2024-01-26:
Can you elaborate a little further on how you optimized the power consumption? In deep sleep I was getting 5mA at 5V, after desoldering the PW LED around 3.5mA, but once I desoldered the regulator I was reading 4.4mA, therefore higher consumption than before. At 3.3V i got around 1.2mA, which makes sense.

I am not getting anywhere close to µA, in fact, I already fried 2 of the boards trying to figure it out :laughing: 

Which pins are you using to power the board and at what voltage? I would really appreciate some help.

NorbertRoller, 2024-09-10:
Hi, I was working on the same subject since a few weeks and came to a similar conclusion like leNnard-808 and arduino12.
As my project is permanently installed in a car (mobile home) with a 12V environment, I needed to have a Voltage converter down to 3.3V.

I choose the BA33BC0FP-E2 from Rohm that has a much lower quiescent current ~10uA than the 78L05 with 5-6mA.

Unfortunately the pinout is different and therefore it can not be soldered directly onto the pads.
I will keep the 16MHz external oscillator and the USB-Port.
This dropped the power consumption in sleep mode from 6.5mA to 1.5mA.