SpenceKonde/ATTinyCore #436: Optimisation on main()

prandeamus, 2020-07-22:
Enhancement suggestion

Since arduino's main() never terminates (ends in infinte calls to loop()
1. You could add __attribute__((OS_Main)) to remove unneeded register save/load
2. Declare it as void rather than int, then you don't need the "return 0" which only exists to make compiler happy

I see that optiboot main() already has the OS_Main set up. Seems like a simple way to save a few flash bytes without getting caught up with massive assembler re-writes. I see there's also an OS_task attribute in the AVR-GCC compiler documentation which does something similar but subtly different, just a little over my head.

SpenceKonde, 2020-07-24:
1. Doesn't work: `warning: 'OS_Main' attribute directive ignored [-Wattributes]`
2. No, main() must return an int - gcc conforms to the C++ standard, even though it makes no sense on AVR :-/. Try changing it... `error: '::main' must return 'int' `

closed 2020-07-24 by SpenceKonde

SpenceKonde, 2020-07-24:
I think optiboot can do it because it is straight C.