SpenceKonde/ATTinyCore #437: millis()/micros() calculations incorrect at certain speeds, some of them relevant

SpenceKonde, 2020-07-26:
At 20, 16.5, and 12 MHz, millis() and micros() are inconsistent with timing derived from the same timebase, ie, the math is wrong. 

At 12.8 MHz, the timing with micros() is correct but millis is  way the hell wrong. 

These issues need to be corrected; particularly the issue with 12.8, as that is flagrantly wrong (millis isn't even close), millis() and micros() are inconsistent, and it is a key speed for future releases.

SpenceKonde, 2020-07-26:
20 and 12 are still wrong, but not too badly. The speeds where F_CPU % 1000000 !=0 are still weird (micros and millis not in sync), but this is documented, and if you want a wacky speed like that, you're probably willing to sacrifice for it.

SpenceKonde, 2020-08-02:
At 12, millis and micros are dead-on sync'ed with eachother, and *very* close to real time. I declare that to be acceptable.
At 12.8, micros is dead on syn'ed with real time, but millis drifts a lot. 
16.5, millis and micro are dead on sync'ed with eachother, but drift rather quickly vs real time
20, millis and micros are dead on sync'ed with eachother, and *very* close to real time. I declare that to be acceptable.

12.8 and 16.5 both must be fixed for 1.5.0.

cburstedde, 2020-12-17:
Would you find it worthy to bring millis() down to exact zero error? I could add code that does it at relatively little expense.

SpenceKonde, 2021-02-08:
@cburstedde has this all been sorted out by your PRs? Trying to figure out where the core stands right now, now that I have 1.3.1 DxCore out.

cburstedde, 2021-02-08:
> @cburstedde has this all been sorted out by your PRs? Trying to figure out where the core stands right now, now that I have 1.3.1 DxCore out.

Oh you've integrated a ton of stuff!

I believe the code is fine.  Without this PR, the story would fall slightly short.

Note that exactness is not usually met, but we are at 30 ppm worst case drift
for arbitrary frequencies (such as 12345678).  Should be roughly 500x more
precise than some of the older approximations.

closed 2021-02-12 by SpenceKonde