SpenceKonde/ATTinyCore #439: No USB detection after flashing MH t88 V1.4.0 bootloader (Micronucleus on t85 still works)

pcfreak1201, 2020-07-27:
Hi!
I'm not sure what to check, but after burning bootloader on my MH-ET t88 there ist no device on USB and upload via USB doesn't work. I flashed the bootloader (also manualy) with USBasp and upload sketches with it still works. 
Console output when flasing bootloader by Arduino IDE 1.8.12 on Linux Mint 19.3 x64:

```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/avrdude.conf"
         User configuration file is "/home/stefan/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny88
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 187500 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9311 (probably t88)
avrdude: erasing chip
avrdude: set SCK frequency to 187500 Hz
avrdude: reading input file "0xfe"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xfe:
avrdude: load data efuse data from input file 0xfe:
avrdude: input file 0xfe contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file "0b11011111"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0b11011111:
avrdude: load data hfuse data from input file 0b11011111:
avrdude: input file 0b11011111 contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xE1"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xE1:
avrdude: load data lfuse data from input file 0xE1:
avrdude: input file 0xE1 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex"
avrdude: writing flash (8172 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 8172 bytes of flash written
avrdude: verifying flash memory against /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex:
avrdude: load data flash data from input file /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex:
avrdude: input file /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex contains 8172 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 8172 bytes of flash verified

avrdude done.  Thank you.
```

SpenceKonde, 2020-07-27:
Thanks, will investigate. IIRC I never actually checked that one... Maybe will have to rebuild the bootloaders sooner than I had planned...

prandeamus, 2020-08-04:
I'm seeing similar symptoms.
- Installed a known & working mtehlive board, connected to breadboard but with no other hardware connected
- Problems with version of micronucleus front end as previously described in #442, resolved by rebuilding latest micronucleus loader from source (details in ticket)
- Worked but after a couple of sessions and a few hours uploading small test sketches, micronucleus would not work. Still, no attached hardware so it was very unlikely that I bricked it with bad voltages etc. It just seemed to be time and repetition of micronucleus bootloading
- Attached ICSP and was able to upload same sketches (without micronucleus). One such test program used v-usb and that works fine, so it doesn't seem to be the USB hardware at fault
- Attempted to reflash the micronucleus bootloader and that runs without errors
- If I power up the board, with mn bootloader, but do not attach it to a USB port, I see the rhythmic three flashes on the onboard LED that suggests it is hunting for USB
- If I connect to the PC, the physical act of plugging in causes the following in dmesg

[87112.219387] usb 1-1.2: new low-speed USB device number 10 using ehci-pci
[87112.299387] usb 1-1.2: device descriptor read/64, error -32
[87112.487385] usb 1-1.2: device descriptor read/64, error -32
[87112.675382] usb 1-1.2: new low-speed USB device number 11 using ehci-pci
[87112.755378] usb 1-1.2: device descriptor read/64, error -32
[87112.943307] usb 1-1.2: device descriptor read/64, error -32
[87113.051679] usb 1-1-port2: attempt power cycle
[87113.655378] usb 1-1.2: new low-speed USB device number 12 using ehci-pci
[87114.071375] usb 1-1.2: device not accepting address 12, error -32
[87114.151376] usb 1-1.2: new low-speed USB device number 13 using ehci-pci
[87114.567369] usb 1-1.2: device not accepting address 13, error -32
[87114.567595] usb 1-1-port2: unable to enumerate USB device

The RED led flashing stops almost immediately and switches to solid red, so it looks as if micronucleus is unable to establish itself as a VUSB port.

LD;DR Hardware seems OK, micronucleus is in flash, but is unable to establish a connection

BURN BOOTLOADER OUTPUT

/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/avrdude.conf -v -pattiny88 -cusbasp -e -Uefuse:w:0xfe:m -Uhfuse:w:0b11011100:m -Ulfuse:w:0xE1:m -Uflash:w:/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/avrdude.conf"
         User configuration file is "/home/robert/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 5.0
         AVR Part                      : ATtiny88
         Chip Erase delay              : 15000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff
           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

[... 48 lines cut ...]
avrdude: load data lfuse data from input file 0xE1:
avrdude: input file 0xE1 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex"
avrdude: writing flash (8172 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 8172 bytes of flash written
[... cut ...]
avrdude: load data flash data from input file /home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex:
avrdude: input file /home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/bootloaders/micronucleus/t88_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex contains 8172 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 8172 bytes of flash verified

avrdude done.  Thank you.

SpenceKonde, 2020-08-09:
Flash the new bootloaders I just pushed ( 9842da8e8d173af340969e54679dbbf94f2c2332 ) with ISP programmer, 

The bootloader configurations were wrong - had D+ and D- swapped....

closed 2020-08-09 by SpenceKonde