SpenceKonde/ATTinyCore #442: Experiences with MH ET Live and Micronucleus bootloader version

prandeamus, 2020-08-02:
Adding this in the hope that it may be useful to others.

I have an MH ET Live Tiny88, about a year old. A simple application using V-USB to emulate a MIDI source and send a few MIDI note messages up to the PC: To be honest the core and the 3rd party libraries are the bulk of the code

Compile with the stock MH ET Live distribution, which is a lightly-hacked DigiSpark board definition, as far as I can see. 4118 bytes flash and 236 bytes of global variables.  With ATTinyCore 1.4.0: 3714 bytes flash and 234 bytes global - so there's a nearly 10% saving, for whatever reason, and well worth having! 

However, was not able to upload via micronucleus

_Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.2_

I wasn't sure exactly what to do here, and wasn't really interested in downgrading the firmware (beside, I hadn't soldered in the ICSP connector to any pins and was feeling lazy. I noted there's an existing issue #439 too.  I'm assuming here that the installed firmware was put in for a reason and didn't think downgrading it was a good idea.

So I built a new version of micronucleus from https://github.com/micronucleus/micronucleus, built a new front-end and copied it over /packages/ATTinyCore/tools/micronucleus. This works for me.

I'm NOT saying this is the perfect solution but it worked for me, with minimal fussing with the existing firmware.

SpenceKonde, 2020-08-02:
Hm, what version of micronucleus does the core report having? I do NOT see this issue....

SpenceKonde, 2020-08-02:
I would very much like to see more details on this, as I definitely was able to program them with the version that this core pulls in...

prandeamus, 2020-08-02:
OK, always happy to help with a mystery...

Firstly I had this problem before the 1.4.0 ATTinyCore release, and as far as I recollect the 1.3.x series didn't support this micronucleus bootloader. I never tried to program this particular device in any other way than with the mhetlive core downloaded from https://raw.githubusercontent.com/MHEtLive/arduino-boards-index/master/package_mhetlive_index.json 

So it's definitely not something that has been recently introduced as a regression in 1.4.x.

I went back and configured Arduino to use the mhetlive board manager (as I said before, looks like a hacked around digistump). The error message is the same 

Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.2

Swapping out the "original" micronucleus file with the "latest" one built from github resolves the problem, as noted earlier.

Doesn't seem to matter what the code base is - same results with a stub for setup() and loop()

So, here's the results compiling the sketch

/home/robert/Downloads/arduino-1.8.12/arduino-builder -dump-prefs -logger=machine -hardware /home/robert/Downloads/arduino-1.8.12/hardware -hardware /home/robert/.arduino15/packages -tools /home/robert/Downloads/arduino-1.8.12/tools-builder -tools /home/robert/Downloads/arduino-1.8.12/hardware/tools/avr -tools /home/robert/.arduino15/packages -built-in-libraries /home/robert/Downloads/arduino-1.8.12/libraries -libraries /home/robert/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx8micr:LTO=enable,sketchclock=16external,pinmapping=mhtiny,bod=4v3,resetpin=reset,burnmode=upgrade,millis=enabled,neopixelport=porta -ide-version=10812 -build-path /tmp/arduino_build_640167 -warnings=all -build-cache /tmp/arduino_cache_178238 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.micronucleus.path=/home/robert/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/home/robert/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4 -prefs=runtime.tools.avrdude.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/robert/Arduino/sketch_aug02a/sketch_aug02a.ino
/home/robert/Downloads/arduino-1.8.12/arduino-builder -compile -logger=machine -hardware /home/robert/Downloads/arduino-1.8.12/hardware -hardware /home/robert/.arduino15/packages -tools /home/robert/Downloads/arduino-1.8.12/tools-builder -tools /home/robert/Downloads/arduino-1.8.12/hardware/tools/avr -tools /home/robert/.arduino15/packages -built-in-libraries /home/robert/Downloads/arduino-1.8.12/libraries -libraries /home/robert/Arduino/libraries -fqbn=ATTinyCore:avr:attinyx8micr:LTO=enable,sketchclock=16external,pinmapping=mhtiny,bod=4v3,resetpin=reset,burnmode=upgrade,millis=enabled,neopixelport=porta -ide-version=10812 -build-path /tmp/arduino_build_640167 -warnings=all -build-cache /tmp/arduino_cache_178238 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.micronucleus.path=/home/robert/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=/home/robert/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4 -prefs=runtime.tools.avrdude.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/robert/Arduino/sketch_aug02a/sketch_aug02a.ino
Using board 'attinyx8micr' from platform in folder: /home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0
Using core 'tiny' from platform in folder: /home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0
Detecting libraries used...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10812 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/cores/tiny -I/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/variants/tinyX8_MH /tmp/arduino_build_640167/sketch/sketch_aug02a.ino.cpp -o /dev/null
Generating function prototypes...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny88 -DF_CPU=16000000L -DCLOCK_SOURCE=18 -DARDUINO=10812 -DARDUINO_AVR_ATTINYX8 -DARDUINO_ARCH_AVR -DHASUSB -DNEOPIXELPORT=PORTA -I/home/robert/.arduino15/packages/ATTinyCore/hardware/avr/1.4.0/cores/tiny -I/home/rob [... cut ...]
	at sun.nio.cs.StreamDecoder.read(StreamDecoder.java:112)
	at java.io.InputStreamReader.read(InputStreamReader.java:168)
	at processing.app.debug.MessageSiphon.run(MessageSiphon.java:81)
	at java.lang.Thread.run(Thread.java:748)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device search timed out	
	
Replacing micronucleus executable with new version, and no other changes, fixes the issue

I'm using 64bit Linux distro

I can't vouch for the bootloader itself on the board. This is what came "from the factory".

prandeamus, 2020-08-02:
Also, the MHETLIVE stuff I can find on github https://github.com/MHEtLive/MHEtLiveArduino has no mention of bootloaders all. https://github.com/MHEtLive/Boards/tree/master/MHetLive/avr/bootloaders doesn't mention micronucleus.

I'm out of ideas. To be fair, the solution does seem to be "get the latest version of micronucleus and it works", so this is something that can be easily worked around.

prandeamus, 2020-08-02:
Update - after trying various combinations and permutations, the board in question has stopped entering the bootloader at all. At least, there is no characteristic flashing on the onboard LED, just the green power LED. Something like this has happened to me before - with the stock MHETLIVE package.

To be honest, it's entirely possible that the boards I bought are defective, with an trashy bootloader perhaps. I will try and see if ICSP still works using my trusty USBAsp...

SpenceKonde, 2020-08-03:
Huh! I thought I *HAD* the latest version!!

SpenceKonde, 2020-08-08:
Wait, you say the MHET core pulls in a version of micronucleus executable that DOES work for uploads? But... but.... their json file references the same file that mine does....

prandeamus, 2020-08-08:
I'll be honest, losing track of this myself I think it might be a good idea to re-install all the Arduino environment from scratch along with ATTinyCore and take stock of what I get. I've also got an unopened MHETLIVE board from the same order which is literally unused and still in the packing, so I'll also do an inspection of what fuse values and flash image it ships with, and how well they work with micronucleus. If there's anything specific you want me to record at each step, let me know.

Just to stress, this isn't a super-critical problem for me, just mildly confusing. I can still program reliably using the ICSP pins.

pcfreak1201, 2020-08-10:
**_Interesting!_**
First: Yes, new Bootloader from #439 works! -- And now, I have ... äh, had (;-) ... "the same problem":
```
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=attiny88 -o /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/sketch/Blink_Tiny88.ino.cpp.o /tmp/arduino_build_916419/core/core.a -L/tmp/arduino_build_916419 -lm
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/Blink_Tiny88.ino.eep
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/Blink_Tiny88.ino.hex
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino_build_916419/Blink_Tiny88.ino.elf
Der Sketch verwendet 704 Bytes (10%) des Programmspeicherplatzes. Das Maximum sind 6780 Bytes.
Globale Variablen verwenden 9 Bytes (1%) des dynamischen Speichers, 503 Bytes für lokale Variablen verbleiben. Das Maximum sind 512 Bytes.
/home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4/launcher -cdigispark --timeout 60 -Uflash:w:/tmp/arduino_build_916419/Blink_Tiny88.ino.hex:i 
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.5
[... more lines ...]
Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.5
[... plug off ...]
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device search timed out
```
I remembered, I'd the same problems between t85 and MH_t88 and already downloaded all parts to recompile micronucleus 2.0a4. So, I've changed MICRONUCLEUS_MAX_MAJOR_VERSION in
**micronucleus_lib.h** from 2 to 3 and replaced the binary micronucleus by my recompiled version:
```
/********************************************************************************
* USB details
********************************************************************************/
#define MICRONUCLEUS_VENDOR_ID   0x16D0
#define MICRONUCLEUS_PRODUCT_ID  0x0753
#define MICRONUCLEUS_USB_TIMEOUT 0xFFFF
#define MICRONUCLEUS_MAX_MAJOR_VERSION 3
[...]
```
_... and, it works!_
```
[...]
Linking everything together...
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=attiny88 -o /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/sketch/Blink_Tiny88.ino.cpp.o /tmp/arduino_build_916419/../arduino_cache_322051/core/core_143379349e39cb4d67038585bd823e17.a -L/tmp/arduino_build_916419 -lm
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/Blink_Tiny88.ino.eep
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_916419/Blink_Tiny88.ino.elf /tmp/arduino_build_916419/Blink_Tiny88.ino.hex
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-size -A /tmp/arduino_build_916419/Blink_Tiny88.ino.elf
Der Sketch verwendet 704 Bytes (10%) des Programmspeicherplatzes. Das Maximum sind 6780 Bytes.
Globale Variablen verwenden 9 Bytes (1%) des dynamischen Speichers, 503 Bytes für lokale Variablen verbleiben. Das Maximum sind 512 Bytes.
/home/stefan/.arduino15/packages/ATTinyCore/tools/micronucleus/2.0a4/launcher -cdigispark --timeout 60 -Uflash:w:/tmp/arduino_build_916419/Blink_Tiny88.ino.hex:i 
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
> Please plug in the device ... 
> Press CTRL+C to terminate the program.
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 2.5
> Device signature: 0x1e9311 
> Available space for user applications: 6780 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 106  page size: 64
> Erase function sleep duration: 742ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!

```

pcfreak1201, 2020-08-10:
_Wait!_
It works only by recompiling? Yes! I've changed MICRONUCLEUS_MAX_MAJOR_VERSION back to 2 and it works too!?
But I can't remember, if I did anything at the sources before (January?). So may be I removed some other restrictions / tests for sub-version. Please check the commandline.zip

But it works ... so also fine for me ;-)

prandeamus, 2020-08-11:
Release 1.4.1 and the improved bootloaders means that bootloaders can be uploaded via ICSP and in that sense things are better. However, the version problems still remain, and I suspect that this is a micronucleus fault. I've tried a variety of the supplied bootloader images from 1.4.1 and they do identify themselves to linux (in dmesg)

New USB Device Found, idVendor=16d0, idProduct=0753, bcdDevice=2.05

That sounds reasonable, and I'm sure these are the hex images provided in 1.4.1 because I wiped all the Arduino content and re-installed from scratch today.

When uploading a sketch, the micronucleus front end reports 

This tool doesn't know how to upload to this new device. Updates may be available.
Device reports version as: 2.5
As pcfreak reported above

So it seems to me that 1.4.1 has fixed the firmware (and given lots of new options for MHETLIVE board, great!) but the linux version of the front-end isn't compatible with it. Maybe Windows micronucleus.exe is fine.

The repository for www.github.com/micronucleus/micronucleus (which I assumed naively was the most recent) has an awareness of firmware up to 2.04  with release 2.04 committed on 15 Jan 2019. The firmware configurations in this repo are limited compared to the ones in you've just dropped in ATTinyCore 1.4.1 - what repo are you getting micronucleus from?

pcfreak1201, 2020-08-11:
Because Spece forked https://github.com/ArminJo/micronucleus-firmware and this ist Version 2.5 - I think, that's the source.
And with commits 23 days ago (today) it's quite actual.

pcfreak1201, 2020-08-11:
And here are the tools: https://github.com/ArminJo/DigistumpArduino/tree/master/tools

SpenceKonde, 2020-08-11:
Well, yrah, but those binaries (other than the tweak to windows one to remove the post_install.bat which arduino won't autorun anymore) are 5 years old...

Does this just mean we need to rebuild the linux binaries and host them somewhere?

prandeamus, 2020-08-11:
That would probably work for me. I mean, if you're using the ArminJo fork I can download and build that locally too. If you are going to package the binaries, I'm not sure if you want to package 32-bit and/or 64-bit. 64 bit is the present and future, 32-bit is the past but someone might want them.... suggest you package the 64 bit binary and maybe indicate where you built it from?

pcfreak1201, 2020-08-11:
Well, I see, the problem under linux is same old as the code/binaries:
Topic: Warning: device with unknown new version of Micronucleus detected.
https://digistump.com/board/index.php?topic=1834.0
Dated: 2015-08-10 (great date, when we today have 2020-08-11 ... 5 years problems not solved :-))
Would be great, to have sources, we could follow the library updates (e.g. libusb). And I think, we should start with 64bit because _it's_ the present.

SpenceKonde, 2020-08-11:
Worth noting that the repo @pcfreak1201 linked doesn't appear to contain the source code? 

Wherever we're building it from (let's try to find the most recently updated one), I should fork it so there's a copy under this github user, and no ambiguity about which version the binaries are built from.

pcfreak1201, 2020-08-11:
Strange! It looks like they loose linux and only work on Windows and Mac?
I found the archive, I used (Dec 2018): DigistumpIDE1.5.8C-linux64.tar.gz
It looks like this source, it came from: https://github.com/digistump/DigistumpArduino/issues/5

SpenceKonde, 2020-12-28:
This is no longer a mystery; it is well understood, #465 (new micronucleus binaries) is the solution.

closed 2020-12-28 by SpenceKonde