SpenceKonde/ATTinyCore #443: Moving TX to another port on Tiny85

git4eb7, 2020-08-03:
Hi Spence,
is it possible to define the TX pin (at Tiny85 sticking to Serial.* ) away from AIN0 to another port e.g. ADC3(PB3/Phys.Pin2) ? 
Why? It blocks parallel use of I2C's SDA.
Best,
Robert

SpenceKonde, 2020-08-06:
There's nothing special about the TX pin - I use AIN0 simply because it is naturally paired with the AIN1 pin for RX (which *is* special - it's one of very few pins that can be used without blocking PCINTs on all pins). That makes the documentation simpler and the pin mapping consistent across parts. I really wish there were a way to make that configuration accessible to the user. The problem is that AFAIK it can't be done without a tools submenu, because it has to be compiletime known, and I'm not confident of how else to make that happen. 

I am loathe to add more of those (I just stripped out some from this core and almost all of them from megaTinyCore) because they are evil - they add a dependence to your code which is not visible in the code (and isn't saved on a per-sketch basis) - that is, a change in the tools submenu selection will result in sketch compiling without error, but not working. This is maddening to debug, and I rightly received many complaints about the number of submenus I had allowed to build up before I saw the error of my ways. 

If you can think of any ways to make this work without a tools submenu, I would *love* to hear them!

Also, ugh, that code is awful - I just took it without modification except to fix the TX pin to AIN0 - and I even do that stupidly (that was EONS AGO that I did that,;I was much worse at programming than I am now. Honestly, about 50% of the code in this core is so bad I deserve to be taken out and shot for leaving it in something that so many people use for so long...). In a near future version, I'll do some initial work on making this suck less... though I still dont see how to change the TX pin gracefully? 

In any case the quick and dirty workaround is to just change ANALOG_COMP_AIN0_BIT in the x5 variant from 0 to 3 (for PB3). I wouldn't put any effort into doing it more gracefully, because future update will do a much better job of this, and may even make it possible to change the pin without quick and dirty crap like that.

liebman, 2020-08-06:
One option would be to allow a compile time option to override it.  While this would not help those using ArduinoIDE it would be of use with platformio and some others.  Something like:
```c
#if !defined(ATTINY_CORE_TX_PIN)
#define ATTINY_CORE_TX_PIN A0
#endif
```

SpenceKonde, 2020-08-06:
I do not add features to cater to other development environments - I have my hands full supporting just the one... you are welcome to submit PRs that do so, though. 

I do with the Arduino IDE had a way to do that though :-(

pcfreak1201, 2020-10-19:
I also would love the possibility to unblock I2C, but it may be could be easier to to switch between RX+TX on the actual pins and TX ONLY on an other pin (may be also predefined, but not blocking other esssential functions)?
Mostly I use TX ONLY only for debugging.

SpenceKonde, 2020-10-19:
It is totally arbitrary which pin it uses for TX. RX can only be on one pin using the trick this implementation uses (basically, you need to use either the analog comparator, which *nobody* in arduino land ever touches on classic AVRs, an INTn interrupt pin, or a PCINT, but in the latter case, the interrupt handler is the only one that can be used for PCINTs on that port (the tiny85, having only one port, is particularly screwed by this, at first glance - except everyone's normal go-to, software serial, in order to support being used on any pin, goes ahead and grabs every PCINT handler the chip has :-P

SpenceKonde, 2020-11-07:
durrrrrrr
I was answering w/regards to how I assumed it was written, instead of how it was actually written....  Actual implementation was trivial, because it DOESN'T need to be compiletime known... they set two properties of the TinySoftSerial object, _txmask and _txunmask to (1<<txbit) and ~(1<<txbit)... there's no SBI/CBI stuff here, nor is there really much reason to, not when interrupts are disabled one way or another anyway

Added Serial.setTxBit(uint8_t txbit)

call it with a number 0~7 - the bit index of pin you want to use for TX. Must be on same PORT as AIN1. PB3 as originally requested would be `Serial.setTxBit(3);` before Serial.begin();

closed 2020-11-08 by SpenceKonde

michalsovcik, 2022-10-12:
Hi @SpenceKonde ,

I am also using this modification with ATtiny85 to separate serial pins from I2C, same as @git4eb7 . However, after including Serial.setTxBit(3); , SDA (PB0) stays high and SCL (PB2) sends only short pulses with 30 Hz frequency. I attach screens and my code. Also PB3 is high all the time.

Could Tx pin modification touch in any way I2C please?

Without mod:
[image]

With mod:
[image]

Code (I2C) scanner:
[scanner.txt](https://github.com/SpenceKonde/ATTinyCore/files/9762469/scanner.txt)