SpenceKonde/ATTinyCore #445: Delay/Micros Timing Off By a Factor of 32 on ATTiny 841 with 16MHz external oscillator

arlarson, 2020-08-08:
Hi I have a ATTiny 841 that I'm programming using ATTiny Core 1.4.0. I'm having timing issues when I use the delay/micros functions in that they seem ~32 times faster than expected. If I put a delay of 1000ms the delay seems to only last ~31.5 ms as measured on my oscilloscope. My ATTiny 841 is using a 16 MHz external oscillator and I've set the 

I'm using an Arduino Uno as an ISP

SpenceKonde, 2020-08-08:
Yes I know there is a serious issue of some sort of the 841 currently, once DxCore 1.0.0 is released this is next on my list. Downgrade to previous version in the meantime. 

Somehow this was missed during pre-release checks of 1.4.0

SpenceKonde, 2020-08-08:
Thank you for figuring out more regarding exactly what that issue is - I had not gotten beyond being aware that blink seemed to be running many times too fast.

arlarson, 2020-08-08:
No problem, I'll try downgrading to an older version of ATTinyCore for now. Thanks for the quick response!

brendanmatkin, 2020-08-08:
I just came to say this same thing! It also happens at 8Mhz internal, 1Mhz internal, both at 3v3 and 5v. Although the times differences are different for each. I can do some further testing if you need @SpenceKonde - I have a scope and several types of 841 boards (both with and without 16Mhz crystals).

SpenceKonde, 2020-08-09:
Fix is in github. I'd accidentally commented out the call to sei() to enable interrupts. Didn't realize it had the wacky effects you all were seeing (lessons learned)

SpenceKonde, 2020-08-09:
Now to try to sort out why the new bootloader bricks MHET t88's

SpenceKonde, 2020-08-09:
6a6453c6c45d61ab6466df1ea126b35a15134a0f was the fix

closed 2020-08-09 by SpenceKonde

brendanmatkin, 2020-08-09:
Awesome! Will test tonight or tomorrow!