SpenceKonde/ATTinyCore #447: Attiny841 delay() off by a double digit factor

JustSomeRandomDIYGuy, 2020-08-09:
Hi,

At first... thanks for this great core. It's allways a pleasure working with it. 

However, today I upgraded the ATTinyCore to the most recent version 1.4.0 and am seeing some severe issues with delay(). The described issue is only valid for this most recent version. If I downgrade to version 1.3.3. everthing is running well.

The boundaries are:
- Using Arduino IDE 1.8.12
- Hardware: One of your own "ATTiny841/84 breakout w/serial header" from tindie equipped with a ATTiny841 and a 16MHz crystal
- the descibed behavior is identical for programming the MCU with or without bootloader.
- running this simple blink sketch to generate a 1Hz blinking LED on Pin2 (counterclockwise numbering):
`
    #define LED_BUILTIN 2

    void setup() {
      pinMode(LED_BUILTIN, OUTPUT);
    }

    void loop() {
      digitalWrite(LED_BUILTIN, HIGH);  
      delay(500);                     
      digitalWrite(LED_BUILTIN, LOW);    
      delay(500);                       
    }
`

If I compile and upload this sketch for a 16MHz external crystal I got blinking rate of about 61Hz. If i do the same with internal 8MHz settings (after buning bootloader), I got a frequency of about 31 Hz. 

During failure search I also tried some other board options like enabling or disabling LTO or the millis()/micros() functions. Interestingly, if I disable the millis()/micros() functions, the LED is perfektly blinking with 1Hz.

I hope somebody can reproduce this behaviour and it is not my hardware. 

Regards from Germany,
Marco

SpenceKonde, 2020-08-09:
Yes, I know exactly the problem you describe, I fixed it late last night
(or early this morning) in the github version. Fix is not yet in release
because have another critical regression to fix.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-08-10:
1.4.1 release is now available through board manager.

closed 2020-08-10 by SpenceKonde

JustSomeRandomDIYGuy, 2020-08-10:
Hi,
that was an impressively quick fix. I upgraded to 1.4.1 and now everything is running fine again.

Thanks!

SpenceKonde, 2020-08-10:
The weekend was crazy.... I launch DxCore late saturday night....
then on sunday, all of a sudden like magic, 1.4.0 which I'd released 2 weeks prior, blows the hell up, and I have 5 people contacting me about problems! I had noticed mid-last-week that something was very wrong with x41 timer, but decided I needed to sort out DxCore first....

SpenceKonde, 2020-08-10:
Maybe that guy who was jokingly telling the person asking on forums about a weird timer problem (#441 ) to be quiet until I released DxCore was actually running a much more successful suppression campaign, and you all had been sitting on these problems until I released DxC