SpenceKonde/ATTinyCore #448: (Documentation gotcha) Micronucleus front-end won't load from snap installation of Arduino IDE but ok from tar install

prandeamus, 2020-08-11:
In my experience, micronucleus front end fails if Arduino IDE is installed from a snap (snap install arduino). It complains that it cannot dynamically link the libusb-0.1 library on which it depends. I suspect this is caused by with the snap sandbox mechanism, although I am not a snap expert in any sense of the word, because it limits access to shared libraries not in the snap container? Anyway, if you uninstall the snap and install Arduino via the 64-bit tarball as per instructions on arduino.cc it works as long as libusb is installed on your system).

Chances are this is a fault in the the snap package, so I'm not suggesting you should try and fix it, but you may want to note this behaviour somewhere in the ATTinyCore documentation as a "gotcha". I know some people hate snaps, but on the other hand they are convenient so this situation may arise.

SpenceKonde, 2020-08-11:
What is a "Snap Package"? Is that the technical term for it, or do you mean Arduino IDE from package managers in general?

prandeamus, 2020-08-11:
https://en.wikipedia.org/wiki/Snap_(package_manager)

That is the technical term for a particular kind of package manager, which is being pushed quite hard in a number of Linux packages, esp. Linux. Like any other software package management system there are pros and cons. What it means for Ubuntu/Debian systems in particular is that you can "sudo apt install arduino" which loads the files system-wide (but in this case the DEB package is quite old), or you can "sudo snap install arduino" which gives you the latest version of the IDE and a certain amount of sandboxing which limits the amount of damage a rogue system can do. You can install it by downloading the tar and running a single script, and that's easiest to be honest in this case. I'm not passing judgment, just noting that I think the sandboxing for the arduino package precludes dynamic loading of libusb. It feels like a corner case in that it's only going to affect devs who use linux, who have a target using micronucleus, and use snap packages. Not a totally obscure corner case but not the majority.

closed 2020-08-13 by SpenceKonde