SpenceKonde/ATTinyCore #449: No SerialCDC-Support as in Digistump on t85

pcfreak1201, 2020-08-11:
While I play around the the t85 and t88_MH, I realized that there is no Serial Emulation when using USB?!
When using the original Digistump Digispark t85 board, I can do this:

```
#include <DigiCDC.h>

void setup() {
  pinMode(1, OUTPUT);
  digitalWrite(1, 1);   // Light on during USB init
  SerialUSB.begin();
  digitalWrite(1, 0);   // Flick light when USB ready
  SerialUSB.delay(100);
  digitalWrite(1, 1);
  SerialUSB.delay(5000); // Pause to allow connect
  SerialUSB.println("Hello, world!");

}

long c = 0;
int heartbeat = 1;

void loop() {
  heartbeat = !heartbeat;  // Toggle light
  digitalWrite(1, heartbeat);
  SerialUSB.println(++c);  // And output count
  SerialUSB.delay(1000);
}
```

With ATTinyCore I get:
```
Using board 'attinyx5micr' from platform in folder: /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1
Using core 'tiny' from platform in folder: /home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1
Detecting libraries used...
/home/stefan/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DCLOCK_SOURCE=6 -DARDUINO=10812 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTB -DHASUSB -DBOOTTUNED165 -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny -I/home/stefan/.arduino15/packages/ATTinyCore/hardware/avr/1.4.1/variants/tinyX5 /tmp/arduino_build_986762/sketch/MicronucleusDigiCDC.ino.cpp -o /dev/null
Alternatives for DigiCDC.h: []
ResolveLibrary(DigiCDC.h)
  -> candidates: []
MicronucleusDigiCDC:1:10: fatal error: DigiCDC.h: No such file or directory
 #include <DigiCDC.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
DigiCDC.h: No such file or directory
```
I think it's an enhancement, but for me, an USB connection _without_ serial debug (and without additional hardware :-) doesn't make sense.

SpenceKonde, 2020-08-11:
Yeah I know - I had to get the core released, was having lots of problems making the libraries work (not specific to my core - they don't work for me with the other core either I gave up and deferred to a future version because people were complaining that things that were backed up behind attinycore 1.4.0 were not getting done. Since they I have gotten the joystick/etc to work - but not CDC. 

What OS are you on? How did you get the DigiCDC to work???

SpenceKonde, 2020-08-11:
This issue is covered by the existing Micronucleus issues and projects - closing it just to keep the list of action items clean. I will still read and reply here.

closed 2020-08-11 by SpenceKonde

pcfreak1201, 2020-08-11:
Year, I see, there are some enhancements on https://github.com/ArminJo/DigistumpArduino/tree/master/digistump-avr/libraries/DigisparkCDC and yes, on the "Digispark" t85 board it yet works. But there are a lot of compiler-warnings.
System is 64bit linux (Mint 19.2) - but may be I have a lot of libusb and other "old" stuff installed. I use mostly try and error to get things solved on my linux :-)

SpenceKonde, 2020-12-10:
@pcfreak1201 Hey - I don't suppose you could give me a .hex file (ideally, upload to a digispark-alike without reset disabled, then dump the entire flash with an ISP programmer, so I can "xerox" it and have a board programmed woth known working CDC serial? (plus link to core, code, and library used). It would be really nice to be abkle to get to where I *know* that if it isn't working, it's the driversl so I could fix those, then know that if my core isn't able to create that, that the problem is definitely with my core. This goes for all the USB functionality exposed by the digistump libraries....

pcfreak1201, 2020-12-10:
Well, I don't know if it's, you are looking for, but yes:
https://github.com/pcfreak1201/micronucleus_commandline/tree/main/CDC_Example