SpenceKonde/ATTinyCore #452: Board manager URL corrupt?

ArminJo, 2020-08-19:
Hi.
with the URL http://drazzy.com/package_drazzy.com_index.json I can see a correct json in Firefox.
But Arduino as well as Sloeber only see 1.3.3 as latest version, 1.4.0 and 1.4.1 are missing. 
In Arduino I see the error: `Could not find boards.txt in C:\Users\Armin\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.3.3. Is it pre-1.5?`
Is there something wrong in the 1.3.3 json?

SpenceKonde, 2020-08-19:
There is no "1.3.3. json" - there's only the one json version which gets updated as the core is updated. I think this is some sort of system issue on your end; is the board manager json url present in preferences and all? I'm wondering if it's even trying to fetch a new version at all, or just using the cached version of it. 

As for the error about missing boards.txt, that implies that your 1.3.3 installation is corrupted or incomplete. 

I have no idea what "Sloeber" is....

ArminJo, 2020-08-19:
You were right, my 1.3.3 file was corrupt. After removing the file ATTinyCore-1.3.3.tar.gz dated of 2020-12-21 from Arduino and Sloeber, everything worked fine.
[Sloeber](http://eclipse.baeyens.it/stable.php?OS=Windows) is an Arduino plugin for Eclipse IDE, I use it heavily.

closed 2020-08-19 by ArminJo