SpenceKonde/ATTinyCore #453: arduino-cli (and maybe boards-mangler?) problem with json file

sej7278, 2020-08-24:
Just tried to update with arduino-cli 0.12 (and 0.11 which worked previously) and got this error:

```
$ arduino-cli core update-index 
Updating index: package_index.json downloaded                                                                                                                                                                                                 
Updating index: package_index.json.sig downloaded                                                                                                                                                                                             
Updating index: package_drazzy.com_index.json downloaded                                                                                                                                                                                      
Error updating index: invalid package index in http://drazzy.com/package_drazzy.com_index.json: invalid patch version separator 'a'
```

i assume it doesn't like `"version": "2.0a4"` in the micronucleus definition or something?

Also a bit iffy to be using a non-https website, why not host the json file on github?

```
$ arduino-cli core list
ID              Installed Latest Name                   
ATTinyCore:avr  1.4.1     1.4.1  ATTinyCore         
```

per1234, 2020-08-24:
>i assume it doesn't like "version": "2.0a4" in the micronucleus definition or something?

I've submitted a PR to fix this: https://github.com/SpenceKonde/ReleaseScripts/pull/34

>Also a bit iffy to be using a non-https website,

See: https://github.com/SpenceKonde/ATTinyCore/issues/394

per1234, 2020-08-24:
>i assume it doesn't like "version": "2.0a4" in the micronucleus definition or something?

I should clarify that it's a different non-semver compliant version that caused the problem. I was suspicious that the `2.0a4` thing could cause an additional issue, but I didn't find any problems with it in my tests so I left it alone.

SpenceKonde, 2020-08-25:
Thanks @per1234 - I have not been doing very well at not pushing out bad versions of board manager json. Thanks for bailing me out yet again. 

Board manager json is updated. 

As I said in #394, I am happy to add SSL if someone will give me simple instructions. Everyone says it's easy now, but when I'm like "Okay, how do I do it?" the answer comes back "just do some research, it's easy" - in other words, stop work on all my other projects in order to study something that I am not particularly interested in, so as to provide no actual benefit to anyone. Nope, I am not going to make that sacrifice. Tell me how to do it and I'll gladly do it. Or install the core manually if you're so terrified of using a file served over normal http.

closed 2020-08-25 by SpenceKonde