SpenceKonde/ATTinyCore #455: Problem while programming attiny861a chip.

rakehsaleem, 2020-09-02:
Hi @SpenceKonde,
I am working on a project for which I used attiny861a chip but I am finding trouble in programming the chip. Previously, I used attiny85 for which I2C and power-down sleep mode works fine but then I required more pins for other functions. So I moved to attiny861a as it has more GPIOs and rest of the specs are the same with attiny85. However, I have no success in programming it as I tried to use your core and libraries and as u mentioned for I2C we can use wire library using your core but no SUCCESS so far. 
And then I tried using simple blink example too yet there is no output.
Any pointers to troubleshoot would be great and also how or what library/core can be used to make I2C work with attiny861a chip. 
**For programming, I am using tiny AVR programmer from this below [link.](https://learn.sparkfun.com/tutorials/tiny-avr-programmer-hookup-guide/all).

Thank you!!
-Rakeh

binitshah, 2020-09-10:
Hi @rakehsaleem, were you able to figure out I2C on the attiny861a? I'm using the same chip and facing the same troubles with the Wire lib.

SpenceKonde, 2020-09-10:
Can anyone give me more information on what the issue with wire is???

rakehsaleem, 2020-09-11:
Hello, so my problem initially was that I was unable to program the attiny861 chip to put into power-down-sleep mode but then I found out that my pin assignment was not right. The pin assignments as datasheets suggest—I was not getting it to work but after looking at the describe pin assignment, I made it to work and interrupts were working fine. 
Then, I have to work on the I2C part and I have to configure an RTC (ds1342) and a digital potentiometer (Max5479), but it's not working neither the RTC alarms are setup nor the potentiometer wipers are setting up with my programmed value. Previously, it was working with same lines of code as changing the chip has nothing to do with slave address, except the procedure of calling.

**With attiny85—working**
```
  TinyWireM.begin();
  TinyWireM.beginTransmission(MAX5479_ADD_U2);
  TinyWireM.send(C_BYTEWA);
  TinyWireM.send(0xff);
  int  someByte = TinyWireM.endTransmission();
  TinyWireM.begin();
  TinyWireM.beginTransmission(MAX5479_ADD_U2);
  TinyWireM.send(C_BYTEWB);
  TinyWireM.send(0x96);
  int  someByte1 = TinyWireM.endTransmission();
```

**With attiny861(a)—Not working until now**
```
  // MAX5479 digiPOT configuration for Strain
  Wire.begin();
  Wire.beginTransmission(MAX5479_ADD_U2);
  Wire.write(C_BYTEWA);
  Wire.write(0xff);                   // (0x80)send (0xff=full, 0x80=half, 0x40=one-fourth, 0x20=one-eighth) resistance value
  int Byte1 = Wire.endTransmission();
  Wire.begin();
  Wire.beginTransmission(MAX5479_ADD_U2);
  Wire.write(C_BYTEWB);
  Wire.write(0x80);                   //(26) send (0xff=full, 0x80=half, 0x40=one-fourth, 0x20=one-eighth) resistance value
  int Byte2 = Wire.endTransmission();
```
Keep in mind the slave address, register address and the next data commands are same.

Also, I have tried the clock to be 1Mhz and 8Mhz but for both, they are not working. The hardware connections had pull-up resistors as well as RTC is tested with coin-cell battery as well.
Any advice would be appreciated because I have already assembled 10 pieces with this new 861(a) chip.

SpenceKonde, 2020-09-11:
Does it work if you use Wire.h instead of TinyWireM on the tiny85?

rakehsaleem, 2020-09-11:
No, tiny85 chip works with TinyWireM library only, therefore I also tested TinyWireM lib with 861 just to make sure.

SpenceKonde, 2020-09-11:
So the Tiny85 works with TinyWireM, but does not work with Wire.h? That would mean that I should be looking in the USI portion of Wire.h for general problems - and not looking at anything specific to TinyWireM. TinyWireM IIRC only worked on a few parts, so the fact that it's not working on the 861 isn't a surprise.

SpenceKonde, 2020-09-11:
Which version of TinyWireM is this, the one my core includes?

SpenceKonde, 2020-09-11:
The one in my repo, or one from somewhere else? Link me to it if from elsewhere.

rakehsaleem, 2020-09-11:
I have not used—In fact, didn't look into your TinyWireM library. Instead, I was using the library from this website of Arduino.
[https://playground.arduino.cc/Code/USIi2c/](url). I have used the I2C master library from here and was using this one for so long until now when I changed the chip.

SpenceKonde, 2020-09-11:
Dear god..... these timings are all fucked up!

SpenceKonde, 2020-09-11:
Actually.... maybe not.... but they are doing multiple things that they aren't supposed to for sure.... 

Note: You *must* use resistors on USI parts - you know that right? Using the builtin pullups is never enough, but on the USI parts it's not even possible as I understand...

rakehsaleem, 2020-09-11:
> Actually.... maybe not.... but they are doing multiple things that they aren't supposed to for sure....
> 
> Note: You must use resistors on USI parts - you know that right? Using the builtin pullups is never enough, but on the USI parts, it's not even possible as I understand...

Yes, I have used the pull-up resistors(4.7k and 10k, ohm) and not relying on builtin-pull ups actually. But still, there is no life on the pins. I have tried to troubleshoot with the oscilloscope and see if I get some signals on the SDA/SCL lines but yet no response onto them as well.

> What are you getting back from the calls to Wire.endTransaction()?
> Also check if you're getting a 1 or 0 back from the Wire.write() calls - they return error codes, so that should help narrow things down...

I assume you mean Wire.endTransmission right? Actually, I have not seen the callback output for either the endTransaction or Wire.write() since it requires serial monitor and for some reason, I am getting access to serial but anyways, ill try again in the morning and see if it helps.

Do you think if I2C is making problem is there any other way maybe UART? Would that work using your core and library? Thanks

SpenceKonde, 2020-09-12:
Can you take the modified version of the library I just added to a spin? (I don't want it in a release until this is verified as at least not making anything worse). Also, the USI is a horrible peripheral

Take a look at the pinout chart - there's a builtin software serial implementation named Serial (it is half duplex, and has all the problems of software serial, except that it only uses the analog comparator interrupt, instead of a whole port worth of PCINTs.

SpenceKonde, 2020-09-12:
Like any software serial implementation, it's not great, but it's good enough to get a few bytes out for debug.

binitshah, 2020-09-13:
Hi thanks for the help. I gave the modified version a spin, but no luck. Trying to get I2C communication between the attiny861 and a Raspi. `i2cdetect` doesn't see anything either, but works fine with an atmega328p.

I'll try between the attiny861 and an uno tomorrow, in case there's something about the Raspi acting up.

rakehsaleem, 2020-09-13:
Thanks, @SpenceKonde  for your help and making the necessary mods, I'm out of my station so it's not possible to test at least until Monday or Tuesday, but I would see if that works with new library.

> Hi thanks for the help. I gave the modified version a spin, but no luck. Trying to get I2C communication between the attiny861 and a Raspi. i2cdetect doesn't see anything either but works fine with an atmega328p.

Happens to me also before but I'll check again this week and hope the spin would get me somewhere close to my results.

SpenceKonde, 2020-09-20:
OH! I just noticed another change that might have fixed things without all of this mess - the definition of the SCL and SDA pins was backwards, and was fixed since the 1.4.1 release...

binitshah, 2020-09-22:
Hey @SpenceKonde, I have been using 1.4.1, but no luck. I ended up turning a C library that did work into an [Arduino library](https://github.com/binitshah/usi_i2c_slave/blob/master/DuckieI2C.h). One thing I noticed was that the [861 defs](https://github.com/SpenceKonde/ATTinyCore/blob/71c754303874c5aa4c06176517e4408477e60572/avr/variants/tinyX61/pins_arduino.h#L51-L62) seem to use portB, while the library I forked off of is using [portA pins](https://github.com/binitshah/usi_i2c_slave/blob/master/i2c_slave_defs.h#L22-L24).

Unfortunately, I'm working with a Raspberry Pi and there's some clock stretching issue between USI and Raspi that leads to extremely faulty I2C transactions. I'm not familiar with the issue, so my best guess is that the Arduino core has too much overhead in the main loop to get the timing right. Anyhow, I'm sticking to C and AVR land.

SpenceKonde, 2020-09-22:
Yeah, 1.4.1 does not have the fixes - they are only in the github version. I can't add them to a release until someone has tested them, as I am concerned that this could cause breakage elsewhere.

commit 49b5a77e (2020-09-12): This *may* fix #455....

As far as I could tell, F_CPU was being accounted for utterly incorrectly, among other things. Also added a very crude setClock() implementation; just switches between something a little slower than standard to something somewhat slower than fast mode :-P
PR #479 (2020-12-28): Remove unusable and duplicated USIWire code
USI I2C implementation is merged into Wire.{cpp,h},
the copy in USIWire.{cpp,h} is never compiled (and cannot be compiled).

Merge changes from
49b5a77efc9b89c80391e3dfe0614faad6233bc4 (This *may* fix #455....)
081d765fc299665687d679dc34f147c84df94488 (Fix #278)
b47d0b012736d5f6c3c19e260245a8ab86202170 (Pull in fix per #208)
and remove confusing duplicates.

P.S. regression risk: note that *if* `Wire.setClock` is called somewhere by sketches/libraries, (merging of) 49b5a77efc9b89c80391e3dfe0614faad6233bc4 will result in notable grow of code size (as  [... cut]
closed 2020-12-28 by SpenceKonde