SpenceKonde/ATTinyCore #456: Reset disable not being disabled using ISP (USBAsp) over Arduino IDE on Digispark Clone (Attiny85)

Rimbaldo, 2020-09-02:
Hi!
I have a Digispark Clone that I want to disable reset to use PB5 as GPIO, on Arduino IDE (1.8.13)

So I set my board as an Attiny85(Micronucleous/Digisapark) on AttinyCore submenu

So I set my USBasp-clone (latest firmware)  as an ISP programmer using Attiinycore (USBasp Attinycore).  I set the Reset pin as GPIO on the dropdown list. I set burn Bootloader Method as Fresh Install (via  ISP). I didn't mess with the other options as clock speed etc.

Then I clicked Burn Bootloader and it works all the way to the end, and  says I fuse was changed and gives a correct output with no errors or warnings

But the reset pin was not disabled at all and I can still re-do the process again (burn bootloader) using ISP and it again recognizes the Attiny and works with no errors and says again 1 fuse changed. And again reset is still enabled. 

Shouldn't it not work anymore as ISP, requering to use a High Voltage Programmer, after disabling reset?

I only managed to disable reset by using this AVRDUDE GUI (AVRDUDESS) technique in this site: 

http://ihackedathing.blogspot.com/2016/11/digispark-reset-disable.html

Then reset was really disabled and my USBAsp doesn't recognize the Attiny anymore after that.

Am I doing something wrong? Did I forget to do something else to be able to disable reset within arduino IDE?

Regards, Rodrigo

SpenceKonde, 2020-09-15:
Yes, what you are doing wrong is using a core whose author didn't finish implementing this option....  

Board manager version is now corrected and it will be in the next board manager release.

SpenceKonde, 2020-09-15:
(edited the above comment - wanted to make sure you saw that bit if you were just going off email. Github version is updated, will be in next board manager version. 

No ETA on next board manager release, so if this is a pressing issue for you, please do a github install.

closed 2020-09-15 by SpenceKonde