SpenceKonde/ATTinyCore #457: Don't use sbi/cbi macros to improve performance and reduce compiled size

MCUdude, 2020-09-08:
As pointed out in the link below, the sbi and cbi macros rarely compile down to a single instruction. If you replace them with "regular" bit shifts, code will be faster and occupy less space.

https://github.com/arduino/ArduinoCore-avr/issues/357

I've fixed this in my own [MCUdude_corefiles repo](https://github.com/MCUdude/MCUdude_corefiles).

SpenceKonde, 2020-09-09:
As it happens, on the tinies, most of them ARE... but whoever wrote that code massively overused it... like, they do it while setting every bit in a register, and 2 bytes and 2 clock cycles per (AVR/AVRe/AVRe+ have 2 clock sbi/cbi - only xmega and AVRxt have single cycle bit-level access - when they could have just set it to the desired value, sometimes doing every bit that way, even when calling it right after reset, and setting most of the bits to their POR values.... 16 bytes and 32 clocks.... instead of 4 bytes and 2 clocks. Or in cases where they know (because of recent reset or having already set them) the values of the other bits, but instead use sbi/cbi. It's like someone told them they were better than assignment - probably for doing reg=reg|1<<x|1<<y; (well, they're 2 bytes smaller assuming reg works with sbi/cbi, and 1 clock slower)... and then they just assumed it was better everywhere they set a register....

SpenceKonde, 2020-10-03:
This was already fixed? 

Anyway, found issue with PLL for timer1 clock source config while looking for this, fixed in 5882079d05d20489d816ecb1cc92bbe2e2aa3699 and saved a couple of words of flash on a few devices. PWM can be improved on x61-series, but have better things to work on ATM. Will be easy fix when I get around to it.

closed 2020-10-03 by SpenceKonde