SpenceKonde/ATTinyCore #462: AttinyCore, FastLED and ColorPalette Error

link0687, 2020-09-27:
Hi,
I am using AttinyCore (using an attiny2313a) and FastLED library. Once I've added the blank sketch with the FastLED library included it will compile fine with no errors. As soon as I add the "ColourPalette" example from FastLED library I get a compiler error. I have attached this below in a text file. If you can help me trouble shoot this or point me in the write direction that would be great. I have labeled in the text file where a compiler error begins by using "Orange Error Writing". [Error COde.txt](https://github.com/SpenceKonde/ATTinyCore/files/5287835/Error.COde.txt) Please excuse the crudeness I am new.
. Thanks!

PS. Thanks for adding Atmel ICE to list of debuggers!!!!!!!

pcfreak1201, 2020-09-27:
Hi link0687,

please always add the code of your sketch, if an error occurs with it.
The Error is:
```
C:\Program Files (x86)\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified 
  static_assert(validpin(), "Invalid pin specified");
```

Looks like a predefined pin isn't avaliable, try to use an other one for this function:
`static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 5; EOrder RGB_ORDER = (EOrder)66] `

SpenceKonde, 2020-09-27:
I can't really understand their code very well - it's sort of an incomprehensible monstrosity - but it looks to me like FastLED doesn't support the ATtiny x313 parts. Notice how https://github.com/FastLED/FastLED/blob/master/platforms/avr/fastpin_avr.h has defines for a bunch of parts, but not the x313. Likely this is because the highly constrained resources of these parts make them a poor fit for controlling addressable LEDs (for example, the 2313 has only 128b of ram - meaning, in practical terms maybe 30 LEDs maximum). It certainly doesn't have enough flash for a bulky sketch like ColourPalette, either - it compiles to 4.3k of flash on an ATtiny84, and you have only 2048 on the '2313. 

This core includes the tinyNeoPixel library which is optimized to reduce the flash overhead so you can control, it duplicates the API of the widely used AdafruitNeoPixel library. See https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/tinyNeoPixel.md

The "exact" copy of the Adafruit API doesn't really fit in the flash of the 3213 (the Simple example compiles... with 8 bytes of flash to spare!), but tinyNeoPixel_Static should be usable for very short strings of lights and very simple code (the simple example compiles to 1.2k with the Static version)

In general, I would recommend using a more modern, higher spec'ed microcontroller for controlling addressable LEDs. 

In fact, I would recommend a newer processor in general, for any and all projects, the '2313 is nearly 2 decades old, and was the cut-price half-flash version of the ancient 4313. It has been a poor choice for any and all projects for around a decade (and for at least that long, higher spec parts with more flash, more ram, and better peripherals have been available for a lower price). There is no reason to use an ATtiny2313 in the year 2020 unless you are reprogramming existing equipment with that processor. 

Ex, the AVR128DA28 is under $3 in DIP-28 package (cheaper in smd packages, even with far more pins, which also have more features), has 128k flash, 16k ram, 3 serial ports, 10 pwm pins, a 12-bit ADC (instead of 10) and even a 10-bit DAC, plus the event system and the rest of the whizz-bang features of the post-2016 AVRs. Oh, and it can run up to 24 MHz across it's whole operating voltage range of 1.8~5.5v without a crystal (actually, you can crank them up to 32 MHz - at least at room temperature)

SpenceKonde, 2020-09-27:
In general, I don't recommend FastLED for ATtiny processors - it suffers from "kitchensinkism", that is, the creators have thrown everything and the kitchen sink in the library, and the result is an inappropriately heavyweight library for these resource constrained devices. The more minimal featureset provided by tinyNeoPixel is a better fit.

If you do use tinyNeoPixel,, be sure to select the correct option for Tools -> tinyNeoPixel port - the selected port must match the pin you're using. See the link I posted above for tinyNeoPixel documentation.

closed 2020-09-27 by SpenceKonde

link0687, 2020-10-01:
> In general, I don't recommend FastLED for ATtiny processors - it suffers from "kitchensinkism", that is, the creators have thrown everything and the kitchen sink in the library, and the result is an inappropriately heavyweight library for these resource constrained devices. The more minimal featureset provided by tinyNeoPixel is a better fit.
> 
> If you do use tinyNeoPixel,, be sure to select the correct option for Tools -> tinyNeoPixel port - the selected port must match the pin you're using. See the link I posted above for tinyNeoPixel documentation.
> 
> Also, somewhat surprisingly, FastLED also still doesn't support the post-2016 AVRs - if you want my guess, the library has become such a monstrosity that expanding it to new families of parts has gotten hard, even for it's own creators....

Thanks for your reply. Would you think any of the ATTiny range could handle it?

SpenceKonde, 2020-10-01:
Any of the 8k or 16k parts (85, 84, 841, 861, 87/167, 1634). All have 512b
of sram except the 1634, which has 1k.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

saddys, 2022-02-28:
> Any of the 8k or 16k parts (85, 84, 841, 861, 87/167, 1634). All have 512b of sram except the 1634, which has 1k.

are you sure? the attiny167 compile fine but with fastled library but the 1634 give the same error on pin mapping