SpenceKonde/ATTinyCore #463: ATTiny85 1MHz PWM error

rkrussell, 2020-10-05:
@SpenceKonde:

Thank you for the great work on the Tiny cores for Arduino.

I have discovered a problem with the ATTiny85 core PWM function in release 1.4.1 that does not appear in prior releases. The problem appears when the 1MHz internal clock is used.

The standard PWM frequency for Arduinos is around 500 Hz or a period of about 2 milliseconds. All of the ATTinyCore releases up to and including 1.4.0 have a PWM frequency of 500 Hz in both the 8 MHz internal and 1 MHz internal clock modes. In release 1.4.1, the PWM function also has a frequency of 500 MHz in the 8 MHz internal clock mode. However, the PWM frequency is about 125 Hz in the 1 MHz internal clock mode. The PWM ratio (on time to period time) seems to be correct, but the frequency is one fourth of what it should be.

I have taken care to burn the bootloader each time I change clock modes, so I know that's not the problem. Also, if it were a bootloader issue, the error factor would be 8 instead of 4.

Please let me know if I can provide any further information.

Thanks again,

           Keith Russell

SpenceKonde, 2020-10-12:
And that's with the *released* 1.4.0 version, not a manually installed one from github? 1.4.0 was in development for a LOOOOONG time, and accumulated a HUGE list of changes, and that's when I went through wiring.c with an axe and removed a ton of kruft (and, uh, the initialization code for the millis timer on the 861 and 841). 1.4.1 was a very small update compared to it, and I didn't think I touched the tiny x5 stuff... I mostly just fixed things I'd broken in 1.4.0... 

Which PWM pins is it? Some pins are on the Timer0, some are on Timer1, 

490 Hz is 8 MHz single slope prescaler of 64. I don't know off hand what the timer settings are for all the timers at the various speeds on ATTinyCore are (I never did out charts for these parts like I did for megaTinyCore - it would be much harder anyway, as the 0/1-series parts are all very similar, while the classic Tinies are.... uh... not.... the '85 is particularly weird (that's an 8-bit high speed async timer1.... 490 Hz at 1MHz system clock is prescaler of 8, but if you're doing that on the millis timer, 

I don't have any records of it, but honestly I could see myself, after I did the deep dive into millis timing on the 0/1-series, going back to ATTinyCore and making sure that Timer0 (the one used for millis) *wasn't* running at 490 Hz when the board was at 1 MHz

Actually - try it with the github version too... As more recently, I *was* messing with that, and fixed some things....

SpenceKonde, 2020-11-08:
the weird part is that it's happening on the pin controlled by timer1 not one controlled by timer 0, so it's not an attempt to reduce time spent in the millis ISR, which Is a rat hole I'm not going to look at at all...

closed 2020-11-09 by SpenceKonde

SpenceKonde, 2020-11-09:
This issue impacted only the pin controlled by timer1, which was being incorrectly set to the /32 prescaler instead of /8. This is now corrected. Note that I also switched pin 1 to be controlled from timer1 now. PWM frequency is also within the 500~1000 Hz range at 4~8 Mhz system clock as well. Note that all bets are off if you use the PLL as timer 1 clock source - if you do that, it is assumed that you will also be taking control of the prescaling as well to get your desired frequency, since otherwise why the heck are you clocking timer1 from the PLL.

commit 9e65ca11 (2020-11-09): PWM frequencies

Fixes #463 - for some reason I selected the /32 prescale instead of /8 at low clock speeds (though the #defines just after indicate I had intended to select /8)
Switched pin 1 to timer1 on ATtiny x5
Also now on the parts that can do it, betweeen 3 and 8 MHz we prescale the PWM timers by 32. this means pins 1 and 4 on the t85, all pwm pins 861, and the lonely timer 0 pwm pin on the x7, and get the PWM frequency back towards the 500 Hz ~ 1000 Hz that attinycore tries to do.