SpenceKonde/ATTinyCore #464: Documentation suggestion about I2C pullups

technoblogy, 2020-10-11:
I've just noticed that the USI documentation says:

> Pull-ups on the SDA and SCL port pin are disabled in Two-wire mode.

This indicates that, unlike on the ATmega328P, you can't use internal pullups instead of external pullups when using USI-based I2C on the ATtiny84, ATtiny85 etc.

This might be worth mentioning in the section:

https://github.com/SpenceKonde/ATTinyCore#i2c-support

SpenceKonde, 2020-10-11:
Thanks, I'll try to call that out more clearly. That said - it's not reliable on the ATmega328p either... A 30k internal pullup doesn't come within a mile of the I2C standard (though unless you're working with bare parts, almost any I2C device on a breakout board has it's own pair of 10k pullups...)

technoblogy, 2020-10-11:
Great, thanks. I'm building a project based on an ATtiny84 driving a Chinese I2C OLED display that doesn't have pullups on board, and I've had to redesign my PCB because I foolishly assumed I could use internal pullups, like on the ATmega328P :(

SpenceKonde, 2020-10-11:
heh... euhm.... I would take the opportunity to change the processor too >.> 

Maybe a nice ATtiny1614, if you like the SOIC-14 package? Supported by my megaTinyCore, and overall a much nicer part, and it's got a real I2C peripheral (a better one than the classic AVRs had) instead of that awful USI

technoblogy, 2020-10-11:
Good suggestion. I'll perhaps consider that for a Version 2, as I'm too far down the road now to change for this version!

closed 2020-11-08 by SpenceKonde