SpenceKonde/ATTinyCore #466: 上傳草稿碼時發生錯誤

kely543, 2020-10-13:
您好Spence，
Arduino:1.8.13 (Windows 10), 開發板:"ATtiny45/85 (Optiboot), Enabled, CPU (CPU frequency), ATtiny85, 8 MHz (internal), B.O.D. Disabled (saves power), Enabled"

草稿碼使用了 3360 bytes (44%) 的程式儲存空間。上限為 7616 bytes。

全域變數使用了 247 bytes (48%) 的動態記憶體，剩餘 265 bytes 給區域變數。上限為 512 bytes 。

上傳草稿碼時發生錯誤

avrdude: verification error, first mismatch at byte 0x0044

         0xf0 != 0xf8

avrdude: verification error; content mismatch

CODE:
#include <SoftwareSerial.h>
#include <Wire.h>
#define RxD 1
#define TxD 0
#define DEBUG_ENABLED 1
SoftwareSerial BTSerial(RxD, TxD);

int dt1 = 500;
int dt2 = 1000;
int upin = 3;
int dpin = 4;

void setup()
{
  pinMode(RxD, INPUT);
  pinMode(TxD, OUTPUT);
  Serial.begin(9600);
  BTSerial.begin(9600); // HC-05 default speed in AT command more
  pinMode(upin,OUTPUT);
  pinMode(dpin,OUTPUT);
  digitalWrite(upin,LOW);
  digitalWrite(dpin,LOW);
}
void loop()
{
  byte cmmd[20];
  int insize;
  while(1){
    if ((insize=(BTSerial.available()))>0){
       Serial.print("input size = "); 
       Serial.println(insize);
       for (int i=0; i<insize; i++){
         Serial.print(cmmd[i]=char(BTSerial.read()));
         Serial.print("\n"); 
       }
    }  
    switch (cmmd[0]) {
      case 97: //"a"     
          digitalWrite(upin,HIGH);
          delay(dt1);
          digitalWrite(upin,LOW);
          cmmd[0]= "";
          delay(dt2);
          break;  
      case 98://"b"
          digitalWrite(dpin,HIGH);
          delay(dt1);
          digitalWrite(dpin,LOW);
          cmmd[0]= "";
          delay(dt2);
          break;
      default:
          cmmd[0]= "";
    } //Switch
      
  } //while 
}

SpenceKonde, 2020-10-24:
Ugh, I don't really know how to approach this, is this a new problem (ie, after an update?)

You don't happen to have anything connected to the pins you're using to upload do you? >.>

SpenceKonde, 2020-12-28:
No updated information, closing.

closed 2020-12-28 by SpenceKonde