SpenceKonde/ATTinyCore #467: Attiny 24/44/88 (No Bootloader) Arduino Upload Error FIX

ajaybnl, 2020-10-14:
Hi, I have installed This Arduino core (1.4.1) to my Arduino 1.8.13

I Selected Attiny 24/44/88 (No Bootloader) and pressed Upload

Arduino Gave me errors :  invalid baud rate specified '{upload.speed}'

I fixed the Boards.txt by adding these lines : 

`attinyx4.name=ATtiny24/44/84(a) (No bootloader)
attinyx4.upload.speed=19200
attinyx4.upload.maximum_size=8192
attinyx4.upload.maximum_data_size=512
attinyx4.upload.tool=avrdude
attinyx4.upload.protocol=arduino`

And Its not working.

Please fix the core files.

SpenceKonde, 2020-10-14:
First, undo those changes you made to boards.txt - that will guarantee that it will not work, because those are for use with optiboot, but you want to use it with no bootloader... And you certainly haven't installed a bootloader

What programmer did you have selected? I suspect this is a problem with one of the many programmer options. 
For that matter, are you sure the tools -> programmer option matches the programmer you are using? What physical programmer are you using?

ajaybnl, 2020-10-14:
I have selected USBASP (AttinyCore)

sleemanj, 2020-10-14:
You should try to use "Upload Using Programmer" in the Sketch menu of the IDE and see if that works.

ajaybnl, 2020-10-14:
For UsbAsp i always use Program with programmer

ajaybnl, 2020-10-15:
Sorry i wanted to write that with the lines added, it is Working.

SpenceKonde, 2020-10-22:
It looks to me like - somehow - it is using tools.avrdude.upload instead of tools.avrdude.program recipe for me. I would dearly like to see logs of an (attempt to) upload with and without those changes.

SpenceKonde, 2020-10-26:
Closing as bug does not reproduce. There was something very strange about the original situation that led to this issue.

closed 2020-10-26 by SpenceKonde