SpenceKonde/ATTinyCore #470: attiny84a / PWM duty cycle issue

captainmac40, 2020-10-24:
Hi, after last updates (core and ide), my program on ATTINY84 doesn't work has expected :
I use an pot on analog pin (A2), setting up by pinMode with A2 not 2 and using map() to generate pwm output on analog pin 8.
Before update everything is ok but after when my pot is at 80% i give 25% of dutyCycle ...

ATTINY84a @ 1MHZ internal

An link to update of ATTINYcore ou IDE ?

Using 1.8.13 ide and last ATTINYcore.

Thanks !

SpenceKonde, 2020-10-24:
Please post code that reproduces the issue, and let me know which version you were using previously with different behavior.

I don't suppose you accidentally reconnrected the pot backwards?

captainmac40, 2020-10-24:
It's a long code so I put only the faulty things : 

`void setup()
{
  analogReference( DEFAULT );
  
  // PinMode déclarations
  pinMode ( POT_IN, INPUT );
  pinMode ( MOS_OUT, OUTPUT );
}
void loop()
{ 
int pwm = map (analogRead ( POT_IN ),0,1023,0,255 ));
analogWrite( MOS_OUT, pwm);
}`

and Yes i double check the pot wiring.

About the version.... That's the problem... I don't remember precisely which is correct version... I suppose is 1.8.4 IDE but i'm not sure. I will make tests.

Thanks !

Kind regards

captainmac40, 2020-10-24:
I test with 1.8.4 IDE and 1.3.3 library and it's working as expected. Maybe an core issue between library and IDE with recent versions ?

SpenceKonde, 2020-11-09:
Dollars to donuts it's nothing to do with the IDE, and I broke something in 1.4.0, haven't had a chance to try to reproduce yet.

Sebastian-MrBeam, 2020-12-01:
I see same behavior on attiny88   @ 8 MHz  on 1.3.3 all working fine  PWM Frequency ~250 Hz   PWM on 0-4 ms 
                                                                                                                                                   (val 0 - 255 -> 0 - 100%)
                                                                     on 1.4.0 and 1.4.1          PWM Frequency   ~62 Hz    PWM on 0-4 ms 
                                                                                                                                                    (val 0 - 255 ->0 - 25%) 
looks like Scaling is broken ( factor 4 ) for the PWM Frequency. I used analogWrite(9,val) for testing.

MartinLongUK, 2020-12-01:
I'm getting unusual PWM output on OC1A (Pin D6) when stepping from analogWrite(254) to analogWrite(255), expected to be near indiscernible but observing a large change.

Read-back of the registers shows that the wave generation mode is set to 10-bit (top = 0x03FF), but I think the counter is topping at 8 bits (0xFF).  If I change WGM to 8-bit it behaves as expected.  

Haven't been able to check how OC1B behaves, something else connected to it.  OC0B works as expected.
IDE 1.8.13, ATTinyCore 1.4.1

Sebastian-MrBeam, 2020-12-01:
Make sense - Periode counter runs to 1024,  trigger point is max. set to 255   ->Periode is 4 times longer than expected (62 Hz instead of 250 Hz) and max. possible on time is 25% . Many Thanks for this Information.
Test with "TCCR1A = (TCCR1A & 0b11111100) + 1; //WGM11=0 WGM10=1 8Bit counter" works as expected.
IDE 1.8.13, ATTinyCore 1.4.1  @attiny88  8 Mhz

SpenceKonde, 2020-12-02:
Thanks for the additional debugging help - this issue was easy to fix once I knew where it was (I have no idea when I would have had a chance to do investigation with actual hardware on these, considering the volume of issues on megaTinyCore and DxCore, which being new platforms where all the exciting development is ongoing, are the priority at this point.

One really wonders how it was unreported for so long when it has been broken since the release in august...

closed 2020-12-02 by SpenceKonde