SpenceKonde/ATTinyCore #472: ATTiny85 Simple serial code not working

Tinynja, 2020-10-31:
Hello, I would like to understand why this seamingly simple code doesn't work (the LED on PB4 doesn't light up when I send a message). I have spent many hours now trying to understand what is happening and I couldn't for the life of me get what piece of code I'm missing. Please help me!

    void setup()
    {
      OSCCAL = 0b1011101;
      Serial.begin(9600);
      DDRB |= 1 << 4;
    }
    
    void loop()
    {
      if (Serial.available())
        PORTB |= 1 << 4;
    }

Notes:

- Running on ATTiny85
- Using FTDI232R as USB/Serial converter
- Arduino IDE 1.8.9
- Code uploaded using Arduino as ISP
- Connections/wiring milluple checked with various other working examples
- Build options:
[image]

Let me know if you need any additional information.

pcfreak1201, 2020-11-01:
And the Blink-Sketch with PIN_PB4 works?

Tinynja, 2020-11-01:
> And the Blink-Sketch with PIN_PB4 works?

Yes, the following sketch works without issue:

    void setup() {
      OSCCAL = 0b1011101;
      DDRB |= 1 << 4;
    }
    
    void loop() {
      PORTB = 1 << 4;
      delay(500);
      PORTB = 0 << 4;
      delay(500);
    }

Tinynja, 2020-11-08:
So..... any hint on what I could do next to figure out what I'm doing wrong? Could anyone try the first code I provided and tell me if it works on their end?

SpenceKonde, 2020-11-08:
Yeah, I don't see anything showing up either.... transmit is working though

closed 2020-11-08 by SpenceKonde

SpenceKonde, 2020-11-08:
Fixed - the test case was simple enough that head and tail never got pushed out of the registers and reloaded from main memory, and they were not declared volatile; first off, even in the absence of the optimizer, with such simple code, they'd never need to be reloaded from system memory - but in any event, I suspect (didn't bother to check) that the optimizer was eliminating the call to available entirely, as it could see that no reachable code could change the values of head and tail, and the tinySoftSerial code didn't tell it that the variables could be modified in an ISR, so the compiler was free to optimize that away.

SpenceKonde, 2020-11-08:
Was probably broken whenever LTO was in use (and only worked without it because the compiler didn't have the power to do that kind of optimization across files); tells you something about how many people are (trying to) use the software serial for RX - it's very widely used for TX for debugging.

Tinynja, 2020-11-09:
Omg thank you so much! It's working now! Thank you for making one of my first "issue" experience on github so pleasant :)

commit 2e5a79ab (2020-11-08): How the hell did tinySoftSerial *ever* work

Fixes #472