SpenceKonde/ATTinyCore #473: Issue with analogRead("PORT B pins")

daanstevenson, 2020-11-04:
I'm seeing some strange behavior in my system.

`analogRead(PIN_PA0)` works fine, with input signal on PA0.
`analogRead(PIN_PB0)` seems to break many things in my system. 
However, when using the MUXPOS for ADC11, system works as expected:
`analogRead(0x0B)` works fine with input signal on PA0.

I have registers set for custom PWM on Timer 1 and 2, and ISR for INT0 and PCINT1 (PIN_PA1).

sleemanj, 2020-11-04:
Spence is probably going to want to know which chip you are using.

daanstevenson, 2020-11-04:
Thanks, updated.

SpenceKonde, 2020-11-07:
From the "gotchas" section at the top of the readme

> When using analogRead(), use the A# constant to refer to the pin, not the digital pin number. Analog channel number (see table in datasheet entry for ADMUX register) can also be used - unlike the official core, you can use analogRead() with the differential ADC channels (for example).

The only reason it happens to work for PORTA is that the ADC channel numbers and the arduino pin numbers are the same with the pin mapping you have selected....

I am extremely dissatisfied with the way attinycore handles analog channels.... but changing that would be an, ah, maximally backwards combatable change.... the way analog channels are handled in classic AVR cores in general is a godawful mess - first the Atmel engineers, wearing blindfolds, threw darts at the pinout chart to place peripheral pins including ADC channels, then Arduino's official core started the analog-pin-digital-pin convention. 

If I had more time to work on ATTinyCore, I would do a 2.0 release, that would make this stuff work in a more intelligent manner, and fix a bunch of other dumb stuff I did in the past that can't be rectified without some measure of breakage.... My core development time is mostly directed at DxCore and megaTinyCore now. 

But yes - you cannot refer to analog pins by the digital pin number on ATTinyCore, only my modern AVR cores.

closed 2020-11-07 by SpenceKonde

daanstevenson, 2020-11-13:
Thanks Spence.
Apologies for missing this in the readme!