SpenceKonde/ATTinyCore #475: Unrelated piece of code breaks Serial functionnality

Tinynja, 2020-11-09:
Hello SpenceKonde, thanks again for fixing a piece of code related to Serial yesterday. I'm encountering another problem now. The following code only works if the `//UNRELATED CODE` section is **present**.

<details>
  <summary>ATTiny85 Code</summary>

    char msg[64];
    byte msg_size = 0;
    
    void setup()
    {
      OSCCAL = 0b1011101;
      Serial.begin(9600);
      DDRB |= 1 << 4;
    }
    
    void loop()
    {
      // UNRELATED CODE
      if (Serial.available()) {
        PORTB |= 1 << 4;
      } else {
        PORTB &= ~(1 << 4);
      }
      // MAIN CODE
      if (Serial.available()) {
        // Read incoming data and populate msg array
        msg[msg_size] = Serial.read();
        // Check if 0x4 
        switch (msg[msg_size]){
          case 0x17: // (end of trans. block)
            // Acknowledge reception of transmission block
            Serial.write(0x6);
            break;
          case 0x4: // (end of transmission)
            // Acknowledge reception of message
            Serial.write(0x6);
            // Append '\0' in order to use Serial.write()
            msg[msg_size+1] = '\0';
            // Send back full message
            Serial.write(msg);
            // Reset msg
            msg_size = 0;
            break;
          default: // (message body)
            msg_size += 1;
        }
      }
    }
</details>

<details>
  <summary>Python code on the computer</summary>

    from math import ceil
    import serial
    from serial.tools import list_ports
    
    tx_buffer = 30
    
    ports = list_ports.comports()
    for i,p in enumerate(ports):
    	print(f'{p.device}: {p.manufacturer} ({p.description})')
    
    tx_buffer -= 2
    ftdi = serial.Serial(input('USB Port: '), 9600, timeout=2)
    while True:
    	msg = input('\nMessage to send: ')
    	# Separate the message into groups of `tx_buffer` chars (ATTiny85 Serial RX buffer size)
    	for i in range(ceil(len(msg)/tx_buffer)):
    		# print('Sending: ', (msg[i*tx_buffer:min((i+1)*tx_buffer,len(msg))] + ('\x04' if (i+1)*tx_buffer > len(msg)-1 else '\x17')).encode())
    		ftdi.write((msg[i*tx_buffer:min((i+1)*tx_buffer,len(msg))] + ('\x04' if (i+1)*tx_buffer > len(msg)-1 else '\x17')).encode())
    		while not ftdi.in_waiting:
    			pass
    		resp = ftdi.read()
    		if resp != b'\x06':
    			print('Unexpected control sequence: ', resp)
    			break
    	print('Response: ' + ftdi.read_until('\x04'.encode()).decode()[:-1])
</details>

Notes:

- Running on ATTiny85
- Using FTDI232R as USB/Serial converter
- Arduino IDE 1.8.9
- ATTinyCore pulled on Nov 8, 2020
- Programmer: Arduino as ISP (ATTinyCore)
- Connections/wiring milluple checked with various other working examples
- Build options:
[image]

Let me know if you need any additional information.

SpenceKonde, 2020-11-09:
What does "works" mean as opposed to not working, ie, what is the symptom?

What about if you replace the "unrelated code" block with just `Serial.available()`? 
If that doesn't work, what about `volatile byte dummyvar=Serial.available()`?

SpenceKonde, 2020-11-09:
If one of those fixes it, see if moving that call to setup() also works.

Tinynja, 2020-11-09:
"Works" means the "communication protocol" does what I expect it to. I'm sending bytes with python to the ATTiny and chopping them in a buffer variable.

I'll try all your suggestions and report back the results!

Tinynja, 2020-11-09:
> What about if you replace the "unrelated code" block with just `Serial.available()`?

Same symptoms... the program doesn't seem to reach inside the main code's `if` statement.

> If that doesn't work, what about `volatile byte dummyvar=Serial.available()`?

This works!

I don't know if this could help debugging, but here are the flash and memory usage for each sketch I tried (as displayed from the arduino IDE console when uploading sketch):

- Original code as provided in the 1st post: 818B flash, 135B memory
- `UNRELATED CODE` section commented out: 802B flash, 135B memory
- `UNRELATED CODE` replaced with `Serial.available()`: 802B flash, 135B memory
- `UNRELATED CODE` replaced with `volatile byte dummyvar=Serial.available()`: 828B flash, 135B memory

SpenceKonde, 2020-11-09:
Well now we're getting somewhere!
The fact that there's no change to the flash size with and without Serial.available() without the return value being used clearly indicates that the compiler sees that you're not using the value and is optimizing away that call entirely. So it makes sense that there's no change with and without that line. So it does look like a second call to Serial.available is all that's needed - huh... 
Three new questions:
See if losing the 'volatile' still works. 
See if moving the line (the simplest one that works) to setup() instead of loop() works
See if removing those lines, but setting LTO to "disabled" works (you pay a huge premium in flash for this - goes from 808 to 1100 in a quick test - but if that makes it work, it's a *massive* clue)

Tinynja, 2020-11-09:
> See if losing the 'volatile' still works.

This doesn't work. Flash 802B, memory 135B.

> See if moving the line (the simplest one that works) to setup() instead of loop() works

Moving `volatile byte dummyvar=Serial.available()` to `setup()` doesn't work either. Flash 828B, memory 135B.

> See if removing those lines (that is, so it wouldn't work), but setting LTO to "disabled" works

I'm happy to give you the massive clue you were expecting, this works! I have absolutely no idea what you're telling me to do but I'm glad I can help you debug this haha! Flash 1098B, memory 135B.

SpenceKonde, 2020-11-09:
So gunning LTO fixes it in the absence of any other code that does... (so we know to file this under Optimizers Gone Wild)... but it's not enough to call it a second time anywhere, only in loop() itself... it's as if the compiler doesn't realize it can change...

These two issues both point to a lack of testing with the builtin tinySoftSerial since the LTO support was made enabled by default on June 19th... OF LAST YEAR! I think the first release was that August, and I'm pretty sure it was totally hosed in all cases always for rx until now...

SpenceKonde, 2021-05-28:
Well, I looked back at this with another 6 months of assembly experience. guess what also works? 

just creating a variable that won't be optimized away as a local variable in loop() is enough. I think even setup is enough. 
```
volatile byte dummyvar=0;
```
The big clue about what is going on, I think, is in these 10 bytes that are among the ones that appear in good configurations:

```
int main(void)
{
 2a8:	cf 93       	push	r28
 2aa:	df 93       	push	r29
 2ac:	1f 92       	push	r1
 2ae:	cd b7       	in	r28, 0x3d	; 61
 2b0:	de b7       	in	r29, 0x3e	; 62
```
push the Y register (r28 and r29 onto the stack, push 0 onto the stack) (now stack pointer is decremented by 3) then load it into the Y register.

I can't see how this isn't a compiler bug!

SpenceKonde, 2021-05-28:
Okay, i have a fix for 2.0.0. Not a satisfying one by any means, but.... it does the trick, (based on inspection of assembly - generated code has all the hallmarks of the good one.

SpenceKonde, 2021-05-28:
what a bunch of wacky juju!

SpenceKonde, 2022-01-26:
Needs to be tested if that voodoo is still needed after I refactored the hell ou of tinySoftSerial. Or for that matter if it even works at all...

SpenceKonde, 2022-01-28:
The wacky voodoo fix does appear to be needed still. I do not understand why. it is needed or how it fixes the problem

commit d9037a9a (2021-06-15): Fix #475

there is some bug relating to how the stack gets initialized, which something like this nonsense fixes
closed 2022-01-28 by SpenceKonde