SpenceKonde/ATTinyCore #477: 1.4.x/Micronucleus support breaks installs on ARM Linux systems

stonehippo, 2020-11-17:
The inclusion of Micronucleus in 1.4.x releases of ATTinyCore seem to have broken installation of the package on ARMv7 systems (such as Raspberry Pi OS 32-bit running on a RPi 3) and presumably other ARM-based systems.

For example, trying to install the latest ATTinyCore via `arduino-cli` on a system like the above results in the following error:

```
Error during install: tool ATTinyCore:micronucleus@2.0a4 not available for the current OS
```

Not sure if this is related to the ongoing discussion in #465. Installing 1.3.3 still works fine.

per1234, 2020-11-17:
You can see the hosts the `ATTinyCore:micronucleus` tool is available for here:
https://github.com/SpenceKonde/ReleaseScripts/blob/84083f5b74f3db023fe3501f2665103280ff1fc2/package_drazzy.com_index.json#L826-L855
In order to work on the Raspberry Pi, it would be necessary to have an entry in that list for the `arm-linux-gnueabihf` host (and another for `aarch64-linux-gnu` if you want to support 64 bit ARM linux systems as well)

stonehippo, 2020-11-17:
> In order to work on the Raspberry Pi, it would be necessary to have an entry in that list for the `arm-linux-gnueabihf` host (and another for `aarch64-linux-gnu` if you want to support 64 bit ARM linux systems as well)

I figured as much. Of course, the problem is that Digistump didn’t build for ARM, so @SpenceKonde has nowhere to add those architectures. I guess I can see if I can build the 2.0.a4 micronucleus release and put them somewhere. *sigh*

per1234, 2020-11-17:
I think it should be fine to host them on GitHub. The Linux version is especially small.

stonehippo, 2020-11-17:
I’m not too worried about where to put them, @per1234. I’m just thinking about having to go chase down getting them built (based on #465, I’m gathering that there’s some question about the current versions and getting things built).

stonehippo, 2020-11-17:
Ok, I built an ARM version of the Micronucleus commandline tool and put it in https://github.com/stonehippo/micronucleus-commandline-arm-builds. Based on what I’m seeing in the other issue, it looks like @sanbeg was building and collecting modern versions, so I’ve mentioned this over there, and I’m going to close this one.

closed 2020-11-17 by stonehippo