SpenceKonde/ATTinyCore #478: small delays a few cycles too short

nerdralph, 2020-11-17:
rcall takes 3 cycles and ret takes 4.  2 ldi instructions to initialize a uint16_t takes two more, The comparison and branch takes 4 more cycles, for a total of 13 cycles.  That's 3 cycles short of the 16 needed to make 1us.
https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/cores/tiny/wiring.c#L415

LaZsolt, 2020-11-18:
Hi!
7 days ago I have finished my delayMicroseconds() for LGT8Fx MCUs. This version of delayMicroseconds() solve this problem above. Check my code. https://github.com/LaZsolt/delayMicroseconds/tree/master/for_LGT8F

SpenceKonde, 2020-11-18:
@nerdralph  - thanks; easy enough to correct...  though as you said, cycle accurate timing is a bit much to expect from an Arduino core. I always wince when I see a short delayMicroseconds sandwiched between, say, two calls to digitalWrite()....

SpenceKonde, 2020-11-18:
@LaZsolt - for your 2-cycle NOP, why not use `rjmp .+0` - gets you 2 cycle NOP with only one instruction word...

LaZsolt, 2020-11-18:
@SpenceKonde - `rjmp .+0` takes only 1 clock cycle in LGT8Fx.

LaZsolt, 2020-11-18:
and `sbiw` is 1 clocktick too

SpenceKonde, 2020-11-18:
Eeeeeeew.... So, that means most neopixel libraries for the LGT parts don't work either, lol... compatible my ass! Glad Microchip has been putting out some real AVRs to occupy my time, else I'd have gone down the LGT8F rathole myself...

(see, a rathole is like a rabbithole, and they look the same from the surface, but when you get to the end, instead of awe and wonder, you're just dirty and covered in flea-bites (or in this sort of case, flea-bytes?) )

LaZsolt, 2020-11-18:
Some of Neopixel libraries are got fixed for LGT. https://github.com/dbuezas/lgt8fx/wiki/Libraries-support

nerdralph, 2020-11-19:
> Eeeeeeew.... So, that means most neopixel libraries for the LGT parts don't work either, lol... compatible my ass! Glad Microchip has been putting out some real AVRs to occupy my time, else I'd have [...]

It also means avr-gcc's built-in _delay_us and _delay_ms are way too short.  A couple months ago I tried I played with some LGT8F328P modules, and wasn't impressed.
http://nerdralph.blogspot.com/2020/10/lgt8f328p-edmini-board.html

Getting back on-topic, one thing I still don't like about the delayMicroseconds code is all the #if... #elif blocks.  There's essentially 9 different versions of the function.  In picocore, I originally just redefined delayMicroseconds to _delay_us, the same way @MCUdude does in MicoCore.  However there's more Arduino code in the wild than I thought there would be that won't work with that.  Sometimes the delays are determined at runtime, and often, even when it is known at compile time, gcc (even with LTO) isn't able to propagate the constant through various layers of indirection.  Therefore I just finished rewriting it to support variable delays, and there's only a few relatively simple lines of code for the function to support clock rates 4MHz and up.  Counting the 2 ldi instructions required to initialize the 16-bit us variable, it will delay exactly the number of required us plus one clock cycle.
https://github.com/nerdralph/picoCore/blob/master/avr/cores/picocore/Arduino.h#L72

SpenceKonde, 2020-11-19:
Woah, neat... Thanks @nerdralph  - have you heard of any compatibility problems with that?

nerdralph, 2020-11-19:
picoCore doesn't have near the number of users that ATTinyCore does, so I
don't get a lot of feedback.  However I have been helping Hans with
MicroCore for a couple years now, and other than the expected constraint of
a compile-time constant for the delay, there have been no issues with using
a macro to define delayMicroseconds as _delay_us.  Fundamentally, the
Arduino core version of delayMicroseconds does the same thing as
avr-gcc's _delay_us, which is a cycle-counting busy loop.  _delay_us is
more thoroughly tested, and when used directly it gives single-cycle
accurate delays.

I used the related function, __builtin_avr_delay_cycles, in picoUART to
create what I believe is the first cycle-accurate bit-bang AVR uart that
works over a wide range of baud rates and clock frequencies.
https://github.com/nerdralph/picoUART

LaZsolt, 2020-11-19:
@nerdralph 

>  there have been no issues with using a macro to define delayMicroseconds as _delay_us

I've been thinking about this for a while. So needless to care parameter type of delayMicroseconds() because it will be always a compile-time constant.

SpenceKonde, 2020-11-19:
Well, you can't #define delayMicroseconds() straight up as _delay_us() because as @nerdralph noted (and I was ready to if he didn't) that delayMicroseconds is often called with non-compiletime-known values. I was most interested in his solution for F_CPU >= 4 MHz,

yumkam, 2020-12-03:
Thanks to `LTO`, `delayMicroseconds(1)` is optimized away into no delay at all, which is kinda nasty. E.g. `for(;;) {PORTB &= ~1; delayMicroseconds(1); PORTB |= 1; delayMicroseconds(1); }` results in ~2MHz signal instead of ~500kHz.
Why not just `if(__builtin_constant_p(us)) return _delay_us(us);`?
This should fix constant delays (with `-flto`), won't affect non-constant delays, and trivial.

Related issue: arduino/ArduinoCore-avr#118

LaZsolt, 2020-12-03:
@yumkam 
You don't need to turn off LTO. Try this:
```
/* This structure designed for delayMicroseconds() to make sure compiler will not optimize by linker.
 * The idea came from Raymond Chen https://devblogs.microsoft.com/oldnewthing/20150102-00/?p=43233
 */
struct ForceFunctionToBeLinkedAndNotOptimized
{
  ForceFunctionToBeLinkedAndNotOptimized(void (* p) (unsigned int)) // This stuct will be stored, but never called.
    { GPIOR0=((uint16_t)p) & 0xFF; } // This is the shortest operation with the address in the LGT8Fx 8-bit AVR system without any affects.
};

ForceFunctionToBeLinkedAndNotOptimized ForceNotDoingSomething(delayMicroseconds_v);
```

SpenceKonde, 2020-12-03:
Yuck. FFS... 

Can you explain how the hell that works? 

How would I go about using that? Unless someone can tell me why this shouldn't be done, my inclination is that delayMicroseconds should *always* be passed through that unholy wizardry...

yumkam, 2020-12-03:
@LaZsolt your code will force compiler to generate generic delayMicroseconds instance. But it won't prevent compiler to inline (and optimize into nothing) delayMicroseconds with constant 1.

LaZsolt, 2020-12-03:
Explanation: https://devblogs.microsoft.com/oldnewthing/20150102-00/?p=43233

When a stucture refer to delayMicroseconds's address, and looking like doing something with it, then the compiler unable to determine how to optimize, So every line will be included to created code from the source.

yumkam, 2020-12-03:
@SpenceKonde 
this is a trick to fool *linker* into including generic version of function into resulting binary (so in can be called from debugger).
@LaZsolt 
It won't stop *compiler* from either inlining code, or producing specialized versions of code with specific constant arguments (there are `__attribute__((__noinline__))` and `__attribute__((__noclone__))` in gcc for that).
But we *want* linker to omit unused code and inlining `delayMicroseconds()` with constant arguments is beneficial for both code size and delay accuracy; it's just overhead calculation that is wrong in this case and needs to be fixed (and the least messy way is to just reuse `_delay_us()`, it produces smaller code and more accurate).

LaZsolt, 2020-12-03:
@yumkam 

Yes, you are right. Need some __attribute__`

LaZsolt, 2020-12-03:
Maybe the other solution is:
```
#define delayMicroseconds(us) do { if (__builtin_constant_p(us)) { delayMicroseconds_c(us); } else { delayMicroseconds_v(us); } } while (0)
#define delayMicroseconds_c(us) __builtin_avr_delay_cycles((uint32_t)(  (F_CPU/1e6)*((double)us)  )) // for constant case
 void   delayMicroseconds_v(unsigned int us) __something attribute__ ((xyzinline));                  // for variable case
```
in this case above, `us` value could be lower than 1 if it is not from a variable. (0.5 or even 0.125 microseconds on 16 MHz)

or
```
#define delayMicroseconds(us) do { if (__builtin_constant_p(us)) { _delay_us(us); } else { delayMicroseconds_v(us); } } while (0)
 void   delayMicroseconds_v(unsigned int us) __something attribute__ ((xyznoinline));                // for variable case
```

For variable case there are more very good solution from @nerdralph , @PaulStoffregen  or @matthijskooijman

Variables could be optimized in a register or read from RAM. Load `us` from register is 1 clock cycle and load from RAM 4 clock cycle.

I read that some cases the compiler could not determine by `__builtin_constant_p()` but LTO could. But it is not a problem if it will use the `delayMicroseconds_v()` function.

pcfreak1201, 2020-12-03:
Wouldn't it be easier to prevent the compiler from optimizing the delay at general?
Like here: https://github.com/RalphBacon/200-GCC-Compiler-Optimisation-for-Arduino/blob/main/Arduino_Blink_Optimisation_Demo.ino/Arduino_Blink_Optimisation_Demo.ino.ino

LaZsolt, 2020-12-03:
@pcfreak1201 

But it has no effect on LTO.

SpenceKonde, 2020-12-10:
Okay, if y'all can figure out how to do this and give me a PR, I will be more than happy to merge it! But uhhhh until then, this is a tar pit I choose not to step in.

matthijskooijman, 2020-12-10:
See also https://github.com/arduino/ArduinoCore-avr/issues/118 for some related work on the AVR core (I built an alternative implementation in a WIP-branch that is linked from there, IIRC I have some unpushed changes that I still need to clean up and submit properly). IIUC perfect delays should be feasible for constant delays (by forcing inline so the function call overhead disappears), for non-constant delays there is always some imperfection for short delays and I found a tradeoff where you'd have perfect delays (for longer delays), or rounded delays (to e.g. multiples of 2/3/4 cycles or so) but with better accuracy on shorter delays.

nerdralph, 2020-12-17:
I just pushed a version to picoCore that works for F_CPU >= 1MHz.
https://github.com/nerdralph/picoCore/blob/master/avr/cores/picocore/Arduino.h#L68

My primary goal was doing it in a minimal number of lines of code and without any conditional compilation (I hate when programmers use #ifdef outside of header files).  The code is always inlined for a couple reasons:
1) The code is small, so the cost of inlining is only 5-6 instructions per invocation
2) Correcting for function call overhead would add complexity to the function
3) The overhead of a function call is variable, and can only be determined by looking at the disassembled code.

The third point may not be obvious, as the naïve view of calling delayMicroseconds() assumes 4 instructions:
2 * ldi for the 16-bit parameter, rcall, and ret.
However, since the avr-gcc abi allows any function to clobber r18-r27 & Z, in all but the simplest of use cases, calling a function involves pushing and popping a few registers.

The delay will be the result of rounding to the nearest 4us, plus 3 clock cyles.  This should be consistent for any clock frequency >=1MHz that is an integer when multiplied by 4.  So even unusual clock frequencies like 10.5MHz should work.

LaZsolt, 2020-12-17:
@nerdralph 
Your code is great. Using `__builtin_avr_delay_cycles()` instead of `_delay_us()` inside of cycle could be better. What is your opinion?

nerdralph, 2020-12-17:
> 
> 
> @nerdralph
> Your code is great. Using `__builtin_avr_delay_cycles()` instead of `_delay_us()` inside of cycle could be better. What is your opinion?

_delay_us takes a floating point argument, and internally does the necessary math and rounding to call __builtin_avr_delay_cycles().  So for example when F_CPU is 8MHz, _delay_us(1.5) will delay for 12 cycles.

LaZsolt, 2020-12-17:
Ahh. I didn't know it.

SpenceKonde, 2020-12-28:
If anyone wants to submit a pull request that implements this in a way that *doesn't* break code for the damned fools who use delayMicroseconds() with non-compiletime-known arguments, I would welcome such a PR (I *wish* people didn't write code like that, but this is Arduino-land; There is amazing code, awful code, and everything inbetween.

PaulStoffregen, 2020-12-28:
You're welcome to help yourself to Teensy's delayMicroseconds (and delayNanoseconds) if you like.  The microsecond delay handles both const and variable inputs with different code using __builtin_constant_p().

https://github.com/PaulStoffregen/cores/blob/a0f8e1107a7d7fdb62f21b63842f70fe31db6367/teensy/core_pins.h#L2353

LaZsolt, 2020-12-28:
@PaulStoffregen 
Is there any reason you didn't to use in your code `_delay_us()` or `__builtin_avr_delay_cycles()` when input is const?

PaulStoffregen, 2020-12-28:
Yes.  As I understand (might be wrong?) the _delay_us() function requires a const input.

Until now, I did not know of __builtin_avr_delay_cycles().

LaZsolt, 2020-12-28:
@PaulStoffregen 

> the _delay_us() function requires a const input

But your code testing the input is a const or not: `if (__builtin_constant_p(usec))`

LaZsolt, 2020-12-28:
In delay.h:
static __inline__ void _delay_us(double __us) __attribute__((__always_inline__));

PaulStoffregen, 2020-12-28:
Yes, good point.

If you or anyone else wants to use the code, especially at other clock speeds, indeed it would make sense to replace the const case with _delay_us().  Probably good to use those other macros too.  Maybe make other improvements too?

Please understand my purpose here to is not debate any particular implementation or syntax.  It's only meant as a response to "If anyone wants to submit a pull request that implements this in a way that doesn't break code for the damned fools who use delayMicroseconds() with non-compiletime-known arguments, I would welcome such a PR".  Rather than a PR, it merely a link to the code we've been using successfully on AVR-based Teensy for the last ~10 years.  Indeed it does meet that requirement, though I personally wouldn't call people who use non-const inputs "damned fools".

If this helps, great.  If you or others redesign it with avr-libc macros, even better.

If you object based on style, if you expect me to rewrite this very mature code that's been in widespread use for about a decade, well, that's far more than I intend to do to help.  Really, the extent of my contribution here is just pointing to the const & variable solution that was worked well all these years.  The rest is up to you.

LaZsolt, 2020-12-28:
Thank you your answer. I could redesign this function. MCUdude accepted my first version of `delayMicroseconds()` modification.

LaZsolt, 2021-01-07:
I just finished my new `delayMicroseconds()` function. Here is my design guide: https://github.com/LaZsolt/delayMicroseconds/blob/master/README.md
And this is <b>a part of the code</b> wich is not tested yet. Maybe tomorrow I will test how it will work.
I write it for 32, 30, 25, 24, 22, 20, 18 (=18.432), 16, 15 (=14.7456), 14, 12, 11 (=11.0592), 10, 9, 8, 7, 6, 5, 4, 2, 1 MHz
```
static inline void delayMicroseconds(uint16_t) __attribute__((always_inline, unused));

#define _NOP0_ ""
#define _NOP1_ "nop \n\t"
#define _NOP2_ "rjmp .+0 \n\t"
...
// Multiply by two takes 2 clock cycles
#define _MULby2_                     \
          "add    %A0,%A0      \n\t" \
          "adc    %B0,%B0      \n\t" 

// Multiply by three takes 4 clock cycles
#define _MULby3_                     \
          "add    %A0,%A0      \n\t" \
          "adc    %B0,%B0      \n\t" \
          "add    %A0,%A1      \n\t" \
          "adc    %B0,%B1      \n\t"

// Multiply by four takes 4 clock cycles
#define _MULby4_                     \
          "add    %A0,%A0      \n\t" \
          "adc    %B0,%B0      \n\t" \
          "add    %A0,%A0      \n\t" \
          "adc    %B0,%B0      \n\t" 

#if   F_CPU >= 32000000L

  #define _CCALCULATE_   _MULby4_    \
          "sbiw   %A0,1        \n\t"
  #define _TUNINGNOPS_   _NOP2_
  #define _NOPSinCYCLE_  _NOP4_
  #define _SBIWinCYCLE_              \
          "sbiw   %A0,1        \n\t"

#elif F_CPU >= 18000000L

  #define _CCALCULATE_  _MULby3_     \
          "sbiw   %A0,1        \n\t"
  #define _TUNINGNOPS_  _NOP0_
  #define _NOPSinCYCLE_ _NOP2_
  #define _SBIWinCYCLE_              \
          "sbiw   %A0,1        \n\t"

#elif F_CPU >= 1000000L

  #define _CCALCULATE_  ""
  #define _TUNINGNOPS_  _NOP0_
  #define _NOPSinCYCLE_ _NOP0_
  #define _SBIWinCYCLE_              \
          "sbiw   %A0,4        \n\t"

#endif

static inline void delayMicroseconds(uint16_t us)
{
    if (__builtin_constant_p(us)) {
      _delay_us(us);
    } else {
        uint16_t tmp;
        asm volatile(
                    // Movw compensates for the missing cycle in
                    // the last brge below, and also prevents the
                    // compiler from inserting this one. Both operands
                    // might actually be the same register, but that's
                    // ok.
          "movw   %A0, %A1     \n\t"  // 1 ( tmp = us; )
          _CCALCULATE_                // Cycle counter multiplication and subtraction if needed
          _TUNINGNOPS_
          "1:                  \n\t"
          _NOPSinCYCLE_
          _SBIWinCYCLE_               // 2
          "brge   1b           \n\t"  // 2 (1 on last)
          : "=w" (tmp)
          : "r"  (us)
        );
    }
}
```

LaZsolt, 2021-01-12:
Now my tested code and documents are available here: https://github.com/LaZsolt/delayMicroseconds/tree/master/for_AVR

Possible improvement in future if needed:
- 1.) In constant delay case the requested timings may have fractional part and shorter delay than 1 microsecond.
- 2.) Different working frequencies for the case of change the clock speed at the fly. (Different clock prescaler settings.)

SpenceKonde, 2021-05-29:
Will use the LaZsolt solution for 2.0.0. Much thanks.

SpenceKonde, 2022-02-07:
Does anyone remember why I wasn't jacking the version by @nerdralph instead of @LaZsolt? 

One of those two will be used  I'm not going with the one derived from the modern AVR parts. I'm trying to get rid of makework additions that won't be noticed by users.

commit 4c9c151c (2022-02-11): fix #478

Nobody objected to this in #478 so;....
closed 2022-02-11 by SpenceKonde