SpenceKonde/ATTinyCore #485: Remove 5 second boot delay on digispark pro (attiny 167)

Rimbaldo, 2020-12-19:
Digispark Pro (attiny167) comes with a 5 second delay before running the code, to allow programming over usb.  There’s some ways of programming it to remove this delay and choose a pin to be grounded  to be able to code into it. Could this “feature” be added into your core? Like a menu option to remove the 5 second delay and maybe be able to choose a pin to be used as a programming pin when grounded? I’ve got 4 ready made hex files with 4 different pins (and no delay to start the code) , but I lack knowledge to create the hex with any other pin I wanted to choose.

tomaskovacik, 2020-12-27:
looks like only another bootloader need to be used:

http://digistump.com/wiki/digispark/tricks:

Get rid of 5 second startup delay

When the digispark is powered up, it waits 5 seconds to give your computer a chance to upload a different program. For some projects this delay is less than ideal. For these situations we have another version of the bootloader which only accepts uploads if you connect D5 to ground with a wire or a button before plugging the digispark in. You can add an 'update' button to your project for times you want to change its software, and use the button for other stuff in your app, or use the reboot to bootloader code above to make it reboot and accept a program without having to unplug and replug the device.

Instructions are on the forum for Mac and Windows. Linux users should follow the Mac instructions pretty much. Note that this is an unsupported modification and does currently void your digispark's warranty. So far nobody seems to have had any trouble with it though.

The newest version of the jumper firmware can be found here: https://github.com/Bluebie/micronucleus-t85/tree/master/upgrade/releases

Rimbaldo, 2020-12-27:
> looks like only another bootloader need to be used:
>
> http://digistump.com/wiki/digispark/tricks:
>
> Get rid of 5 second startup delay
>
> When the digispark is powered up, it waits 5 seconds to give your [...]

Hi! Thanks for the answer.  Unfortunately this text and method you posted is for the 5 second delay on the DIGISPARK, which uses an ATTINY85.

 I´m asking for a "menu driven" option on Spencekonde´s Attiny library for generating these HEXes for removing the 5 second boot delay and choosing any  pin to act as a programming pin on the DIGISPARK PRO, which uses an ATTINY167. 

For the Digispark PRO (Attiny167) I have three Hexes already made (not by me)  that removes the 5 second delay and each HEX uses a different PIN to act as a programming pin. But it would be nice to have an option to use any pin as a programming pin, and generate and hex on the fly,  and it would be easier if it was inside the menu on the Arduino IDE.

tomaskovacik, 2020-12-28:
it was just note
I do not plane to work on it, but since it only requires new bootloader, the simplest way to implement will be to add  another bootloader to existing menu in boards.txt:

add these lines after "attinyx7micr.menu.burnmode.install.bootloader.tool=avrdude" in boards.txt 

note:(change path to your existing hex files and change pin numbers)

attinyx7micr.menu.burnmode.installnodelaypin1=Fresh Install (via ISP) - no delay/pin 1
attinyx7micr.menu.burnmode.installnodelaypin1.bootloader.file=micronucleus/t167_entry_on_power_on_no_pullup_fast_exit_on_no_USB_no_delay_pin_1.hex
attinyx7micr.menu.burnmode.installnodelaypin1.bootloader.tool=avrdude
attinyx7micr.menu.burnmode.installnodelaypin2=Fresh Install (via ISP) - no delay/pin 2
attinyx7micr.menu.burnmode.installnodelaypin2.bootloader.file=micronucleus/t167_entry_on_power_on_no_pullup_fast_exit_on_no_USB_no_delay_pin_2.hex
attinyx7micr.menu.burnmode.installnodelaypin2.bootloader.tool=avrdude
attinyx7micr.menu.burnmode.installnodelaypin3=Fresh Install (via ISP) - no delay/pin 3
attinyx7micr.menu.burnmode.installnodelaypin3.bootloader.file=micronucleus/t167_entry_on_power_on_no_pullup_fast_exit_on_no_USB_no_delay_pin_3.hex
attinyx7micr.menu.burnmode.installnodelaypin3.bootloader.tool=avrdude
attinyx7micr.menu.burnmode.installnodelaypin4=Fresh Install (via ISP) - no delay/pin 4
attinyx7micr.menu.burnmode.installnodelaypin4.bootloader.file=micronucleus/t167_entry_on_power_on_no_pullup_fast_exit_on_no_USB_no_delay_pin_4.hex
attinyx7micr.menu.burnmode.installnodelaypin4.bootloader.tool=avrdude
attinyx7micr.menu.burnmode.installnodelaypin5=Fresh Install (via ISP) - no delay/pin 5
attinyx7micr.menu.burnmode.installnodelaypin5.bootloader.file=micronucleus/t167_entry_on_power_on_no_pullup_fast_exit_on_no_USB_no_delay_pin_5.hex
attinyx7micr.menu.burnmode.installnodelaypin5.bootloader.tool=avrdude
attinyx7micr.menu.resetpin.reset=Reset

it will look something like this:

[image]

btw, bootloader files are not compiled on demand, so if you like to have all pins as options, you have to have as much hex files as there are pins on the chip suitable for this function

SpenceKonde, 2020-12-28:
Yeah - there are already plans to have a considerable variety of bootloaders with different entry conditions (eg, whether reset will enter it, specific pin and so on...) I was part way through this when I had to direct my immediate attention to megaTinyCore and DxCore, after DxCore 1.3.0 and megaTinyCore 2.2.0 are released, I will be able to come back to this; among other things, we're also getting micronucleus support for the t861 and t1634.

SpenceKonde, 2020-12-28:
The implementation will be a new submenu for the bootloader entry condition - I think there are already some parts that have that, check around the other micronucleus board definitions - that way you'll be able to change that setting via either upgrade or fresh install.

SpenceKonde, 2020-12-28:
This falls under #429 - closing.

closed 2020-12-28 by SpenceKonde