SpenceKonde/ATTinyCore #486: ATtiny84A PWM issue with Timer1

JChristensen, 2020-12-21:
I am still experiencing the PWM issue as described in issue #470. My wiring.c file (packages/ATTinyCore/hardware/avr/1.4.1/cores/tiny/wiring.c), does not match wiring.c in commit 106d0a8. 
[wiring.zip](https://github.com/SpenceKonde/ATTinyCore/files/5725070/wiring.zip)

I've been using ATTinyCore for a while, I really like it. Thank you!
```c++
// Demonstrate ATTinyCore PWM issue, incorrect PWM on Timer1.
//
// Issue occurs with ATTinyCore 1.4.1 and 1.4.0, not with 1.3.3.
// Chip: ATtiny84A, no bootloader, 1MHz internal clock, BOD disabled, EEPROM retained,
// CW pin mapping, LTO enabled, NeoPixel Port A, millis() enabled.
// Arduino IDE 1.8.13.

uint8_t
    led0 {8},   // LED on Timer0
    led1 {6};   // LED on Timer1

void setup()
{
    pinMode(led0, OUTPUT);
    pinMode(led1, OUTPUT);
    delay(2000);
}

void loop()
{
    analogWrite(led0, 255);     // full brightness
    analogWrite(led1, 255);     // full brightness
    delay(2000);

    analogWrite(led0, 254);     // not noticeably different, as expected
    analogWrite(led1, 254);     // noticeably dimmer
    delay(2000);
}
```

SpenceKonde, 2020-12-21:
That is correct the fix is not yet in a release. It will be in 1.4.2. I do
not have any eta on 1.4.2 release.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

JChristensen, 2020-12-21:
Oh, I misunderstood, sorry. Thanks, I'll watch for it.

closed 2020-12-21 by JChristensen