SpenceKonde/ATTinyCore #487: Support for Microchip ATtiny202/402

ergerd, 2021-01-05:
Is there any chance to support the Microchip AVR ATtiny202/402 CPUs?

freemovers, 2021-01-05:
ATtiny202/402 have a different core and can be found here:
https://github.com/SpenceKonde/megaTinyCore
For more information go to:
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x02.md

SpenceKonde, 2021-01-05:
Yup -all the tinyAVR 0/1-series parts from the lowly 202's to the 3217 are supported by megaTinyCore - and soon the 2-series (if Microchip ever releases it!)

SpenceKonde, 2021-01-05:
To be clear - the architecture and peripherals of those parts are so different that they belong in an entirely separate core.

freemovers, 2021-01-05:
> 
> 
> Yup -all the tinyAVR 0/1-series parts from the lowly 202's to the 3217 are supported by megaTinyCore - and soon the 2-series (if Microchip ever releases it!)

Looks like I have to complete some pin-out diagrams for those as well!

SpenceKonde, 2021-01-05:
Yup. The 2-series appears to be a few months away. Which is about how far
off it has been since the summer... my guess is that crazy new ADC has been
giving them trouble >.>

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

closed 2021-01-06 by SpenceKonde