SpenceKonde/ATTinyCore #490: ATTiny841 Pin_PB2 outputs high, 3X 64ms square wave during Optiboot

daanstevenson, 2021-01-07:
Hi Dr. Azzy,

I'm noticing that Pin_PB2 on the ATTiny841 produces this output wave after a reset, even with empty code. 
It is a square wave of 3x 64ms high output pulses. None of the other pins exhibit any unexpected behavior like this.
I'm guessing it occurs during the Optiboot bootloader routine, as it doesn't happen on supply powerup. 
8MHz internal clock, UART0 used for bootloader. 

[image]

Unfortunately in my application this pin powers a high voltage output that is only designed to take specific input PWM signal, and is pulling a lot of current that dips the supply during startup. I might be able to move some pins around in the design, but let me know if you can think of why this is happening and if there is a workaround.

Thanks!

SpenceKonde, 2021-01-08:
That is the pin that is assumed to have the LED connected to it. It is doing the Optiboot "triple blink" so you know the bootloader is running. I admit that the documentation does not do a good job (or, for that matter, a bad job) of communicating which pins are used for the LED blink functionality. It's the same pin as LED_BUILTIN - but that's also not listed in the documentation.

If you need that pin to be left unmolested, either build Optiboot with LED_START_FLASHES=0, or don't use the bootloader at all.

daanstevenson, 2021-01-08:
Thank you Spence. I hope it's ok to post issues like this here, even if they don't turn out to be issues.
Happy to message you instead...

SpenceKonde, 2021-01-08:
Posting issues like this is not a problem. Certainly, one could also email me ( spencekonde@gmail.com ), but marking issues as tech support and closing them is prety easy fpor me - and in this case, it did raise an issue that does deserve to be fixed, namely that it's not documented ( #491 ) 

I should have been able to point you to the part specific docs and been like "This behavior is clearly documented, see the part-specific documentation pages" and linked you to it :-P

closed 2021-01-08 by SpenceKonde

daanstevenson, 2021-01-08:
After a long night and a long morning I finally figured out how to build the optiboot bootloader on my Windows machine, so I figured I would post here to help others.

There is good information on this process in this post, but some references and repos are outdated:
https://forum.arduino.cc/index.php?topic=424433.0

Here is my approach:
1. Clone SpenceKonde/ATTinyCore to ‘C:\Users\$username$\Documents\Arduino\hardware\’ (no Boards Manager)
2. Install make GNUWin32 from here:
https://sourceforge.net/projects/gnuwin32/
3. Include make.exe in set path (‘C:\Program Files (x86)\GnuWin32\bin’)
4. Copy ‘omake.bat’ and 'install-avr-tools.bat’ from:
https://github.com/Optiboot/optiboot/tree/master/optiboot/bootloaders/optiboot
To: ‘...\ATTinyCore\avr\bootloaders\optiboot’ (back up original omake.bat)
5. In Powershell, execute: 
>> cd C:\Users\$username$\Documents\Arduino\hardware\ATTinyCore\avr\bootloaders\optiboot
>> .\omake.bat attiny841     AVR_FREQ=8000000L  BAUD_RATE=57600 LED_START_FLASHES=0 LOWERCAL=7
(for _int version)
6. Rename hex file as appropriate
7. Edit boards.txt file in '...\ATTinyCore\avr’ as appropriate

I still get "Error while burning bootloader." message about 50-75% of the time while burning bootloader (even with the original bootloaders), even with USBTinyISP SLOW.
Verbose output is the same as with a successful bootloader burn, and the chips seems to behave ok with either result.

Thank you Dr. Azzy for your awesome core!

SpenceKonde, 2021-01-08:
Huh - what I do is take an Arduino 1.0.6 installation (it has all the build tools), copy modern compiler over the old one, and copy the optiboot directory into (arduino 1.0.6 with updated compiler)/hardware/arduino/bootloaders/ - then I can use the version of omake included in the repo. I think somewhere there's a text file that describes that process, albeit a very terse one....