SpenceKonde/ATTinyCore #494: ATTiny1634 reading and flashing issues

vishalbhat88, 2021-01-14:
Hey,
I am using an ATTiny1634 QFN-20 package and am trying to read and flash it. However i am facing issues with reading the controller. I have tried using an Arduino Uno with ArduinoISP sketch as an ISP programmer as well as a USBasp programmer, with no sucess. I have used the arduino IDE as well as avrdude to try to read it and then flash, with no avail. 
With avrdude using USBasp programmer, this is the output i get:
$ avrdude -P usb -c usbasp -p t1634

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

If i override this using -F, i get an invalid device signature error as so:

$ avrdude -P usb -c usbasp -p t1634 -F

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x601d8a
avrdude: Expected signature for ATtiny1634 is 1E 94 12

avrdude done.  Thank you.

I tried using different sck frequencies from -B1 to -B20, but it didnt work. I have checked the USBasp programmer and have been able to read and flash an atmega328p controller.
 
I have made changes to the avrdude.conf file as suggested in another post, this is my avrdude.conf file:

#------------------------------------------------------------
# ATtiny1634
#------------------------------------------------------------

part
    id              = "t1634";
    desc            = "ATTINY1634";
     has_debugwire = yes;
     flash_instr   = 0xB6, 0x01, 0x11;
     eeprom_instr  = 0xBD, 0xF2, 0xBD, 0xE1, 0xBB, 0xCF, 0xB4, 0x00,
	             0xBE, 0x01, 0xB6, 0x01, 0xBC, 0x00, 0xBB, 0xBF,
	             0x99, 0xF9, 0xBB, 0xAF;
    stk500_devcode  = 0x86;
    # avr910_devcode = 0x;
    signature       = 0x1e 0x94 0x12;
    pagel           = 0xd7;
    bs2             = 0xc2;
    chip_erase_delay = 9000;
    pgm_enable       = "1 0 1 0 1 1 0 0 0 1 0 1 0 0 1 1",
                       "x x x x x x x x x x x x x x x x";

    chip_erase       = "1 0 1 0 1 1 0 0 1 0 0 x x x x x",
                       "x x x x x x x x x x x x x x x x";

    timeout         = 200;
    stabdelay       = 100;
    cmdexedelay     = 25;
    synchloops      = 32;
    bytedelay       = 0;
    pollindex       = 3;
    pollvalue       = 0x53;
    predelay        = 1;
    postdelay       = 1;
    pollmethod      = 1;

    pp_controlstack     =
	0x0E, 0x1E, 0x0F, 0x1F, 0x2E, 0x3E, 0x2F, 0x3F,
	0x4E, 0x5E, 0x4F, 0x5F, 0x6E, 0x7E, 0x6F, 0x7F,
	0x66, 0x76, 0x67, 0x77, 0x6A, 0x7A, 0x6B, 0x7B,
	0xBE, 0xFD, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00;
    hventerstabdelay    = 100;
    progmodedelay       = 0;
    latchcycles         = 5;
    togglevtg           = 1;
    poweroffdelay       = 15;
    resetdelayms        = 1;
    resetdelayus        = 0;
    hvleavestabdelay    = 15;
    resetdelay          = 15;
    chiperasepulsewidth = 0;
    chiperasepolltimeout = 10;
    programfusepulsewidth = 0;
    programfusepolltimeout = 5;
    programlockpulsewidth = 0;
    programlockpolltimeout = 5;

    memory "eeprom"
        paged           = no;
        page_size       = 4;
        size            = 512;
        min_write_delay = 3600;
        max_write_delay = 3600;
        readback_p1     = 0xff;
        readback_p2     = 0xff;
        read            = " 1 0 1 0 0 0 0 0",
                          " 0 0 0 x x x x a8",
                          " a7 a6 a5 a4 a3 a2 a1 a0",
                          " o o o o o o o o";
    
        write           = " 1 1 0 0 0 0 0 0",
                          " 0 0 0 x x x x a8",
                          " a7 a6 a5 a4 a3 a2 a1 a0",
                          " i i i i i i i i";

	loadpage_lo	= "  1   1   0   0      0   0   0   1",
			  "  0   0   0   0      0   0   0   0",
			  "  0   0   0   0      0   0  a1  a0",
			  "  i   i   i   i      i   i   i   i";

	writepage	= "  1   1   0   0      0   0   1   0",
			  "  0   0   x   x      x   x   x  a8",
			  " a7  a6  a5  a4     a3  a2   0   0",
			  "  x   x   x   x      x   x   x   x";

	mode		= 0x41;
	delay		= 5;
	blocksize	= 4;
	readsize	= 256;
        ;

    memory "flash"
        paged           = yes;
        size            = 16384;
        page_size       = 32;
        num_pages       = 512;
        min_write_delay = 4500;
        max_write_delay = 4500;
        readback_p1     = 0xff;
        readback_p2     = 0xff;
        read_lo         = " 0 0 1 0 0 0 0 0",
                          " 0 0 0 a12 a11 a10 a9 a8",
                          " a7 a6 a5 a4 a3 a2 a1 a0",
                          " o o o o o o o o";
        
        read_hi          = " 0 0 1 0 1 0 0 0",
                           " 0 0 0 a12 a11 a10 a9 a8",
                           " a7 a6 a5 a4 a3 a2 a1 a0",
                           " o o o o o o o o";
        
        loadpage_lo     = " 0 1 0 0 0 0 0 0",
                          " 0 0 0 x x x x x",
                          " x x x x a3 a2 a1 a0",
                          " i i i i i i i i";
        
        loadpage_hi     = " 0 1 0 0 1 0 0 0",
                          " 0 0 0 x x x x x",
                          " x x x x a3 a2 a1 a0",
                          " i i i i i i i i";
        
        writepage       = " 0 1 0 0 1 1 0 0",
                          " 0 0 0 a12 a11 a10 a9 a8",
                          " a7 a6 a5 a4 x x x x",
                          " x x x x x x x x";

        mode        = 0x41;
        delay       = 6;
        blocksize   = 128;
        readsize    = 256;

        ;
        
    memory "lfuse"
        size            = 1;
        min_write_delay = 4500;
        max_write_delay = 4500;
        read            = "0 1 0 1 0 0 0 0 0 0 0 0 0 0 0 0",
                          "x x x x x x x x o o o o o o o o";
        
        write           = "1 0 1 0 1 1 0 0 1 0 1 0 0 0 0 0",
                          "x x x x x x x x i i i i i i i i";
        ;
    
    memory "hfuse"
        size            = 1;
        min_write_delay = 4500;
        max_write_delay = 4500;
        read            = "0 1 0 1 1 0 0 0 0 0 0 0 1 0 0 0",
                          "x x x x x x x x o o o o o o o o";
        
        write           = "1 0 1 0 1 1 0 0 1 0 1 0 1 0 0 0",
                          "x x x x x x x x i i i i i i i i";
        ;
    
    memory "efuse"
        size            = 1;
        min_write_delay = 4500;
        max_write_delay = 4500;
        read            = "0 1 0 1 0 0 0 0 0 0 0 0 1 0 0 0",
                          "x x x x x x x x x x x x x o o o";
        
        write           = "1 0 1 0 1 1 0 0 1 0 1 0 0 1 0 0",
                          "x x x x x x x x x x x x x i i i";
        ;
    
    memory "lock"
        size            = 1;
        min_write_delay = 4500;
        max_write_delay = 4500;
        read            = "0 1 0 1 1 0 0 0 0 0 0 0 0 0 0 0",
                          "x x x x x x x x x x o o o o o o";
        
        write           = "1 0 1 0 1 1 0 0 1 1 1 x x x x x",
                          "x x x x x x x x 1 1 i i i i i i";
        ;
    
    memory "calibration"
        size            = 1;
        read            = "0 0 1 1 1 0 0 0 0 0 0 x x x x x",
                          "0 0 0 0 0 0 0 0 o o o o o o o o";
        ;
    
    memory "signature"
        size            = 3;
        read            = "0 0 1 1 0 0 0 0 0 0 0 x x x x x",
                          "x x x x x x a1 a0 o o o o o o o o";
        ;
;

According to this post (https://alainsprojects.com/2019/09/23/brand-new-attinys-and-the-problems/) I also tried high voltage programming to set the fuse bits so that the device signature can be set. I tried high voltage programming but however the fuse bits too aren't being read.
I even tried giving an external clock signal of 8MHZ to XTAL1 pin in case the fuse bits were set to an external clock but the attiny1634 still could not be read or flashed
I haven't changed the fuse bits before nor have i burnt the bootloader, so i am not sure what the issue could be. Any tips that you could provide with would be really helpful.

SpenceKonde, 2021-01-14:
Well, the -B parameters ain't gonna do shit - almost all usbasp firmwares
ignore them (you see that warning?) and use their own auto-SCK algorithm.

There *is* something about the ATtiny1634, ATtiny841, and ATtiny828 that is
fiddlier about programming, When I was first using them, sometimes, I just
*could not make them program* with my USBAsps... But that problem hasn't
appeared for me in eons... I don't really understand that phenomenon.

However:
* One theory is that they are pickier about the SCK clock than most classic
AVRs when running at 1 MHz. When I was plagued by it - maybe I had a pair
of USBAsps that came preloaded with lousy firmware? Those two got the
upgraded fw, but I never bothered on subsequent ones (I now have between
half a dozen and a dozen USBAsps in a bin, as well as about as many
USBTinyISPs... The USBTinyISPs listen to the -B parameter (and don't try to
auto-SCK), though you need to choose slow option when bootloading virgin
chips). The ArduinoAsISP also has an option to set the SCK speed.
* Another theory is that reset gets mishandled by some programmers, and
again, that these parts are pickier about it; I'm suspicious that there may
be something about the timing of asserting or releasing reset or something
(which, of course, would also be more sensitive at 1 MHz internal on virgin
chip) . I recall that the "fixed" USBAsp firmware dealt with some issue
about mishandling reset. I seem to recall on one occasion manually holding
the reset line up against ground and letting go of it at appropriate times
while programming, way back).
* I was always able to talk to them with an Arduino as ISP (I forget
whether I slowed the SCK or not), and USBTinyISPs at the slow speed also
seem to work reliably (though at the same time, my standard programmer was
a USBAsp for years, and I only had problems a couple of times years ago);
it wasn't until recently when I realized the USBTinyISPs come with microusb
connectors, that I started using those more often.

I've thought about this a few times since that original round of problems,
and it seems more and more plausible that the orignal issue (assuming it
wasn't a PEBCAK) was a batch of USBAsps with worse firmware than usual...
Especially since I've seen a few times that, despite the same SCK speed vs
main clock specs, these will require a slower clock, while most classic
AVRs don't.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

vishalbhat88, 2021-01-14:
Hey,
Could you help me by pointing out or sharing the USBasp firmware that has worked for you?

pcfreak1201, 2021-01-14:
Hi! If you have a "classic" chinese USBAsp like this: https://blog.podkalicki.com/wp-content/uploads/2019/03/usbasp-v2.0-board-pinout-1024x537.jpg (https://blog.podkalicki.com/how-to-update-avr-usbasp-firmware-to-latest-version/) set (solder in) the jumper J3 - works fine with my version.

vishalbhat88, 2021-01-14:
Thank you for your help, updating the firmware for the programmer helped

SpenceKonde, 2021-01-14:
Oh god, that's an ancient version of the f/w! https://forum.arduino.cc/?topic=363772 is the latest/best version afaik.

closed 2021-01-14 by SpenceKonde