SpenceKonde/ATTinyCore #496: package_drazzy.com_index.json - the megaTinyCore entry for version 2.2.2 is dislocated

ArminJo, 2021-01-14:
My Sloeber (Eclipse pluging) complains 
```
java.lang.NullPointerException
	at io.sloeber.core.managers.Tool.setParent(Tool.java:33)
	at io.sloeber.core.managers.Package.setParent(Package.java:45)
	at io.sloeber.core.managers.PackageIndex.setOwners(PackageIndex.java:40)
	at io.sloeber.core.api.PackageManager.loadPackage(PackageManager.java:686)
	at io.sloeber.core.api.PackageManager.loadJson(PackageManager.java:673)
	at io.sloeber.core.api.PackageManager.loadJsons(PackageManager.java:645)
	at io.sloeber.core.managers.InternalPackageManager.startup_Pluging(InternalPackageManager.java:74)
	at io.sloeber.core.Activator$2.run(Activator.java:247)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
```
about the new http://drazzy.com/package_drazzy.com_index.json.
It is because the entry at line 3432
```
"name": "megaTinyCore",
          "architecture": "megaavr",
          "version": "2.2.2",
```
was slipped behind the tools section at line 3379.
Putting it at the right place - before "version": "2.2.3", fixes the error.

ArminJo, 2021-01-14:
I got an 404 after first submit, so I submitted it again.

closed 2021-01-14 by ArminJo

SpenceKonde, 2021-01-14:
Yeah, I have been getting the 404 after creating an issue too lately. Couldn't tell you why...