SpenceKonde/ATTinyCore #497: ATTiny1634 USART0 code

vishalbhat88, 2021-01-16:
Hey,
Can someone help me out by providing a complete example code for USART0 using hardware serial just so that i can test the UART functionality

SpenceKonde, 2021-01-16:
Uh? Like, it's literally exactly like the serial port on any other normal Arduino... 

```
void setup() {
// TX is pin 0 (PB0);
// RX is pin 1 (PA7);
Serial.begin(19200);
Serial.println(F("Hi from USART0"));

// the 1634 has **TWO** hardware serial ports! 
//They both work the same way. 

//TX is on pin 15 (PB2)
//RX is on pin 16 (PB1)
Serial1.begin(19200);
Serial1.println(F("Hi from USART1"));
}
void loop() {
  while (Serial.available()) {
    Serial1.print(Serial.read());
  }
  while (Serial1.available()) {
    Serial.print(Serial1.read());
  }
}
```

Off the top of my head j

Are you having problems? 

Remember the "standard" check list: 
* "Burn bootloader" to get it running on either internal 8 MHz, or the external crystal you connected per pinout diagram). 
*  If you're running it from internal at 5v, be sure to choose the "Internal 8 MHz, Vcc >4.5" clock source. The internal oscillator has a very strong voltage dependence - enough that you get gibberish out of serial at 8 MHz/5v... A few versions ago I realized just how consistent the oscillators are on the 841/828/1634, and added that option - they just take a guess at how much OSCCAL0 needs to be nudged down by and do it on startup; it''s good enough to make sure serial works even with internal oscillator.

vishalbhat88, 2021-01-16:
The normal hardware Serial libraries are not working, i have even tried the register level C code that's given in their datasheet but i'm still having problems. I've burnt the boot loader and also using the "Internal 8 MHz, Vcc >4.5" clock source. Any other suggestions that may help?

SpenceKonde, 2021-01-16:
So what isn't working?

vishalbhat88, 2021-01-18:
I checked the TX and RX pins on a digital oscilloscope and do not see the start bit, data bits and stop bits. I just get a flat 1V output on the TX pin when i use the hardware serial libraries.

vishalbhat88, 2021-01-29:
Is there any way to tune the OSCCAL values for ATTiny1634MUR? Or is there some other issue other than the internal oscillator accuracy?

SpenceKonde, 2021-01-29:
Yes. Write a different value to OSCCAL1? It is volatile - reset every time the chip starts up (the factory calibration can't be changed, and at Vcc > 4 V or so, it's always fast, above 4.5v too fast for serial to work; if you choose >4,5V, the core should be nudging the speed down enough for serial to work though!

Have you gotten beyond the issue above, where it sounds like you must have been on the wrong pin, or using a pin not set output (which my core's Serial implementation does

vishalbhat88, 2021-01-29:
Nope I've checked the pin-outs multiple times and i am using UART0 which is PB0 (TX) and PA7(RX) . I have also chosen the >4.5V option when flashing so that the OSCCAL1 values can be nudged down as you have mentioned that your core does. However i am still getting nothing when i check the TX and RX pins on a digital oscilloscope. I tried with an external 8MHZ crystal oscillator to XTAL1 and XTAL2 and was able to flash the controller a couple of times, though the behavior on an external crystal was unpredictable. Even then i was getting nothing on the TX and RX pins. Am i missing out something else to get UART working?

vishalbhat88, 2021-01-29:
Also on the Arduino IDE, other than choosing the >4.5V, internal 8MHZ option, do i have to add any other header files other than <ATTinyCore.h> to get it working?

closed 2021-02-03 by vishalbhat88