SpenceKonde/ATTinyCore #498: No option for disabling RESET pin

ideaalab, 2021-01-18:
Im using an Attiny84A, and found that the microcontroller resets if the circuit is touched in a certain way. Very sure this is because the RESET pin is acting like a capacitive touch button. This is a special circuit, and the design can not be changed, so i thought of disabling the RESET pin with the fuses. Unfortunately i can not find any option to do this.
I know that doing this would disable serial programming, and i would need a HV programmer to change fuses or reprogram, but im OK with that. Just need to disable RESET pin to avoid resets.
Is it possible to implement or do in a simple way?
Thanks!

SpenceKonde, 2021-01-18:
That's correct - there are a lot of novices using this core, and damned near nobody has an HVSP - I try to handle all plausible use cases within my core... except for this one, because the portion of ATTinyCore users who can benefit from it is very small, and the portion who . My longstanding position is that a manual step should be required for this, as that will keep people from bricking their boards and then crying about it. (I'd punt and say "Well, I don't have an HV programmer, so I wouldn't be able to test it (which is true), but let's face it, I don't exactly let a need for testing 

File -> Preferences -> Verbose output during: Upload (check) 
Choose desired options, but dont connect the programmer to the chip. Tools -> Burn Bootloader.
Above the orange error text will be a line like this: 
```
C:\arduino-1.8.13-portable-skbk\hardware\tools\avr/bin/avrdude -CC:\Users\Spence\Documents\Arduino\hardware\ATTinyCore\avr/avrdude.conf -v -pattiny84 -cusbtiny -e -Uefuse:w:0xFF:m -Uhfuse:w:0b11010111:m -Ulfuse:w:0xE2:m -Uflash:w:C:\Users\Spence\Documents\Arduino\hardware\ATTinyCore\avr/bootloaders/empty/empty_all.hex:i 
```

Copy/paste into notepad or other text editor. Remove the -e, -Uefuse, -Ulfuse, and -Uflash. 

Edit the -Uhuse parameter to set the RSTDSBL fuse (the high bit becomes a 0).

You'll end up with something like 
```
C:\arduino-1.8.13-portable-skbk\hardware\tools\avr/bin/avrdude -CC:\Users\Spence\Documents\Arduino\hardware\ATTinyCore\avr/avrdude.conf -v -pattiny84 -cusbtiny -Uhfuse:w:0b01010111:m  (the exact fuse value will depend on the other settings you have)
```
Copy and paste that into a command prompt. 

This simple manual step seems a reasonable requirement for exotic, advanced options like this that could brick boards if it were made too easy.

ideaalab, 2021-01-18:
Wow, great explanation. Worked perfectly!
But do you think that an option inside the menu under a label like "DO NOT TOUCH" would be something to worry about? It would make it much easier for the ones that need it. And for most others, it would be not touched (hopefuly)

prandeamus, 2021-01-18:
"The trouble with making things foolproof is that fools are actually 
quite resourceful"

It's not my call to make, but never underestimate the ability for people 
to press buttons they should not.

:)

(#498)

ideaalab, 2021-01-18:
Yeah, i can understand... the suggested method works, but its a pity not to have an option for that anyway

SpenceKonde, 2021-01-26:
I may end up just putting this option in with a warning, since I already have it for the digispark parts...

ideaalab, 2021-02-06:
Any plan with this? Any way we can help you implement soon?
We plan to use the enable/disable of reset often, and the CMD option is not very practical.
More over, do you think if we use a HVP like the one here https://td0g.ca/2020/04/13/high-voltage-flash-programming-on-attiny85/
It would be possible enable/disable reset from within arduino IDE using the "Burn bootloader" option?

ideaalab, 2021-02-06:
@SpenceKonde i would like to do a proposal on "freelance work", but i can not find the way to contact you.

SpenceKonde, 2021-02-06:
Spencekonde@gmail.com would be the correct contact info - I try to make it available, but people keep finding it difficult to find x_x

ideaalab, 2021-02-07:
Thanks, i will contact you today or tomorrow as soon as i have some time.

Meanwhile, do you know why the program working as soon as i disable the reset and try to use it as an input?

Case 1:
Reset enabled or disabled, but pin not used, then the program works ok.

Case 2:
Reset pin declared as input_pullup in setup but reset still as reset in the fuse, program  works, but of course, when using the reset pin as input button, the program resets.

Case 3:
Same program as above, just disable reset in fuse, program stop working :(

Do you think rst pin is correctly declared in AttinyCore? What could be the cause of disabling reset and using it as input cause program to stop working?

SpenceKonde, 2021-02-24:
Will be part of the massive refactoring of boards.txt in the big release. See discussions thread on change of [plans for releases

ideaalab, 2021-02-25:
> Thanks, i will contact you today or tomorrow as soon as i have some time.
> 
> Meanwhile, do you know why the program stop working as soon as i disable the reset and try to use it as an input?
> 
> Case 1:
> Reset enabled or disabled, but pin not used, then the program works ok.
> 
> Case 2:
> Reset pin declared as input_pullup in setup but reset still as reset in the fuse, program works, but of course, when using the reset pin as input button, the program resets.
> 
> Case 3:
> Same program as above, just disable reset in fuse, program stop working :(
> 
> Do you think rst pin is correctly declared in AttinyCore? What could be the cause of disabling reset and using it as input cause program to stop working?

Disregard this message. I found what i was doing wrong. Program stopped working because writing fuse also erase memory. So i had to set the fuses before programming, otherwise "it will not work" :)

SpenceKonde, 2021-02-25:
Huh, somehow I missed that part of the question o_o  sorry about that.... 

Programming the fuses does not have to clear the flash, I didn't think it did via avrdude. You don't by any chance have a -e in the command do you (which tells it to perform a chip erase, even if not writing to the flash)? avrdude is only supposed to do a chip erase if you pass -e, or are writing something to the flash (there's a different flag you can pass, maybe -D, to tell it not to erase even when writing to the flash)

ideaalab, 2021-02-25:
If i click on the "Burn bootloader" option the "-e" flag is added to the avrdude command. I thought that if no bootloader was used, then the flash was not affected. But once i noticed the flash was erased, i started to use the avrdude command line to write the fuses.

ideaalab, 2021-03-03:
Just realize that if the fuse to disable reset is programmed with the "burn bootloader" option, then it will effectively disable reset, but will at the same time erase the flash. So it will not be very useful.. Would only be useful if could preserve flash while disabling reset. Maybe a new option in the menu just to burn fuses?

SpenceKonde, 2021-03-08:
Good point.... 
I can't create new "command" level options. Sp then I thought I'd just make a new board definition that would do things like blowing reset disable when you did burn bootloader (and print errors if you tried to compile or upload with it - but I can't read current fuse value and base what I write on that either.... 

I think we're back to it being an option so dangerous that it can't be exposed. Maybe a commented out option in boards.txt that will add a menu option to let you set reset-disable on all ISP uploads or something?

SpenceKonde, 2021-05-29:
As described above I don't see hope for a software solution here. A firmware solution. based on a modified Arduino ISP sketch that fires the HVSP reset sequence at the start of programming and then modified the high fuse to set reset disable would, I think, be the best solution here.

closed 2021-05-29 by SpenceKonde